MichaIng/DietPi #663: V140: LMS installation fails on oDroid C2 (aarch64)

audiomuze, 2016-12-25:
Apt-get has failed:                                                │                                                                                    
                                                                                   │  - Apt string = zlib1g-dev libjpeg-dev libpng-dev                  │                                                                                    
                                                                                   │ libjpeg62-turbo-dev                                                │                                                                                    
                                                                                   │  - Error code = 100                                                │                                                                                    
                                                                                   │  - Software Title = SqueezeBox: logitech media server (lms)        │                                                                                    
                                                                                   │                                                                    │                                                                                    
                                                                                   │ Please report this issue to DietPi so it can be investigated.      │                                                                                    
                                                                                   │                                                                    │                                                                                    
                                                                                   │ DietPi-Software will now exit.                                     │                                                                                    
                                                                                   │                                                                    │                                                                                    
                                                                                   │                                                                    │                                                                                    
                                                                                   │                               <Ok>

Fourdee, 2016-12-26:
@audiomuze 
Thanks for report, i'll take a look

Fourdee, 2016-12-26:
@audiomuze 
Tested ok on fresh C2 install here.

Lets try manually installing those packages and see what we get (please paste results):
```
apt-get install zlib1g-dev libjpeg-dev libpng-dev libjpeg62-turbo-dev
```

audiomuze, 2016-12-27:
`The following packages have unmet dependencies:
 libjpeg62-turbo-dev : Depends: libc-dev
 zlib1g-dev : Depends: libc6-dev but it is not going to be installed or
                       libc-dev
` 

So it would seem build-essential needs to be installed?

audiomuze, 2016-12-27:
` Welcome to DietPi-Software 

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT
 Please wait...

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  apt apt-utils libapt-inst1.5 libapt-pkg4.12
4 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 2,348 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://security.debian.org/ jessie/updates/main libapt-pkg4.12 arm64 1.0.9.8.4 [738 kB]
Get:2 http://security.debian.org/ jessie/updates/main apt arm64 1.0.9.8.4 [1,084 kB]
Get:3 http://security.debian.org/ jessie/updates/main libapt-inst1.5 arm64 1.0.9.8.4 [168 kB]                                                                                                                                               
Get:4 http://security.debian.org/ jessie/updates/main apt-utils arm64 1.0.9.8.4 [358 kB]                                                                                                                                                    
Fetched 2,348 kB in 15s (147 kB/s)                                                                                                                                                                                                          
(Reading database ... 16176 files and directories currently installed.)
Preparing to unpack .../libapt-pkg4.12_1.0.9.8.4_arm64.deb ...
Unpacking libapt-pkg4.12:arm64 (1.0.9.8.4) over (1.0.9.8.3) ...
Setting up libapt-pkg4.12:arm64 (1.0.9.8.4) ...
Processing triggers for libc-bin (2.19-18+deb8u7) ...
(Reading database ... 16176 files and directories currently installed.)
Preparing to unpack .../apt_1.0.9.8.4_arm64.deb ...
Unpacking apt (1.0.9.8.4) over (1.0.9.8.3) ...
Setting up apt (1.0.9.8.4) ...
Processing triggers for libc-bin (2.19-18+deb8u7) ...
(Reading database ... 16176 files and directories currently installed.)
Preparing to unpack .../libapt-inst1.5_1.0.9.8.4_arm64.deb ...
Unpacking libapt-inst1.5:arm64 (1.0.9.8.4) over (1.0.9.8.3) ...
Preparing to unpack .../apt-utils_1.0.9.8.4_arm64.deb ...
Unpacking apt-utils (1.0.9.8.4) over (1.0.9.8.3) ...
Setting up libapt-inst1.5:arm64 (1.0.9.8.4) ...
Setting up apt-utils (1.0.9.8.4) ...
Processing triggers for libc-bin (2.19-18+deb8u7) ...

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software
 Please wait...

 [Info] Alsa will be installed

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop
 Please wait...

 [Ok] cron

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Install software
 Please wait...

 ALSA: linux sound system
 [Info] Please wait...

 #               processing | D i e t  P i             #

Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  libasound2 libasound2-data libsamplerate0
Suggested packages:
[... 204 lines cut ...]
Building dependency tree       
Reading state information... Done
Note, selecting 'libpng12-dev' instead of 'libpng-dev'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libjpeg62-turbo-dev : Depends: libc-dev
 zlib1g-dev : Depends: libc6-dev but it is not going to be installed or
                       libc-dev
E: Unable to correct problems, you have held broken packages.
 [Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken pacakages. DietPi-Software will now exit.

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues
`

Just tried a clean install for the 3rd time.  Could you explain how you installed?

audiomuze, 2016-12-27:
`# apt-get install build-essential
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 build-essential : Depends: libc6-dev but it is not going to be installed or
                            libc-dev
                   Depends: g++ (>= 4:4.9.1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
`

Are we working off the same build?

Fourdee, 2016-12-27:
@audiomuze 

> Are we working off the same build?

Yes.

Try fixing the broken packages and deps with:
```
apt-get install -f
```

audiomuze, 2016-12-27:
Will do.

Your install works out the box?

audiomuze, 2016-12-27:
```
# apt-get install -f zlib1g-dev libjpeg-dev libpng-dev libjpeg62-turbo-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libpng12-dev' instead of 'libpng-dev'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libjpeg62-turbo-dev : Depends: libc-dev
 zlib1g-dev : Depends: libc6-dev but it is not going to be installed or
                       libc-dev
E: Unable to correct problems, you have held broken packages.

```

audiomuze, 2016-12-27:
```
# apt-get install -f build-essential
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 build-essential : Depends: libc6-dev but it is not going to be installed or
                            libc-dev
                   Depends: g++ (>= 4:4.9.1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

```

Fourdee, 2016-12-29:
@audiomuze 
Please run and paste results if any errors:
```
apt-get install libc6-dev g++ build-essential
```

Fourdee, 2016-12-29:
@audiomuze 
Could you also please send us a bug report, I'd like to check system setup and installed software:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184

Fourdee, 2016-12-29:
Note: Also report by another user:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1109#p4662
Reference CODE: f1ac5977-f26c-45f4-9eff-668573de4827-0

Fourdee, 2016-12-31:
Impossible package version situation with ```libc6``` and ```libc6-dev```:

cause:
```
root@DietPi:~# apt-get install zlib1g-dev libjpeg-dev libpng-dev libjpeg62-turbo-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libpng12-dev' instead of 'libpng-dev'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libjpeg62-turbo-dev : Depends: libc-dev
 zlib1g-dev : Depends: libc6-dev but it is not going to be installed or
                       libc-dev
E: Unable to correct problems, you have held broken packages.
```

```
root@DietPi:~# apt-get install zlib1g-dev libjpeg-dev libpng-dev libjpeg62-turbo-dev libc6-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libpng12-dev' instead of 'libpng-dev'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libc6-dev : Depends: libc6 (= 2.19-18+deb8u6) but 2.19-18+deb8u7 is to be installed
E: Unable to correct problems, you have held broken packages.

```

Fourdee, 2016-12-31:
Attempt force install of request u6 package for libc6
```
wget http://ftp.debian.org/debian/pool/main/g/glibc/libc6_2.19-18+deb8u6_arm64.deb -O package.deb
dpkg -i package.deb
rm package.deb
```

Fourdee, 2016-12-31:
@audiomuze 
As a temp fix for this run these commands: https://github.com/Fourdee/DietPi/issues/663#issuecomment-269865721

Then reinstall LMS with:
```
dietpi-software install 35
```

Fourdee, 2016-12-31:
Added to existing v140 patch (our image is < 140). Applies to all installations (fresh and existing).

Resolved for v141

Fourdee, 2016-12-31:
🈯️ Tested with fresh install of C2

closed 2016-12-31 by Fourdee