MichaIng/DietPi #664: DietPi-Software - Semi automated reverse proxys

LexiconCode, 2016-12-25:
As the many of dietpi software packages are meant for web facing services we need to be easy way to set up a reverse proxy. This ticket is to discuss methods for making this easy for the end user based on the web services provided.

Apache - Lighttpd - Nginx

Third-party software
http://www.apsis.ch/pound/
http://www.haproxy.org/ - Well documented

joaofl, 2017-03-19:
I actually though about this as well this week, after configuring it for Transmission in Apache.
I'm not sure on the other servers, but on Apache it is quiet simple to do. 
(There is a simple step by step here: http://www.htpcbeginner.com/transmission-apache-proxy-setup/)

In this case, a script would need to append some extra proxy config for each app installed.
First run it it install the modules and create the proxy file. Every time you run it, it checks for each server you are using, and which apps you wanna add to the proxy list. With some sort of check boxes, e chose the apps that should be added to the proxy config file to be generated.

Fourdee, 2018-04-05:
Marking as closed due to:
- No activity/interest from our user base or dev team.

closed 2018-04-05 by Fourdee