MichaIng/DietPi #666: DietPi-Software | Nextcloud requires database set up when using dietpi.txt global password

LexiconCode, 2016-12-26:
When installing Nextcloud via dietpi.txt global password utilizing the global password setting causes a bug. When viewing the web interface for the first time with Nextcloud where you would use admin and the global password Nextcloud asks for the database to be set up. This should already be done and is not included in the instructions to do so dietpi.com

I followed the instructions using a password for the global password that excludes the following characters \"$

An example of the password I used
45nau@55q3J%44^ayr6#354hyt5n^oy2.Nnim^ByZhy

Fourdee, 2016-12-26:
@LexiconCode 

Thanks for report 👍, i'll take a look

Linked for reference: https://github.com/Fourdee/DietPi/issues/647#issuecomment-267019815

Fourdee, 2016-12-28:
@LexiconCode 

Turns this also effects all programs with SQL DB generation, when a custom password is used. Lets gets this fixed.

Fourdee, 2016-12-28:
Tested with password provided by @LexiconCode:
- 🈯️ 

```
root@DietPi:~# cat /DietPi/dietpi.txt | grep Global
#Global Password to be applied for the system
AUTO_Global_Password=45nau@55q3J%44^ayr6#354hyt5n^oy2.Nnim^ByZhy
```
```
dietpi-software install 114
```

```
[Info] Creating MySql DB for nextcloud

 [Ok] Create database nextcloud | Completed

Nextcloud is not installed - only a limited number of commands are available
Nextcloud was successfully installed
```

Completed. Fixed for v141.

closed 2016-12-28 by Fourdee

LexiconCode, 2016-12-28:
Originally what password was being used? or did the DB Set up fail?

Fourdee, 2016-12-28:
@LexiconCode 

> Originally what password was being used? or did the DB Set up fail?

DB creation was set to use ```dietpi``` as the MySQL root account, instead of AUTO_Global_Password=