MichaIng/DietPi #669: DietPi-Software | Apt simulation failing

Fourdee, 2016-12-27:
> ```
Update & upgrade APT
Please wait...

Ign http://ftp.cn.debian.org jessie InRelease
Get:1 http://ftp.cn.debian.org jessie-updates InRelease [145 kB]
Hit http://security.debian.org jessie/updates InRelease
E: The method driver /usr/lib/apt/methods/https could not be found.
N: Is the package apt-transport-https installed?
[Info] Running apt simulation to check for errors, please wait...
[Failed] Apt simulation | An issue has occured

[Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken pacakages. DietPi-Software will now exit.

[Info] Please try:
[Info] - Fixing apt package errors with 'apt-get install -f'
[Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
[Info] - Clear and refresh apt 'apt-get update'
[Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
- https://github.com/Fourdee/DietPi/issues
```

Fourdee, 2016-12-27:
Iits as if Debian has forced https redirect on their security repo overnight.

Solution:
- ARMv7 = 
```
wget http://security.debian.org/debian-security/pool/updates/main/a/apt/apt-transport-https_1.0.9.8.4_armhf.deb
dpkg -i apt-transport-https_1.0.9.8.4_armhf.deb
wget http://security.debian.org/debian-security/pool/updates/main/c/curl/libcurl3-gnutls_7.38.0-4+deb8u5_armhf.deb
dpkg -i libcurl3-gnutls_7.38.0-4+deb8u5_armhf.deb
```

Fourdee, 2016-12-27:
```apt-transport-https``` will now be installed by default on all DietPi devices.

thanhvo31, 2016-12-28:
Hi

Here is my error

> dpkg: dependency problems prevent configuration of apt-transport-https:
 apt-transport-https depends on libapt-pkg4.12 (>= 1.0.9.8.4); however:
  Version of libapt-pkg4.12:armhf on system is 1.0.9.8.3.

>

k-plan, 2016-12-28:
Hi,

you can test, if this is working for you:

```
root@Odroid-C1:~# apt-get install apt-transport-https
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  apt-transport-https
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/135 kB of archives.
After this operation, 170 kB of additional disk space will be used.
Selecting previously unselected package apt-transport-https.
(Reading database ... 19297 files and directories currently installed.)
Preparing to unpack .../apt-transport-https_1.0.9.8.4_armhf.deb ...
Unpacking apt-transport-https (1.0.9.8.4) ...
Setting up apt-transport-https (1.0.9.8.4) ...
```

If you will ask that packet can't be verify, answer to install it anyway.

Fourdee, 2016-12-28:
@thanhvo31 

If @k-plan post above does not resolve it.

Try installing the request version of this package (``` libapt-pkg4.12_1.0.9.8.4```):
```
wget http://security.debian.org/debian-security/pool/updates/main/a/apt/libapt-pkg4.12_1.0.9.8.4_armhf.deb
dpkg -i libapt-pkg4.12_1.0.9.8.4_armhf.deb
```

thanhvo31, 2016-12-28:
@Fourdee , @k-plan 
 Confirm @k-plan post above does not resolve it.
 installing the request version of this package (libapt-pkg4.12_1.0.9.8.4): worked!

Fourdee, 2016-12-30:
Dammit,

I'll need to update all our images affected by this, pre-installing ```apt-transport-https```. Else, all users will experience this issue on each fresh install.

- RPi (is not effected)
- Everything else

aliosa27, 2016-12-31:
I'll regenerate a few of those. all i need to add is the apt-transport-https package correct?

Fourdee, 2016-12-31:
@aliosa27 

> I'll regenerate a few of those. all i need to add is the apt-transport-https package correct?

Yep. Please, would be a great help 👍 

I'll try and get v141 released early today. Added patch to grab ```apt-transport-https``` package from apt, if that fails, it will manually pull them in via ```wget```.

Fourdee, 2016-12-31:
I'll mark this as resolved, patch system will fix fresh and existing installations, however, i'd still like to update our images and add ```apt-transport-https``` by default, just incase.

@aliosa27 
I need to re-do the RPi+XU4 image, if you can do any others i'd be grateful. New ticket: https://github.com/Fourdee/DietPi/issues/676

closed 2016-12-31 by Fourdee