MichaIng/DietPi #671: DietPi on Berryboot (Raspberry Pi)

agoldcheidt, 2016-12-27:
I convert DietPi for Berryboot using the following method:
http://www.berryterminal.com/doku.php/berryboot/adding_custom_distributions

On the first boot DietPi give some errors about the detection of the device. I check a lot of DietPi scripts and from this point and I don't think I can't fix this from my end.

Fourdee, 2016-12-28:
@agoldcheidt 

Thanks, i'll take a look. Most likely we need to add some extra code to support BerryBoot partitioning.

Fourdee, 2016-12-28:
Notes:

```
#7Zip prereqs
apt-get install -y p7zip

wget 'http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z' -O image.7z
p7zip -d image.7z

```

```
sudo kpartx -av image_you_want_to_convert.img 
add map loop0p1 (252:5): 0 117187 linear /dev/loop0 1
add map loop0p2 (252:6): 0 3493888 linear /dev/loop0 118784
$ sudo mount /dev/mapper/loop0p2 /mnt
$ sudo sed -i 's/^\/dev\/mmcblk/#\0/g' /mnt/etc/fstab
$ sudo mksquashfs /mnt converted_image_for_berryboot.img -comp lzo -e lib/modules
$ sudo umount /mnt
$ sudo kpartx -d image_you_want_to_convert.img 
```

k-plan, 2016-12-28:
Hi,

what's your concern? DietPi a prebuilt image like this one? 

https://sourceforge.net/projects/berryboot/files/os_images/

Yes, Berryboot is really cool stuff, use it and I like it much, before I discovered DietPi.
Especially the new universal version for all sort of RPis: 

http://www.berryterminal.com/doku.php/berryboot#berryboot_v20_-_bootloaderuniversal_operating_system_installer

And I like to have support for some new ARM devices DietPi assisted:

k-plan, 2016-12-28:
thinks, to get supported (by default):

- [keyboard layout](http://www.berryterminal.com/doku.php/berryboot/adding_custom_distributions#tweaks)

- [Headless installation](http://www.berryterminal.com/doku.php/berryboot/headless_installation)

- [Storing your files on a NAS (using iSCSI)](http://www.berryterminal.com/doku.php/berryboot/headless_installation) and  [on network share (CIFS)](http://www.berryterminal.com/doku.php/berryboot/local_repository_on_network_share)

Fourdee, 2016-12-28:
@k-plan 

> And I like to have support for some new ARM devices DietPi assisted:
https://github.com/maxnet/berryboot#to-add-support-for-a-new-arm-device
But it is a more a "wet dream" and a lot of coding work ... for @Fourdee 

😆. I'll get this working for RPi for now. See how it goes, may be a one off to support RPi + DietPi on BerryBoot.

Fourdee, 2016-12-28:
Looking good.

I believe the issue was simply due to the BerryBoot image creation process, not copying ```/boot``` to the newly created single partition.

So:
```
#Copy DietPi scripts from p1 boot and transfer to p2
cp -R /mnt/loopp1/* /mnt/loopp2/boot/
```

I'll run some tests.

k-plan, 2016-12-28:
> _Tweaks
If the image you are converting is based on Debian/Raspbian delete the etc/console-setup/cached_UTF-8_del.kmap.gz file before converting the image to force regeneration of the cached keyboard mapping on first boot. This is to make sure it uses the keyboard layout set in Berryboot instead of default British._

Fourdee, 2016-12-28:
@agoldcheidt 

Ok working fine. Steps i used to create the squashed image:
https://github.com/Fourdee/DietPi/issues/671#issuecomment-269476497

Also uploaded image for loading in BerryBoot:
http://dietpi.com/downloads/testing/DietPi_berryboot_RPi.7z

There may be issues/bugs with certain features that BerryBoot controls at boot time (eg: GPU memory splits). 
But we'll need users help to test and report issues back to us so we can try and get them resolved. Although, there will be some issues that cannot simply be resolved due to the "container" BerryBoot puts DietPi in. So expect a few bugs here and there.

agoldcheidt, 2016-12-30:
Thanks @Fourdee, fast support ever, I will testing the OS image.

Fourdee, 2017-01-06:
@agoldcheidt 

Any chance you have been able to test the BootBerry image (http://dietpi.com/downloads/testing/DietPi_berryboot_RPi.7z)?

agoldcheidt, 2017-01-06:
@Fourdee 

Yes, tested on a Raspberry Pi 2 and it works great. I share your OS image for Berryboot through my website (http://berryboot.alexgoldcheidt.com/images/) in there I added a link to this issue post.

Thank you so much for converting DietPi for Berryboot.

Fourdee, 2017-01-07:
@agoldcheidt 

> Yes, tested on a Raspberry Pi 2 and it works great. I share your OS image for Berryboot through my website (http://berryboot.alexgoldcheidt.com/images/) in there I added a link to this issue post.

Excellent, great to hear its working 👍 

I'll mark this as closed for now, but if anyone experiences any issues with the BerryBoot image, please reopen or create another ticket.

closed 2017-01-07 by Fourdee

cms-, 2017-05-02:
Hello, I'm attempting to use DietPi with Berryboot for a project and seem to have hit a little snag with accessing kernel modules. Using the dietpi_jessie_v145_all_rpi_2017.03.08_berryboot.img.tar image, both my Pi 2 and Samsung Pi 1 exhibit the following behavior. 

Namely, any packages that depend on kernel modules seem to fail. I noticed there seems to be a difference between the kernel version in uname and the directory names in /lib/modules, but my poking at this with a symlink did not yield much insight.

Can anyone confirm whether loading kernel modules is possible using this image?

``[    0.000000] Linux version 4.4.50v7-aufs (max@lynx) (gcc version 4.8.4 (Buildroot 2015.02-gf4a5ebb-dirty) ) #1 SMP Wed Mar 8 12:18:30 CET 2017``
``[   24.110761] fuse: disagrees about version of symbol module_layout``
``[   24.123051] systemd[1]: Failed to start Load Kernel Modules.``

/lib/modules:

> drwxr-xr-x  3 root   root   4096 Apr 27 17:04 4.4.50+
> drwxr-xr-x  3 root   root   4096 Apr 30 02:16 4.4.50-v7+
> lrwxrwxrwx  1 root   root     12 Apr 30 01:56 4.4.50v7-aufs -> ./4.4.50-v7+

With no added symlink, boots as follows:

``[   28.921308] systemd[1]: systemd-modules-load.service: main process exited, code=exited, status=1/FAILURE``
``[   28.924592] systemd[1]: Failed to start Load Kernel Modules.``
``May  2 01:29:02 dietpi systemd-modules-load[167]: could not open moddep file '/lib/modules/4.4.50v7-aufs/modules.dep.bin'``
``May  2 01:29:02 dietpi systemd-udevd[215]: could not open moddep file '/lib/modules/4.4.50v7-aufs/modules.dep.bin'``

``May  2 01:44:26 dietpi systemd[1]: Starting Load Kernel Modules...``
``May  2 01:44:26 dietpi systemd-modules-load[2644]: could not open moddep file '/lib/modules/4.4.50v7-aufs/modules.dep.bin'``
``May  2 01:44:26 dietpi systemd-modules-load[2644]: Failed to lookup alias 'fuse': Function not implemented``
``May  2 01:44:26 dietpi systemd[1]: systemd-modules-load.service: main process exited, code=exited, status=1/FAILURE``
``May  2 01:44:26 dietpi systemd[1]: Failed to start Load Kernel Modules.``
``May  2 01:44:26 dietpi systemd[1]: Dependency failed for netfilter persistent configuration.``
``May  2 01:44:26 dietpi systemd[1]: Unit systemd-modules-load.service entered failed state.``

symbios24, 2018-02-22:
This method Method is not working neither for Jessie or for Stretch boot is halt and i get the following errors:

failed to start dietpi-ramlog
failed to start dietpi-ramdisk
dependency failed for /etc/rc.local compartibility
dependency failed for Dietpi-Boot

symbios24, 2018-02-22:
This Method is not working neither for Jessie or for Stretch boot is halted and i get the following errors:

failed to start dietpi-ramlog
failed to start dietpi-ramdisk
dependency failed for /etc/rc.local compartibility
dependency failed for Dietpi-Boot

vikasjee, 2018-05-10:
What is the plan to fix this issue?

Fourdee, 2018-05-11:
@vikasjee 

We do not officially support BerryBoot or third party images at this stage. This remains a community effort.