MichaIng/DietPi #677: DietPi-Software | Public key is not available from 'apt-get update'

LexiconCode, 2016-12-29:
Hardware: oDroid C2 (aarch64)
Configuration: Completely fresh install

During the use of the command
apt-get update

**Produces**
Reading package lists... Done
W: GPG error: http://www.deb-multimedia.org jessie InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 5C808C2B65558117

Fourdee, 2016-12-31:
@LexiconCode 

Thanks for report.

Yep, it appears we left in the deb multimedia repo on our current C2 image. This is patched for new installations in v141, removing the repo and therefore error.

Fourdee, 2017-01-01:
Resolved for v142 patch:
```
		#-------------------------------------------------------------------------------
		#Deb multimedia repo left on C2 image, remove for ALL installs: https://github.com/Fourdee/DietPi/issues/677
		#	debmultimedia.list is used by dietpi-software.
		rm /etc/apt/sources.list.d/deb-multimedia.list &> /dev/null
```

closed 2017-01-01 by Fourdee