MichaIng/DietPi #684: General | Disable auto mount of USB_1

Fourdee, 2017-01-01:
Although extremely useful, when the drive is not attached:
- Any access to ```/mnt/usb_1``` will result in a hang at system level.

This effects all software ```ls -lha``` Kodi etc.

Fourdee, 2017-01-01:
This will break: 
https://github.com/Fourdee/DietPi/issues/501

Fourdee, 2017-01-02:
This will only take effect when user next runs ```dietpi-drive_manager```, or, we need to patch their ```/etc/fstab```

xenfomation, 2017-01-04:
Out of curiosity, cant' we leverage dmesg & or lsblk before/inside surrounding scripts as so we prevent a false action on a missing drive?  And yes, it is good to be back!  New career and as you can tell -- free time!

Fourdee, 2017-01-04:
@xenfomation 

> New career and as you can tell -- free time!

Great to hear 👍 Hope your enjoying your new career? 

> Out of curiosity, cant' we leverage dmesg & or lsblk before/inside surrounding scripts as so we prevent a false action on a missing drive?

I think the issue is no response is given by system, when attempting to access a location that is unavailable using ```x-systemd.automount``` in fstab. 
Everything hangs, even a simple ```ls -lha /mnt/usb_1```, or when a user browses the ```/mnt``` folder in Kodi. So if you can find a way to resolve this at system level and keep the auto mount enabled, i'am game 😃

Fourdee, 2017-01-04:
Ok patch done for this. I'll mark this as closed.

@xenfomation 

If you find a way to make this work with ```x-systemd.automount```, please reopen 👍

closed 2017-01-04 by Fourdee