MichaIng/DietPi #686: DietPi-Software | Gitea

Facel3ss1, 2017-01-02:
Hello, I see you have Gogs on your software list. Gitea ([gitea.io](https://gitea.io)) is a fork of Gogs which looks to be better maintained gogits/gogs#1304 (?) So, I'd imagine the installation process would be similar to Gogs. Gitea is on it's 1.0 release as of writing this. I also would say that Gitea is more active than Gogs (Gitea has merged 42 PRs in the last week, Gogs merged 2). If adding Gitea to the software list is possible, it would be appreciated. Thank you for your help.

DJCrashdummy, 2017-07-12:
documentation on their homepage: https://docs.gitea.io/en-us/install-from-binary/

Fourdee, 2017-07-13:
> Binaries: https://dl.gitea.io/gitea/master/

Bridouz, 2017-07-21:
From what I understand it will be really the same process as gogs to include gitea in dietpi.

@Fourdee If you want testers I'm in :) Trying right now to install gitea on my DietPi powered Rpi3

Bridouz, 2017-09-25:
@Fourdee 

Did a bit of testing yesterday and tried to make a PR on `dietpi-software` but the file is damn too big for my poor skills (And I probably messed up the file )

I created a gist with a few things I managed to format according to Dietpi-Software syntax 

(https://gist.github.com/Bridouz/4a264d317c9690b7b28b1246b011ca92)

Obviously just as Gogs it needs `openssh`.

And I think that's it, tell me if something it's wrong :)

Thanks again for `Dietpi`, a really nice layer to use on top of Raspbian.

techdabbler, 2017-09-28:
With the binaries available this one looks pretty easy to add. I might take a crack at it though I'm out of town for a few days.  One question -- is this a software request add or a replace? Should I be looking to remove Gogs or just add Gitea?

techdabbler, 2017-10-07:
@Bridouz  I used your gist as a great jumping off point to work on adding this. I made a few changes the biggest one is I used this systemd script for running as a daemon instead of writing the init script from dietpi-software

https://github.com/go-gitea/gitea/blob/master/contrib/systemd/gitea.service

One hangup I'm working on is I have to run gitea one time before it will run as a service. I'm suspecting some sort of .conf file it is looking for. If anyone has any idea throw it my way.  I also haven't figured out how to redirect the repo storage to diet_userdata.

Check it out if you like
https://github.com/techdabbler/DietPi/blob/gitea-add/dietpi/dietpi-software

techdabbler, 2017-10-07:
Ok I'm at the initial config point and I'm having trouble knowing what the pw for MySQL is. I tried 'dietpi' from the gogs setup in the wiki and that didn't work.  So either I'm reading the code/wiki wrong or I've got a different issue.  I got this running gitea both as root and as git (the user the systemd file uses - though I can change it if I need to). 

[image]

Bridouz, 2017-10-08:
thanks for the fixes, rewrites and commits. I will test it tomorrow on my dietpi.

what do you think about creating a _git_ user @Fourdee ?

Is it OK or do you want to stick with `root` user only for now ?

techdabbler, 2017-10-08:
@Bridouz I look forward to hearing how your test goes. Also, if you have any notes on uninstallation I'd appreciate them. Removing the executable and data is easy it is the command to drop the db from mysql that I haven't tried hunting down yet.

@Fourdee when marking an install as needing a pre-requisite software package (ie mysql, git, openssh in this case) when the software (gitea) is un-installed does diet-pi software automatically remove mysql or git for example? Or roll back dropbear? Or is there a way for the un-installer to call that via dietpi commands? Or is this something that is even recommended to consider?

Fourdee, 2017-10-08:
@techdabbler 

> @Fourdee when marking an install as needing a pre-requisite software package (ie mysql, git, openssh in this case) when the software (gitea) is un-installed does diet-pi software automatically [...]

We don't remove pre-requisite/library software, during uninstalls, unless the user has specifically selected it for uninstall.

EG: Remove MPD, does not remove ALSA, as it could be needed for other install programs.
EG: Remove MPD, and user also selects ALSA for removal, removes both.
[image]

techdabbler, 2017-10-08:
@Fourdee Thanks. That is what I expected. Just want to make sure I'm doing uninstalls as complete as possible.

Bridouz, 2017-10-11:
Gitea working on DP Stretch Rpi3 :)

Do we stick with that `git` user creation or do we keep all software running with `root` @Fourdee ?

Fourdee, 2017-10-11:
@Bridouz 

> Do we stick with that git user creation

Ideally, stick with the user as defined by the software, just be aware may need to set permissions for that user in ```/mnt/dietpi_userdata/SOFTWARETITLE``` if used and requires it.

> PS: I also have a nginx config for a redirection from localhost:3000 to server_name/gogs if you want.

I personally prefer to keep the installation URL's and ports stock. The only exception being http://0.0.0.0/pihole because it make sense :)
Leave redirection and customization down to the end user.

techdabbler, 2017-10-11:
@Bridouz Good to hear it is working for you too. I just committed uninstall section. I haven't tested yet but would you mind giving it a test in RPi3 stretch for me? Once that is in place I think we are readying to pull request it.

Fourdee, 2017-10-19:
@techdabbler Thanks for adding this in 👍 

### Ready for testing, Install and uninstall:
- ARMv6
- ARMv7
- x86_64

Fourdee, 2017-12-09:
Online Doc:
- 🈯️ http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=9863#p9863

Fourdee, 2017-12-10:
Tests:
- 🈯️ ARMv7
- 🈯️ ARMv8
- 🈯️ x86_64

Completed.

closed 2017-12-10 by Fourdee