MichaIng/DietPi #688: snmpd not working for polling from a manager!!!

tusharvb19, 2017-01-03:
When installed I am not able to poll from predefined OIDs from MIB browser while this is working in Core ubuntu.

Fourdee, 2017-01-03:
@tusharvb19 

Thanks for the report.

Which device are you running?

tusharvb19, 2017-01-05:
I am using NanoPi NEO.

Fourdee, 2017-01-06:
@tusharvb19 

Thanks.

I believe NanoPi Neo serial is enabled by default but we need to verify.

- Does a basic serial console work (excluding the python script)?
- Which kernel are you running ```uname -a```?. If its 3.x, try using the updated 4.9 kernel image and retest: http://dietpi.com/downloads/images/DietPi_NanoPiNEO-armv7-(Jessie).7z

I'll make a note to test NanoPi Neo serial console, but you may beat me too it 😉

tusharvb19, 2017-01-08:
The serial console is working but I have to interface GSM module to UART1 thus ttyS1 but still after enabling them I am not able to make connection through screen to test basic AT commands.
The screen get exit automatically. And the other python serial code comes out with an error "can not configure port..."
Here is the link of fona(GSM module) tutorial which is working in other distro..core ubuntu!

https://learn.adafruit.com/fona-tethering-to-raspberry-pi-or-beaglebone-black/setup

tusharvb19, 2017-01-30:
Help me please to solve this...I am not able to use UART1!

tusharvb19, 2017-02-08:
Is there anyone who can make the ttyS1 work to get connected to various modules that operates on UART i.e. TX-RX. I have to interface SIM900 module(sic).

tusharvb19, 2017-02-20:
I need the solution to use dietPi as a my deafult os. Can someone try and help me to get it working!
(sic)

Fourdee, 2017-02-24:
@tusharvb19 

> I need the solution to use dietPi as a my deafult os. Can someone try and help me to get it working!

Unfortunately, anything beyond standard serial connection is not something I have experience with. I cannot offer any assistance in this case.

tusharvb19, 2017-02-25:
The serial ports are there ttys1but I am not able to get it working for serial interface..it shows could not configure port/io error!

DickyT, 2017-05-18:
@tusharvb19 I have the same problem, both uart1&2 is not working, but there is a workaround, use ssh to login your pi, and use uart0 (debug uart) for your serial device

Fourdee, 2018-02-22:
Neo is no longer supported. Marking as closed.

closed 2018-02-22 by Fourdee