MichaIng/DietPi #691: USB hardrive permission error

crustovsky, 2017-01-03:
When setting up a USB hard drive on Raspberry Pi 3, I'm getting "DietPi was unable to create your user data directory:  ERROR: Unable to create test file in target directory /mnt/usb_1. Check permissions. DietPi-Software will now exit." error when running DietPi v141. Any workaround or planned fixes for this ?

Fourdee, 2017-01-04:
@crustovsky 

Thanks for report.

Can you please send us a bug report, it will contain some system information I can use to assist in this issue:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184

crustovsky, 2017-01-04:
I just posted bug report as instructed, reference code 197b1674-0716-4b78-a9bb-5c7137a801e0-0

Fourdee, 2017-01-05:
@crustovsky 
Thanks 👍 

Please paste results of following commands:
```
ls -lha /mnt
ls -lha /mnt/usb_*
```

And which drive was used when moving user data (eg: /mnt/usb_1)?

crustovsky, 2017-01-05:
Had to do this on /mnt/usb_2 as usb_1 is currently used by flash drive (spinning drive on usb_2 is still causing the error mentioned earlier)

root@crustovsky-DietPi3:~# ls -lha /mnt
total 40K
drwxr-xr-x 10 root root 4.0K Jan  3 22:36 .
drwxr-xr-x 20 root root 4.0K Jan  3 20:42 ..
lrwxrwxrwx  1 root root   10 Jan  3 22:36 dietpi_userdata -> /mnt/usb_1
drwxr-xr-x  2 root root 4.0K Dec 31 15:44 ftp_client
drwxr-xr-x  2 root root 4.0K Dec 31 15:36 nfs_client
drwxr-xr-x  2 root root 4.0K Dec 31 15:44 samba
drwxrwxr-x  7 root root 4.0K Jan  4 19:28 usb_1
drwxr-xr-x  3 root root 4.0K Jan  5 22:07 usb_2
drwxr-xr-x  2 root root 4.0K Jan  3 20:41 usb_3
drwxr-xr-x  2 root root 4.0K Jan  3 20:41 usb_4
drwxr-xr-x  2 root root 4.0K Jan  3 20:41 usb_5
root@crustovsky-DietPi3:~# ls -lha /mnt/usb_*
/mnt/usb_1:
total 101M
drwxrwxr-x   7 root root 4.0K Jan  4 19:28 .
drwxr-xr-x  10 root root 4.0K Jan  3 22:36 ..
drwxrwxr-x   2 root root 4.0K Jan  3 20:42 downloads
drwxrwxr-x   2 root root  16K Jan  3 22:35 lost+found
drwxrwxr-x 145 root root 4.0K Jan  4 08:39 Music
drwxrwxr-x   2 root root 4.0K Jan  3 20:42 Pictures
-rwxrwxr-x   1 root root 100M Jan  3 22:36 .swapfile
drwxrwxr-x   3 root root 4.0K Jan  3 23:42 Video

/mnt/usb_2:
total 24K
drwxr-xr-x  3 root root 4.0K Jan  5 22:07 .
drwxr-xr-x 10 root root 4.0K Jan  3 22:36 ..
drwx------  2 root root  16K Jan  5 22:07 lost+found

/mnt/usb_3:
total 8.0K
drwxr-xr-x  2 root root 4.0K Jan  3 20:41 .
drwxr-xr-x 10 root root 4.0K Jan  3 22:36 ..

/mnt/usb_4:
total 8.0K
drwxr-xr-x  2 root root 4.0K Jan  3 20:41 .
drwxr-xr-x 10 root root 4.0K Jan  3 22:36 ..

/mnt/usb_5:
total 8.0K
drwxr-xr-x  2 root root 4.0K Jan  3 20:41 .
drwxr-xr-x 10 root root 4.0K Jan  3 22:36 ..
root@crustovsky-DietPi3:~#

Fourdee, 2017-01-06:
@crustovsky 

Thanks for results.

Did you use ```dietpi-drive_manager```, or ```dietpi-software``` to move the user data?

> /mnt/usb_2:
total 24K
drwxr-xr-x 3 root root 4.0K Jan 5 22:07 .
drwxr-xr-x 10 root root 4.0K Jan 3 22:36 ..
drwx------ 2 root root 16K Jan 5 22:07 lost+found

Permissions are fine. Its not the cause. 

Lets try manually writing a file to usb_2:
```
echo 1 > /mnt/usb_2/testfile
```
If that works, try moving userdata again with:
```
/DietPi/dietpi/func/dietpi-set_userdata $(readlink -f /mnt/dietpi_userdata) /mnt/usb_2
```

crustovsky, 2017-01-08:
@Fourdee it failed both at startup setup time and when i used drive manager from launcher. What is strange is that i wasn't able to setup that drive (probably incompatible enclosure) with XU4, i will check with other one once it arrives.

I had to move back to Pi3 to test this (hence swapping back _2 to _1) and I got an error again from setup wizard and commands you mentioned output

root@crustovsky-DietPi3:~# echo 1 > /mnt/usb_1/testfile
-bash: /mnt/usb_1/testfile: Read-only file system
root@crustovsky-DietPi3:~#

root@crustovsky-DietPi3:~# /DietPi/dietpi/func/dietpi-set_userdata $(readlink -f /mnt/dietpi_userdata) /mnt/usb_1

 DietPi
─────────────────────────────────────────────────────
 Mode: Updating user data location
 Please wait...

 [Info]  - From : /mnt/dietpi_userdata
 [Info]  - To   : /mnt/usb_1
 [Info] Please wait...
/DietPi/dietpi/func/dietpi-set_userdata: line 245: /mnt/usb_1/.testfile_dietpi_userdata_basedirectory: Read-only file system
 [Failed] ERROR: Unable to create test file in target directory /mnt/usb_1. Check permissions.

 [Failed] An issue has occured

I have Pi3 dietpi setup on a spare card for more tests.

Fourdee, 2017-01-08:
@crustovsky 

> -bash: /mnt/usb_1/testfile: Read-only file system
/DietPi/dietpi/func/dietpi-set_userdata: line 245: /mnt/usb_1/.testfile_dietpi_userdata_basedirectory: Read-only file system

Looks like the cause. Drive is being mounted as read only. Possible due to a corruption on drive.

Lets try dismounting and remounting drive to check for any errors on screen:
```
umount /mnt/usb_1
mount -a
```

Also, it might be worth trying to format the drive with GPT partition table (if not done already?). GPT is default, required for 2TB+ drives. Backup your data 1st:
```
dietpi-drive_manager
```

crustovsky, 2017-01-08:
@Fourdee 

Unmounting and mounting back produces no output

```
root@crustovsky-DietPi3:~# umount /mnt/usb_1
root@crustovsky-DietPi3:~# mount -a
root@crustovsky-DietPi3:~#
```
And drive is already formatted to GTP EXT4 type

[image]

I will try to reformat it on my Windows machine and scan it there for any errors

crustovsky, 2017-01-10:
@Fourdee so i got a new enclosure and after attaching that drive (full reformat was done under windows first though) to my XU4 I'm now moving my user data to this without any issues, if i find another hardrrive i will try again with that old enclosure and Pi3. Strange. Also believe it or not but drive could not be detected and usb3 pins on a board required clean as instructed here

[](http://forum.odroid.com/viewtopic.php?f=95&t=15302)

I will update this once I get other drive for Pi3.

Fourdee, 2017-02-05:
@crustovsky 

Marking as closed. Please re-open if required.

closed 2017-02-05 by Fourdee