MichaIng/DietPi #692: Adding Support for NeoVim

joshuamcginnis, 2017-01-03:
I've been working to get a successful build of [neovim](https://neovim.io/) on the Odroid 2 and finally had some success, but it took some [tweaks to get it working](https://github.com/neovim/neovim/issues/5861#issuecomment-270216617). 

In my conversation with the project maintainers, they suggested I reach out and inquire as to how we can get `neovim`, `python-neovim`, and `python3-neovim` as supported packages using DietPi's package management system.

neovim is already supported with apt in Debian and Ubuntu as documented here:
https://github.com/neovim/neovim/wiki/Installing-Neovim

Here's the thread re: building neovim from source on Odroid here:
https://github.com/neovim/neovim/issues/5861

Is there a way I can help get neovim supported as DietPi's package? My thanks in advance.

Fourdee, 2017-01-04:
> Is there a way I can help get neovim supported as a supported package for DietPi? My thanks in advance.

Yep, i'll take a look and see whats involved.

Should be two options for install method:
- Sourcebuild: Where its compiled from source during install
- Deb packages: Where we create .Deb's and pull them in to install.

Source build is to ensure latest version, but can fail if libraries/compile steps change.

Deb packages preferred, with testing ensures working, stable installations. But they will need to be re-built to match latest versions.

Either way, i'll know more once I've done some tests based on your notes. Leave it with me.

Fourdee, 2017-01-04:
Quick compile C2 notes (based on @joshuamcginnis original notes):
```
apt-get install -y git build-essential checkinstall libtool libtool-bin autoconf automake cmake g++ pkg-config unzip
apt-get install -y lua5.1 luarocks libuv1-dev

---

#lua stuff
luarocks install lpeg
luarocks install mpack
luarocks install luabitop

---

#Build msgpack-c
cd "$HOME"
git clone --depth=1 https://github.com/msgpack/msgpack-c.git
cd msgpack*
cmake .
make -j $(nproc --all)
#make install

# - Create .deb
#checkinstall

---

#Build neovim
cd "$HOME"
git clone --depth=1 https://github.com/neovim/neovim.git
cd neovim
make -j $(nproc --all)

# - Create .deb
#checkinstall

```

@joshuamcginnis 
Looks like its failing on LuaJIT 2.0.4, dep required by NeoVim, ARM64 (v8) not supported:
```
==== Building LuaJIT 2.0.4 ====
make[5]: Entering directory '/root/neovim/.deps/build/src/luajit/src'
Creating directories for 'msgpack'
lj_arch.h:55:2: error: #error "No support for this architecture (yet)"

#
---

#LUA jit
cd "$HOME"
git clone --depth=1 https://github.com/LuaJIT/LuaJIT.git
cd Lua*
make -j $(nproc --all)
#make install

#Same errors
```
Did you receive the same issues during compile? And, have you been successful in compiling and installing NeoVim on ARM64?

joshuamcginnis, 2017-01-04:
> Looks like its failing on LuaJIT 2.0.4, dep required by NeoVim, ARM64 (v8) not supported:

@Fourdee That's correct. [This issue](https://github.com/neovim/neovim/issues/4714) has a discussion from another Odroid user regarding this failure. It's a little unclear, but I believe the user was able to get a successful build of luajit with a patch.

However, I opted to configure the build to NOT compile luajit (and others) since the neovim maintainers claimed it was only needed for running tests.

This comment has my exact build steps for getting a successful build:
https://github.com/neovim/neovim/issues/5861#issuecomment-270216617

Fourdee, 2017-01-09:
@joshuamcginnis 

Ok, tested again, using your steps located here, still no joy: https://github.com/neovim/neovim/issues/5861#issuecomment-270216617

I cant seem to get past ```make deps DEPS_CMAKE_FLAGS="-DUSE_BUNDLED_LIBUV=OFF -DUSE_BUNDLED_MSGPACK=OFF -DUSE_BUNDLED_LUAROCKS=OFF -DUSE_BUNDLED_LUAJIT=OFF"```:
```
In file included from /root/neovim/.deps/build/src/luv/src/luv.c:18:0:
/root/neovim/.deps/build/src/luv/src/luv.h:19:17: fatal error: lua.h: No such file or directory
 #include <lua.h>
                 ^
compilation terminated.
CMakeFiles/luv.dir/build.make:54: recipe for target 'CMakeFiles/luv.dir/src/luv.c.o' failed
make[6]: *** [CMakeFiles/luv.dir/src/luv.c.o] Error 1
make[6]: Leaving directory '/root/neovim/.deps/build/src/luv-static-build'
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/luv.dir/all' failed
make[5]: *** [CMakeFiles/luv.dir/all] Error 2
make[5]: Leaving directory '/root/neovim/.deps/build/src/luv-static-build'
Makefile:117: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/root/neovim/.deps/build/src/luv-static-build'
CMakeFiles/luv-static.dir/build.make:107: recipe for target 'build/src/luv-static-stamp/luv-static-build' failed
make[3]: *** [build/src/luv-static-stamp/luv-static-build] Error 2
make[3]: Leaving directory '/root/neovim/.deps'
CMakeFiles/Makefile2:235: recipe for target 'CMakeFiles/luv-static.dir/all' failed
make[2]: *** [CMakeFiles/luv-static.dir/all] Error 2
make[2]: Leaving directory '/root/neovim/.deps'
Makefile:76: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/root/neovim/.deps'
Makefile:75: recipe for target 'deps' failed
make: *** [deps] Error 2
```

```
The dependency target "libuv" of target "luv-static" does not exist.
Call Stack (most recent call first):
  CMakeLists.txt:160 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.
```

So as suggested here (```make ```, not ```make deps```: https://github.com/neovim/neovim/issues/4692#issuecomment-216345030

Tried:
```
make distclean
make -j $(nproc --all) DEPS_CMAKE_FLAGS="-DUSE_BUNDLED_LIBUV=OFF -DUSE_BUNDLED_MSGPACK=OFF -DUSE_BUNDLED_LUAROCKS=OFF -DUSE_BUNDLED_LUAJIT=OFF"
```

Which results in:
```
make[4]: Leaving directory '/root/neovim/.deps/build/src/unibilium'
[ 77%] Completed 'unibilium'
make[3]: Leaving directory '/root/neovim/.deps'
[ 77%] Built target unibilium
/usr/bin/install -c -m 755 lib/libjemalloc.a /root/neovim/.deps/usr/lib
/usr/bin/install -c -m 755 lib/libjemalloc_pic.a /root/neovim/.deps/usr/lib
make[4]: Leaving directory '/root/neovim/.deps/build/src/jemalloc'
[ 79%] Completed 'jemalloc'
make[3]: Leaving directory '/root/neovim/.deps'
[ 79%] Built target jemalloc
make[2]: Leaving directory '/root/neovim/.deps'
Makefile:76: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/root/neovim/.deps'
Makefile:75: recipe for target 'deps' failed
make: *** [deps] Error 2
```

joshuamcginnis, 2017-01-09:
Those errors seem to hint at either lua not being installed or the wrong version (which I did the first time around). 

`sudo apt-get install lua5.1 luarocks libuv1-dev`

Are you sure you have lua5.1 and libuv1-dev installed?

Fourdee, 2017-01-10:
@joshuamcginnis 

> Are you sure you have lua5.1 and libuv1-dev installed?

Yes. Followed your instructions exactly: https://github.com/neovim/neovim/issues/5861#issuecomment-270216617

Also tried with ```libuv0.10-dev```, same result.
```
root@DietPi:~/neovim# apt-cache search libuv
libuv0.10 - asynchronous event notification library - runtime library
libuv0.10-dbg - asynchronous event notification library - debugging symbols
libuv0.10-dev - asynchronous event notification library - development files
libuv1 - asynchronous event notification library - runtime library
libuv1-dbg - asynchronous event notification library - debugging symbols
libuv1-dev - asynchronous event notification library - development files
r-cran-httpuv - GNU R package of HTTP and WebSocket Server Library
```

So, either something has changed on NeoVim git, that breaks your original compile notes. Or, possibly we are lacking some additional packages and/or steps?

joshuamcginnis, 2017-01-10:
Are you on slack or freenode? Perhaps I can blow away my installation and we can go through it together.

Fourdee, 2017-01-13:
@joshuamcginnis 

> Are you on slack or freenode? 

Nope, i'd never get anything done lol 😃 

> Perhaps I can blow away my installation and we can go through it together.

Sounds good, may take a bit longer, but we should be able to achieve this here.

Here are my install notes (based on yours), includes all the commands required: https://github.com/Fourdee/DietPi/issues/692#issuecomment-270419916

007lva, 2017-01-15:
I'm testing Dietpi Stretch, and I just noticed that neovim package is already include in the repos:

https://packages.debian.org/stretch/neovim

joshuamcginnis, 2017-01-18:
It looks like Debian 9 Stretch will go [full freeze](https://wiki.debian.org/DebianStretch) on 2-Feb-2017. Though, I don't know how much longer after DietPi Stretch will be released. That said, it might be path of least resistance. 

As an aside, if I were to begin testing DietPi Stretch, will I need to reimage once it released or will the normal update process take care of it?

Fourdee, 2017-01-19:
> As an aside, if I were to begin testing DietPi Stretch, will I need to reimage once it released or will the normal update process take care of it?

If i'am honest, unknown at the moment. It really depends how much Stretch changes. We'll patch if we can, however, do expect a new image required for Alpha > Beta? > Release

Fourdee, 2017-01-19:
@joshuamcginnis 

NeoVim, straightforward apt install on Stretch:
https://github.com/Fourdee/DietPi/issues/475#issuecomment-273755851

Fourdee, 2017-01-19:
Although a simple apt-get install, i've added to ```dietpi-software``` aswell, under ```additional linux software```.

Will mark this as closed. This installation is available for Stretch only.

@joshuamcginnis 

If you get any further success with compiling for Jessie, please reopen as needed. However, from my end, I simply don't have the time available to put more into this, and, i'am a nano user 😉 lol.

closed 2017-01-19 by Fourdee