MichaIng/DietPi #694: General | Incorrect fonts with de_DE@euro

Fourdee, 2017-01-04:
As reported by Rainer (email):

> From my point of view something is rotten in the termcap/-info system. Normally every terminal should go through it and get correct font and control commands.

>DietPi | Update available
âââââââââââââââââââââââââââââââââââââââ
 V140 | NanoPi M1/T1 (armv7l)
âââââââââââââââââââââââââââââââââââââââ

k-plan, 2017-01-09:
@Fourdee 

[image]

❓ 

```
root@NanoPi-NEO:~# locale
LANG=de_DE@euro
LANGUAGE=
LC_CTYPE="de_DE@euro"
LC_NUMERIC="de_DE@euro"
LC_TIME="de_DE@euro"
LC_COLLATE="de_DE@euro"
LC_MONETARY="de_DE@euro"
LC_MESSAGES="de_DE@euro"
LC_PAPER="de_DE@euro"
LC_NAME="de_DE@euro"
LC_ADDRESS="de_DE@euro"
LC_TELEPHONE="de_DE@euro"
LC_MEASUREMENT="de_DE@euro"
LC_IDENTIFICATION="de_DE@euro"
LC_ALL=

```

`de_DE@euro ISO-8859-15 `

... a problem with SSH client or on screen (monitor)?
Sorry but my NanoPi Neo (tested device) have no HDMI output.

Fourdee, 2017-01-09:
@k-plan 

Thanks for running those tests. User is using NanoPi M1, so i'll setup a system at some point and see if its device specific.

Fourdee, 2017-01-14:
Fine:
[image]

Most likely as @k-plan suggested, due to SSH client and/or configuration.

Will mark as closed and discuss with user.

closed 2017-01-14 by Fourdee