MichaIng/DietPi #697: DietPi.com | Create post listing current image versions and kernel versions

Fourdee, 2017-01-05:
>

Fourdee, 2017-01-13:
Complete abuse of printscreen and paste lol:
[image]

MichaIng, 2018-01-30:
@Fourdee 
Maybe simplest is to just add the info to image (archive) names. Somewhere it was recently asked to add kernel version there as well: E.g. `DietPi_6.1-RPi_4.9.59-Stretch`
The CPU architecture for my impression can be left out in cases, where the SBCs name clearly define it anyway. For RPi e.g. it is misleading/confusing, as the different versions offer different CPU architectures, but the installed repo system is always armhf, fitting to all RPi versions (besides AlloGUI/AmiBerry).

MichaIng, 2018-02-03:
I think we can close this. DietPi versions do not make too much sense, as anyway it will be updated to latest automatically on 1st run.
Kernel versions might make slightly more sense, but most of them are installed via APT packages and will also in case update on 1st run. Also then we would need to track those APT updates carefully, which is simply too much work for the little helpful info. Also kernels can easily be updated in most cases.

closed 2018-02-03 by MichaIng