MichaIng/DietPi #698: DietPi-Software | Apt-Fast - Downloading packages in parallel, with multiple connections per package.

LexiconCode, 2017-01-06:
[Apt-Fast](https://github.com/ilikenwf/apt-fast) is a shellscript wrapper for apt-get and aptitude that can drastically improve apt download times by downloading packages in parallel, with multiple connections per package.

**Quick Install**
You can quickly install apt-fast by running:
`/bin/bash -c "$(curl -sL https://git.io/vokNn)"`

Then simply run apt-fast instead of apt-get or aptitude.

We could use apt-fast instead of apt-get.

Fourdee, 2017-01-06:
@LexiconCode 

> drastically improve apt download times 

Sounds awesome 👍 I'll take a look and see if we can integrate it.

Fourdee, 2017-01-06:
LAMP stack install (No apt mirrors added):
- ```apt-get``` = ```real    2m14.208s```
- ```apt-fast``` = ```real    2m20.798s```

😕

Fourdee, 2017-01-06:
@LexiconCode 
I'll mark this as closed, but your welcome to test, support for ```apt-fast``` available in testing branch:
https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md

Make sure to install ```apt-fast```:
https://github.com/Fourdee/DietPi/issues/698#issuecomment-270915190

closed 2017-01-06 by Fourdee