MichaIng/DietPi #699: Image | Sparky (Allo)

Fourdee, 2017-01-06:


Fourdee, 2017-01-07:
dmesg
```
[  126.325167] gsl_ts_read set data address fail!
[  126.325192] ------gslX680 test_i2c error------
[  128.325077] owl_i2c: owl_i2c_fifo_irq(): [i2c1] no ACK, fifostat 0x10013

[  128.325170] gsl_ts_read set data address fail!
[  128.325255] owl_i2c: owl_i2c_fifo_irq(): [i2c1] no ACK, fifostat 0x10013

[  128.325327] gsl_ts_read set data address fail!
[  128.325361] ======int_1st: 0 0 0 0 , int_2nd: 0 0 0 0 ======
```

Fourdee, 2017-01-07:
cpu temp:
```
/sys/class/thermal/thermal_zone1/temp
```

Fourdee, 2017-01-07:
```
root@DietPi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: atm7059link [atm7059_link], device 0: ATC2603C PCM atc2603c-dai-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: atm7059link [atm7059_link], device 1: HDMI PCM atm7059-hdmi-dai-1 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: atm7059link [atm7059_link], device 2: SPDIF PCM atm7059-spdif-dai-2 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

Fourdee, 2017-01-07:
WiFi:
```iw reg set GB``` no effect
```
root@DietPi:~# iw reg get
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 40), (6, 20), (N/A), NO-IR
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, NO-IR
        (5170 - 5250 @ 80), (6, 20), (N/A), NO-IR
        (5735 - 5835 @ 80), (6, 20), (N/A), NO-IR
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

Fourdee, 2017-01-07:
##GPU:

- GPU driver: https://github.com/sparkysbc/sparky_sdk/tree/master/ubuntu/gpu/v2

Modules:
- pvrsrvkm
- drm
- videobuf2_vmalloc?
- bc_example?

Fourdee, 2017-01-09:
Looks like we had a corruption on 1st partition of the initial image. Image and partitions has been re-generated and uploaded.

Just waiting for confirmation from Paul resolves inability to boot on his system.

Fourdee, 2017-01-09:
To Do (Many thanks to Jaikumar from Allo for the info):
- We are using Actions semiconductor S500 chip
- owl_i2c msg are related to touchscreen. It can be disabled in kernel config. or blacklist ctp_gsl3680
- Test EMMC: Kernel Image will work for both sdcard & emmc. kernel.dtb is modified to work for both. If the image is built using https://github.com/sparkysbc/Linux.git it can be used for both sdcard & emmc.

Fourdee, 2017-01-13:
- 🈺 Allo Cheapo requires testing
- GPU support, as this SBC is mostly aimed towards Audiophiles, will most likely leave this out for now, unless we get interest from our users to implement it.

Fourdee, 2017-01-18:
Completed. Will test Cheapo when it arrives.

closed 2017-01-18 by Fourdee

joaofl, 2017-03-21:
@Fourdee 
Could you get USB3 to work full speed on this board?

Because this chip is the same as the one on Roseapple Pi. The issue is that at the armbian distro uses the kernel from lemaker, which does not support USB3.

It seems that the kernel by sparky (the one used here) support USB3... At least there are some commits supposed to fix that.
https://github.com/sparkysbc/Linux/commit/3867b3fd05d2998a5503bb6c72a4946b5580a563

However the sparky does not work on the Roseapple Pi out of the box. Ethernet works, but none of the USB ports work, but I still can't understand why.

kotamotov, 2017-10-15:
Hi  Fourdee! 
I have one more question. Dietpi can't recognize my I2s DAC AK4495.
What driver I need to install?
Here is my log file:
[    0.982476] pcm512x 2-004c: Failed to reset device: -6
[    0.983003] owl_i2c: owl_i2c_fifo_irq(): [i2c2] no ACK, fifostat 0x20013

[   11.397985] soc-audio soc-audio.1: ASoC: machine PianoDACPlus should use snd_soc_register_card()
[   11.400993] soc-audio soc-audio.1: ASoC: CODEC pcm512x.2-004c not registered
[   11.404418] platform soc-audio.1: Driver soc-audio requests probe deferral

[   11.611566] soc-audio soc-audio.1: ASoC: machine PianoDACPlus should use snd_soc_register_card()
[   11.611632] soc-audio soc-audio.1: ASoC: CODEC pcm512x.2-004c not registered
[   11.611692] platform soc-audio.1: Driver soc-audio requests probe deferral
[   12.128501] soc-audio: probe of soc-audio.1 failed with error -22

[   11.611566] soc-audio soc-audio.1: ASoC: machine PianoDACPlus should use snd_soc_register_card()
[   11.611632] soc-audio soc-audio.1: ASoC: CODEC pcm512x.2-004c not registered
[   11.611692] platform soc-audio.1: Driver soc-audio requests probe deferral
[   12.128501] soc-audio: probe of soc-audio.1 failed with error -22