MichaIng/DietPi #700: DietPi-Software | NextCloud Webdav is nonfunctional

LexiconCode, 2017-01-07:
I've spent the last three days trying to make NextCloud Webdav work. I really could use some help. Some of these errors aren't related but may be useful.

**Hardware**
Odroid c2 
**Configuration** 
Dietpi 141
A let's encrypt certificate forcing http to https
'always_populate_raw_post_data' to '-1' in php.ini

I removed special characters from MyUser passwords as apparently in my research discovered that Webdav cannot but the rest of NextCloud cam through the main login.

NextCloud log data

- Warning core Login failed: 'admin' (Remote IP: '192.168.1.1')
I'm a little confused why this exists as '192.168.1.1' is my router. Which shouldn't be trying to login anything.

- Sabre\DAV\Exception\Conflict: HTTP/1.1 409 Parent node does not exist
A Webdav error -I've googled this extensively and Evan found a solution.
https://github.com/stackd/SabreDAV/blob/master/lib/Sabre/DAV/Exception/Conflict.php
`    /var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/CorePlugin.php - line 590: Sabre\DAV\Server->createCollection('InstantUpload/2...', Object(Sabre\DAV\MkCol))
    [internal function] Sabre\DAV\CorePlugin->httpMkcol(Object(Sabre\HTTP\Request), Object(Sabre\HTTP\Response))
    /var/www/nextcloud/3rdparty/sabre/event/lib/EventEmitterTrait.php - line 105: call_user_func_array(Array, Array)
    /var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php - line 479: Sabre\Event\EventEmitter->emit('method MKCOL', Array)
    /var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php - line 254: Sabre\DAV\Server->invokeMethod(Object(Sabre\HTTP\Request), Object(Sabre\HTTP\Response))
    /var/www/nextcloud/apps/dav/appinfo/v1/webdav.php - line 60: Sabre\DAV\Server->exec()
    /var/www/nextcloud/remote.php - line 165: require_once('/var/www/nextcl...')
    {main}`

- Error	no app in context	InvalidArgumentException: The given object name is invalid
`    /var/www/nextcloud/apps/activity/lib/FilesHooks.php - line 892: OC\Activity\Event->setObject('files', 48, false)
    /var/www/nextcloud/apps/activity/lib/FilesHooks.php - line 184: OCA\Activity\FilesHooks->addNotificationsForUser('MyUser', 'created_self', Array, 48, false, true, true, 0, 'file_created')
    /var/www/nextcloud/apps/activity/lib/FilesHooks.php - line 111: OCA\Activity\FilesHooks->addNotificationsForFileAction('/', 'file_created', 'created_self', 'created_by')
    /var/www/nextcloud/apps/activity/lib/FilesHooksStatic.php - line 45: OCA\Activity\FilesHooks->fileCreate('/')
    [internal function] OCA\Activity\FilesHooksStatic fileCreate(Array)
    /var/www/nextcloud/lib/private/legacy/hook.php - line 106: call_user_func(Array, Array)
    /var/www/nextcloud/lib/private/Files/View.php - line 1237: OC_Hook emit('OC_Filesystem', 'post_create', Array)
    /var/www/nextcloud/lib/private/Files/View.php - line 1163: OC\Files\View->runHooks(Array, '/MyUser/files', true)
    /var/www/nextcloud/lib/private/Files/View.php - line 266: OC\Files\View->basicOperation('mkdir', '/MyUser/files', Array)
    /var/www/nextcloud/lib/private/Files/Node/Folder.php - line 149: OC\Files\View->mkdir('/MyUser/files')
    /var/www/nextcloud/lib/private/Files/Node/Root.php - line 380: OC\Files\Node\Folder->newFolder('/MyUser/files')
    [internal function] OC\Files\Node\Root->getUserFolder('MyUser')
    /var/www/nextcloud/lib/private/Files/Node/LazyRoot.php - line 65: call_user_func_array(Array, Array)
    /var/www/nextcloud/lib/private/Files/Node/LazyRoot.php - line 282: OC\Files\Node\LazyRoot->__call('getUserFolder', Array)
    /var/www/nextcloud/lib/private/Server.php - line 948: OC\Files\Node\LazyRoot->getUserFolder('MyUser')
    /var/www/nextcloud/lib/private/User/Session.php - line 411: OC\Server->getUserFolder('MyUser')
    /var/www/nextcloud/lib/private/User/Session.php - line 479: OC\User\Session->prepareUserLogin(true)
    /var/www/nextcloud/lib/private/User/Session.php - line 299: OC\User\Session->loginWithPassword(*** sensitive parameters replaced ***)
    /var/www/nextcloud/core/Controller/LoginController.php - line 244: OC\User\Session->login(*** sensitive parameters replaced ***)
    [internal function] OC\Core\Controller\LoginController->tryLogin(*** sensitive parameters replaced ***)
    /var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php - line 160: call_user_func_array(Array, Array)
    /var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php - line 90: OC\AppFramework\Http\Dispatcher->executeController(Object(OC\Core\Controller\LoginController), 'tryLogin')
    /var/www/nextcloud/lib/private/AppFramework/App.php - line 114: OC\AppFramework\Http\Dispatcher->dispatch(Object(OC\Core\Controller\LoginController), 'tryLogin')
    /var/www/nextcloud/lib/private/AppFramework/Routing/RouteActionHandler.php - line 47: OC\AppFramework\App main('LoginController', 'tryLogin', Object(OC\AppFramework\DependencyInjection\DIContainer), Array)
    [internal function] OC\AppFramework\Routing\RouteActionHandler->__invoke(Array)
    /var/www/nextcloud/lib/private/Route/Router.php - line 299: call_user_func(Object(OC\AppFramework\Routing\RouteActionHandler), Array)
    /var/www/nextcloud/lib/base.php - line 1010: OC\Route\Router->match('/login')
    /var/www/nextcloud/index.php - line 40: OC handleRequest()
    {main}`

Fourdee, 2017-01-08:
@LexiconCode 

Thanks for the report, i'll take a look 👍

LexiconCode, 2017-01-19:
```
    location = /.well-known/carddav {
      return 301 $scheme://$host/remote.php/dav;
    }
    location = /.well-known/caldav {
      return 301 $scheme://$host/remote.php/dav;
    }
```

Fourdee, 2017-01-20:
@LexiconCode 

Thanks, this for Lighttpd?

LexiconCode, 2017-01-20:
@Fourdee
It's for Nginx. Only because there is no official documentation for Lighttpd. Lighttpd is technically not supported by NextCloud. I'm hoping Dietpi Certbot supports Nginx soon. It would make it a lot easier to manage NextCloud and other applications that have documentation for Nginx.

LexiconCode, 2017-02-01:
Defining Well-Known Uniform Resource Identifiers (URIs)
https://tools.ietf.org/html/rfc5785#section-3

LexiconCode, 2017-02-10:
When using certbot special consideration is needed for well-known URLs. 
```
Nginx Note:
To use the webroot plugin, your server must be configured to serve files from hidden directories. If /.well-known is treated specially by your webserver configuration, you might need to modify the configuration to ensure that files inside /.well-known/acme-challenge are served by the webserver.

To obtain a cert using a built-in “standalone” webserver (you may need to temporarily stop your existing webserver, if any) for example.com and www.example.com: 
```
https://certbot.eff.org/#debianother-nginx

MichaIng, 2017-12-27:
@LexiconCode @Fourdee 
Is this still present on current ownCloud/Nextcloud implementations?

I never faced an issue on my RPi2 server with DietPi Nextcloud on LAAP webstack + certbot https.

I just tested VM image with LLAP+Nextcloud and webdav works fine. I thought the .well-known folders are just necessary, if you configure webdav clients to access just domain.org instead of the complete URL domain.org/nextcloud/remote.php/dav/... given inside web ui (files+calendar+contacts apps). Never needed this, whether for any cal/card/webdav access nor for certbot. For latter if I get it right it is also just necessary if the base domain, where the certificate applies, is directly pointing towards nextcloud (sub)dir, instead of nextcloud accessible on domain.org/nextcloud (reasonable default on DietPi) and certificate is needed for webserver root on domain.org (var/www).

MichaIng, 2018-01-18:
Marking as closed, as issue is outdated on several stages:
- Nextcloud installation has been completely reworked.
- Certbot installation now includes nginx auto configuration extension, at least on Stretch, and we could also make this available on Jessie via backports: https://packages.debian.org/de/jessie-backports/python-certbot-nginx

closed 2018-01-18 by MichaIng