MichaIng/DietPi #701: Dietpi folder empty

Dalto1, 2017-01-07:
My pine64 system stoped to work after a reboot. Something like this case : http://fuzon.co.uk/phpbb/viewtopic.php?f=11&t=1135

Dalto1, 2017-01-07:
I tried a backup of the sd card with win32diskimager, i checked the archive with 7zip and im able to access to /DietPi from there, but not on the pine64

Fourdee, 2017-01-08:
@Dalto1 

Thanks for the report 👍 

> I tried a backup of the sd card with win32diskimager,

Is there any chance you could zip this image up and send it to us for investigation?

If yes, send me a email to daniel_haze@hotmail.com and i'll send you a link to dropbox for upload.

Dalto1, 2017-01-08:
The archive is like 15 gb do you want files instead? (I reinstalled a fresh image on my pine)

This was the bug in this order. I was unable to connect to libresonic webinterface, so i restarted the pine via command line, and then no access to the folder. It not seems to be due to the power supply.

Fourdee, 2017-01-09:
@Dalto1 

> The archive is like 15 gb do you want files instead? (I reinstalled a fresh image on my pine)

Yep, thats fine. Zip it up if you can please.

WolfganP, 2017-01-14:
Today after booting my Pi2, I noticed many processes weren't running, and tracing the error source I noticed that /DietPi was empty, also and empty /boot/dietpi and missing /boot/dietpi-ramdisk
Even if I always turn off the Pi with 'sudo shutdown', /boot has a ton of FSCK????.REC files
Not the first time I had to reflash the SD card and start over with dietpi so I wonder if there's anything that fails to commit on shutdown?

Fourdee, 2017-01-16:
@WolfganP 

>  FSCK????.REC files
Not the first time I had to reflash the SD card and start over with dietpi so I wonder if there's anything that fails to commit on shutdown?

Looks like a filesystem recovery has occurred on 1st parition.

DietPi simply copies the RAMdisk files back to disk during shutdown, it won't cause a filesystem corruption.

Which SD/PSU combo are you using for this?

WolfganP, 2017-01-16:
Using a Kingston 4GB Class 4 micro SD card, 2A filtered PSU.

Fourdee, 2017-01-16:
@WolfganP 

Should be fine if the 2A PSU provides a stable 5v?

As a pre-caution, i've recreated the RPi image, and its partitions. If we had a corruption on the image, this will now be resolved.

Please download the image from here: http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z
Run it and let us know if you experience the same issue at any point in the future.

closed 2017-01-25 by Dalto1

Fourdee, 2017-01-25:
Reference:
https://github.com/Fourdee/DietPi/issues/719#issuecomment-275129088