MichaIng/DietPi #716: DietPi-Config | Add support for ES9018K2M

Fourdee, 2017-01-18:
> http://www.audiophonics.fr/en/diy-dac/audiophonics-i-sabre-dac-es9018k2m-raspberry-pi-3-pi-2-a-b-i2s-p-11500.html

Fourdee, 2017-01-27:
note to self:
enable module in ```.config```

Fourdee, 2017-02-12:
No fix. Patch for the known issue with this USB DAC exists in 4.4 RPi kernel by default:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1042&start=10#p5577

Fourdee, 2017-03-29:
Test/build our RPi kernel with new AudioPhonics driver

Fourdee, 2017-04-06:
Looks good, untested currently (lack board):
```
root@DietPi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: DAC [I-Sabre K2M DAC], device 0: I-Sabre K2M DAC i-sabre-codec-dai-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@DietPi:~# lsmod
Module                  Size  Used by
i_sabre_codec           5969  1
regmap_i2c              4078  1 i_sabre_codec
brcmfmac              201639  0
brcmutil                9031  1 brcmfmac
snd_soc_bcm2835_i2s     6994  2
i_sabre_k2m             3331  0
regmap_mmio             3728  1 snd_soc_bcm2835_i2s
cfg80211              500074  1 brcmfmac
snd_soc_core          164324  3 i_sabre_k2m,i_sabre_codec,snd_soc_bcm2835_i2s
snd_pcm_dmaengine       5906  1 snd_soc_core
rfkill                 21283  2 cfg80211
snd_pcm                95601  4 i_sabre_k2m,i_sabre_codec,snd_soc_core,snd_pcm_dmaengine
snd_timer              22556  1 snd_pcm
i2c_bcm2708             5740  0
snd                    68400  3 snd_soc_core,snd_timer,snd_pcm
bcm2835_gpiomem         3759  0
bcm2835_wdt             4133  0
uio_pdrv_genirq         3718  0
uio                    10262  1 uio_pdrv_genirq
fuse                   90447  1
ipv6                  367520  14
```

Fourdee, 2017-04-06:
Will nudge to v149, pending testing.

trs79, 2017-04-06:
I'm curious, are the i_sabre_k2m and i_sabre_codec kernel modules open source? My understanding was there were no  native drivers for this board (due to restrictions by ESS)

Fourdee, 2017-04-06:
@trs79 

As far as i'am aware, the AudioPhonics driver uses generic I2C registers which are not restricted by ESS. However, I'd recommend emailing AudioPhonics to get more info.

trs79, 2017-04-06:
That makes sense. Did AudioPhonics provide you with the drivers directly? I had emailed them a while back and they had mentioned a DietPi solution was in the works but I hadn't heard from them about it since.

Fourdee, 2017-04-08:
@trs79 

> Did AudioPhonics provide you with the drivers directly?

Yep. It may be worth sending another email to AudioPhonics, requesting updated status of their driver.

Fourdee, 2017-07-05:
Marking as closed, outdated info/driver. New ticket for final product: https://github.com/Fourdee/DietPi/issues/1054

closed 2017-07-05 by Fourdee