MichaIng/DietPi #718: DietPi-Software | motionEyeOS

lodgnewt, 2017-01-19:
Please add motionEyeOS https://github.com/ccrisan/motioneyeos 
A Video Surveillance OS For Single-board Computers
Installation https://github.com/ccrisan/motioneye/wiki/Installation

Fourdee, 2017-04-03:
- Online doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=100#p6610

Fourdee, 2017-04-03:
Excellent peice of software, impressive.

Fourdee, 2017-04-03:
Commit:
https://github.com/Fourdee/DietPi/commit/324b6b8dfd64363abc687d3a51d7f7ba58ba0eda

Completed.

closed 2017-04-03 by Fourdee

lodgnewt, 2017-04-03:
Thank you very much!

Invictaz, 2017-04-08:
@Fourdee is it possible to use all sorts of USB camera's?

Fourdee, 2017-04-08:
@Invictaz 

> Fourdee is it possible to use all sorts of USB camera's?

Yep, as far as i'am aware any standard USB camera should (untested) work with Motion Eye.

Sirse, 2017-06-02:
Why i can't install MotionEYE on Orange Pi One/PC?

Sirse, 2017-06-02:
It installs on Orange PI, web interface works, but motion don't work:
```
motion -h
motion: error while loading shared libraries: libmmal_core.so: cannot open shared object file: No such file or directory
```
I don't can find, how to install required libs =(

anselal, 2019-06-27:
There is no picture after refreshing the web interface, although the camera is still active according to the camera's led.
Is this a motioneye bug or a dietpi bug ?

It worked great on motioneyeos

MichaIng, 2019-06-27:
@anselal 
Hmm, we install it according to official instructions and do no custom settings changes, so on first view I would say a MotionEye but.

Was this a fresh install or did it suddenly appear?
Which device you are running this?

Update it: `pip install -U motioneye`
Check service logs state: `systemctl -u motioneye`

anselal, 2019-06-27:
It is a fresh install running on my RPi2. I use the 5MP camera module

githlp, 2021-03-16:
in case you activate this

`local_time_file /etc/localtime`

in /etc/motioneye/motioneye.conf

then the motioneye.service on DietPi is broken:

`sudo journalctl -u motioneye`
...
DietPi meyectl[507]:  ImportError: no module named pytz

it cannot load, although i installed pytz as user with pip/pip2. In python shell the pytz module can be loaded though..

MichaIng, 2021-03-16:
Assure that the module is installed globally by running the pip install as root: `sudo pip2 install -U pytz`