MichaIng/DietPi #719: DietPi-RAMdisk | /DietPi folder empty after reboot

Fourdee, 2017-01-19:
Reports:
> http://dietpi.com/phpbb/viewtopic.php?f=11&t=1148&start=20#p5145
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1148&start=20#p5147

Fourdee, 2017-01-23:
@midwan

Which PSU + SD are you using on the Pine when it caused this: https://github.com/Fourdee/DietPi/issues/726#issue-202412586

Fourdee, 2017-01-23:
> John reports possible flaky SD card (Kinsgton 8 GD SD card).

midwan, 2017-01-23:
@Fourdee 
I used the Official Raspberry Pi PSU (5V, 2.5A) and a Sandisk Extreme 16GB SD card.

Fourdee, 2017-01-23:
@midwan 

Thanks. 

When the folder is missing, can you please do the following:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1148&start=20#p5143

Mostly interested in:
```
systemctl status dietpi-ramlog -l
```

WolfganP, 2017-01-23:
I think the issue is even an step earlier, /boot/dietpi is empty as well so dietpi-ramlog and other scripts aren't present in the system.

Fourdee, 2017-01-23:
> I think the issue is even an step earlier, /boot/dietpi is empty as well so dietpi-ramlog and other scripts aren't present in the system.

Seems to vary:
- Boot exists : http://dietpi.com/phpbb/viewtopic.php?f=11&t=1148&start=20#p5147
- Boot exists (script corrupted) : http://dietpi.com/phpbb/viewtopic.php?f=11&t=1148&start=20#p5145
- Boot does not exist (FSCK recovery): http://dietpi.com/phpbb/viewtopic.php?f=11&t=1148#p4868

midwan, 2017-01-25:
@Fourdee 
Here are the results, as requested:
- Device used: Pine64, 2GB model
- Software installed: MATE desktop, Plex Media Server, Samba server. Though they worked fine until I did the firmware update.
- Last command run: Firmware update. It did the self-update first (was already on the latest version) then downloaded the firmware without errors, proceeded to extract it, then reboot.
- PSU used: Official RPI (5V, 2.5A)
- SD card used: SanDisk Ultra 16GB
- fstab entries:

```
root@DietPi:~# cat /etc/fstab
#Internal Drives---------------------------------------------------
proc            /proc           proc    defaults                                                                                        0 0
/dev/mmcblk0p1  /boot           auto    defaults,noatime,discard                                                        0 2
/dev/mmcblk0p2  /               auto    defaults,noatime,discard                                                        0 1
tmpfs                   /tmp                    tmpfs   defaults,noatime,nodev,nosuid,mode=1777                         0 0
tmpfs                   /var/log                tmpfs   defaults,size=20m,noatime,nodev,nosuid,mode=1777        0 0
tmpfs                   /DietPi                 tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777        0 0

#External Drives---------------------------------------------------
#NB: Please use dietpi-drive_manager to setup and control your external drives.
#/dev/sda1       /mnt/usb_1      auto     defaults,noatime,nofail,x-systemd.automount  0 0
#/dev/sdb1       /mnt/usb_2      auto     defaults,noatime,nofail,x-systemd.automount  0 0
#/dev/sdc1       /mnt/usb_3      auto     defaults,noatime,nofail,x-systemd.automount  0 0
#/dev/sdd1       /mnt/usb_4      auto     defaults,noatime,nofail,x-systemd.automount  0 0
#/dev/sde1       /mnt/usb_5      auto     defaults,noatime,nofail,x-systemd.automount  0 0

#Samba Client------------------------------------------------------
#/mnt/samba . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

#FTP Client Mount--------------------------------------------------
#/mnt/ftp_client . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

#NFS Client Mount--------------------------------------------------
#/mnt/nfs_client . Please use dietpi-config and the Networking Options: NAS menu to setup this mount
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

```

DietPi-RAMlog status:

```
root@DietPi:~# systemctl status dietpi-ramlog -l
● dietpi-ramlog.service - DietPi-RAMlog
   Loaded: loaded (/etc/systemd/system/dietpi-ramlog.service; enabled)
   Active: failed (Result: exit-code) since Sun 2017-01-22 18:05:43 GMT; 6min ago
  Process: 543 ExecStart=/bin/bash -c /boot/dietpi/dietpi-ramlog 0 (code=exited, status=127)

Jan 22 18:05:42 DietPi bash[543]: /bin/bash: /boot/dietpi/dietpi-ramlog: No such file or directory
Jan 22 18:05:43 DietPi systemd[1]: dietpi-ramlog.service: control process exited, code=exited status=127
Jan 22 18:05:43 DietPi systemd[1]: Failed to start DietPi-RAMlog.
Jan 22 18:05:43 DietPi systemd[1]: Unit dietpi-ramlog.service entered failed state.

```

I hope that helps!

Fourdee, 2017-01-25:
@midwan 

Thanks 👍 

Looks like some of the DietPi scripts are missing from ```/boot```:
```
root@DietPi:~# ls -al /boot/dietpi

/bin/bash: /boot/dietpi/dietpi-ramlog: No such file or directory
```

I'll run some tests.

Fourdee, 2017-01-25:
@midwan 

Whats the size used in ```/boot```?

```df -h```

Fourdee, 2017-01-25:
Ok i've enabled verbose debugging for ```dietpi-ramdisk```. 

Logs will be saved to:
```/etc/dietpi/logs/dietpi-ramdisk.log```

This won't be enabled until v143 is released, however, once the issue occurs, logs should give us much needed clues.

Fourdee, 2017-01-26:
Just wondering if this is a simple case of service running before ```local-fs.target```

Adding following should prevent that:
```After=local-fs.target```

Fourdee, 2017-01-29:
### Debugging enabled for v143:

If you experience and issues with missing ```/DietPi``` or ```/boot``` files, please paste results of the following command
```
cat /etc/dietpi/logs/dietpi-ramdisk.log
```

Fourdee, 2017-01-30:
Report 1:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1148&start=20#p5346

WolfganP, 2017-01-30:
@Fourdee , just to avoid reinstalling from scratch, is there any workaround (ie wget github.com/xyz... ) to recover the /dietpi folder and get the system back to full functionality? (and then dietpi-upgrade for this changes be incorporated in future reboots)
Thx

Fourdee, 2017-01-30:
@WolfganP 

> recover the /dietpi folder and get the system back to full functionality?

You could recover the scripts from git, however, unfortunately, all the DietPi settings cannot be recovered. These are created and unique to your system installation and setup.

```
root@DietPi:~# ls -lha /DietPi/dietpi/.
./                       .dietpi-survey           .network
../                      .dietpi-sync_exclude     .patch_emr
.dietpi-autostart_index  .dietpi-sync_settings    .uninstalled
.dietpi-backup_settings  .hw_model                .update_available
.dietpi-cloudshell       .installed               .update_stage
.dietpi-process_tool     .install_stage           .version
```

Following contains ```dietpi-software``` software states and settings, without this file, DietPi assumes nothing is installed and system is essentially a fresh installation.
```
 .installed
```

Following contains ```dietpi-config``` settings:
```
/DietPi/dietpi.txt
```

And not to forget the ```config.txt``` for RPi is also in ```/DietPi```

So yes, you could recover DietPi scripts, however, not recommended due to loss of generated settings files.

joshuamcginnis, 2017-02-01:
I just experienced this exact issue on my Odroid C2 with a fresh install on an emmc.

```
systemctl status dietpi-ramlog -l
● dietpi-ramlog.service - DietPi-RAMlog
   Loaded: loaded (/etc/systemd/system/dietpi-ramlog.service; enabled)
   Active: failed (Result: exit-code) since Tue 2017-01-31 17:32:10 PST; 2min 27s ago
  Process: 511 ExecStart=/bin/bash -c /boot/dietpi/dietpi-ramlog 0 (code=exited, status=2)

Jan 31 17:32:10 andromeda bash[511]: /boot/dietpi/dietpi-ramlog: line 1: syntax error near unexpected token `s", sendername="%'
Jan 31 17:32:10 andromeda bash[511]: /boot/dietpi/dietpi-ramlog: line 1: `(sender)s", sendername="%(sendername)s"]'
Jan 31 17:32:10 andromeda systemd[1]: dietpi-ramlog.service: control process exited, code=exited status=2
Jan 31 17:32:10 andromeda systemd[1]: Failed to start DietPi-RAMlog.
Jan 31 17:32:10 andromeda systemd[1]: Unit dietpi-ramlog.service entered failed state.
```

Let me know If there's any other diagnostic information I provide.

Fourdee, 2017-02-01:
Report:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1135&p=5391#p5391

Fourdee, 2017-02-01:
@joshuamcginnis 

> Let me know If there's any other diagnostic information I provide.

Thanks 👍 

Could you please provide the results of:
```
cat /etc/dietpi/logs/dietpi-ramdisk.log
```

Fourdee, 2017-02-01:
Boards reported with this issue occurring:
- OPi Zero
- RPi 2/3
- OPi PC Plus

Fourdee, 2017-02-01:
@midwan  @WolfganP 
Hi guys,

Can you remember how you wrote the image to SD card, win32diskimager.exe ?

midwan, 2017-02-01:
@Fourdee 
I usually do it with Win32DiskImager, but lately I've been testing Etcher also.

joshuamcginnis, 2017-02-01:
@Fourdee 

```
cat /etc/dietpi/logs/dietpi-ramdisk.log
DietPi-Ramdisk: Already running
DietPi-Ramdisk: Stopping
DietPi-Ramdisk: Completed
DietPi-Ramdisk: Starting
DietPi-Ramdisk: Completed
DietPi-Ramdisk: Stopping
DietPi-Ramdisk: Completed
DietPi-Ramdisk: Starting
DietPi-Ramdisk: Completed
DietPi-Ramdisk: Starting
DietPi-Ramdisk: Completed
DietPi-Ramdisk: Starting
DietPi-Ramdisk: Completed
/boot/dietpi/dietpi-ramdisk: line 2: /: Is a directory
/boot/dietpi/dietpi-ramdisk: line 3: tmpfs: command not found
/boot/dietpi/dietpi-ramdisk: line 4: tmpfs: command not found
/boot/dietpi/dietpi-ramdisk: line 5: tmpfs: command not found
/boot/dietpi/dietpi-ramdisk: line 75: unexpected EOF while looking for matching `"'
/boot/dietpi/dietpi-ramdisk: line 96: syntax error: unexpected end of file
```
[dietpi-ramlog.txt](https://github.com/Fourdee/DietPi/files/745922/dietpi-ramlog.txt)

Fourdee, 2017-02-01:
@joshuamcginnis 

Thanks, look like a possible script/echo leak in tty. I'll retrace steps in code, see if we can figure this one out.

midwan, 2017-02-01:
@Fourdee 
Not sure if you still need this, but just in case:

```
root@DietPi:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2   15G  2.8G   11G  21% /
devtmpfs        962M     0  962M   0% /dev
tmpfs           995M     0  995M   0% /dev/shm
tmpfs           995M  8.5M  986M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           995M     0  995M   0% /sys/fs/cgroup
tmpfs            10M     0   10M   0% /DietPi
tmpfs            20M  4.0K   20M   1% /var/log
tmpfs           995M     0  995M   0% /tmp
/dev/mmcblk0p1   50M   13M   38M  26% /boot
tmpfs           199M     0  199M   0% /run/user/0

```

WolfganP, 2017-02-01:
@Fourdee I use Rufus for the img writing.

Fourdee, 2017-02-04:
I believe this issue may be related to a few others users are reporting, eg:
- C2 wont boot on EMMC after updating.

I'am going to change the way we finalize images as I believe this may be a cause. New system with new methods. Once the main culprit images have been updated, we'll need some testers:
https://github.com/Fourdee/DietPi/issues/719#issuecomment-276695879

I don't believe PSU/SD is an issue, but it can never be ruled out. Hopefully the above provides a solution.

Fourdee, 2017-02-05:
Ok, C2 image:
Loopback reports FAT16, 7Zip reports FAT32:
[image]

---

When running a check/repair on this partition, image is then corrupted with I/O errors on next loopback and mount.

Fourdee, 2017-02-06:
Lets update our images, recreating the partitions after image read:
https://github.com/Fourdee/DietPi/issues/719#issuecomment-277533451
- 🈺 C2
- 🈺 RPi

WolfganP, 2017-02-07:
@Fourdee I'm sorry I've missed the images' update. I was waiting for a new version to re-write my SD and start config from scratch to test and get my system back to stability. Do you suggest to wait to v144 to test the new images + new commits related to this issue?

Fourdee, 2017-02-08:
@WolfganP 

> Do you suggest to wait to v144 to test the new images + new commits related to this issue?

You can test these images now.

The new images are essentially v143.5 and contain all the current commits. In other words, they are v144 at this time, containing all the updated code.

Fourdee, 2017-02-13:
Ok, informed users on forums, awaiting results of the new images:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1148&p=5617#p5617

WolfganP, 2017-02-13:
I rebuilt my SD card from scratch a few days ago with the 1.43.5 RasPi image, using the system as plex media & file server, no issues observed so far. Thanks a lot @Fourdee !

Fourdee, 2017-02-13:
@WolfganP 

> no issues observed so far

Excellent, but please let us know if this changes.

Fourdee, 2017-03-02:
No further reports using the updated images. All images have now been updated using our new image creation procedure and force ```sync``` in ```dietpi-ramlog```.

Marking as closed. Will reopen if further reports come in, using the latest DietPi v145 images.

closed 2017-03-02 by Fourdee

WolfganP, 2017-04-15:
Just to report that I lost /Dietpi folder again today. Something weird about chown in the logs:

```root@DietPi:~$ systemctl status dietpi-ramlog -l
● dietpi-ramlog.service - DietPi-RAMlog
   Loaded: loaded (/etc/systemd/system/dietpi-ramlog.service; enabled)
   Active: active (exited) since Tue 2017-04-11 22:53:07 UTC; 12min ago
  Process: 399 ExecStart=/bin/bash -c /boot/dietpi/dietpi-ramlog 0 (code=exited, status=0/SUCCESS)

Apr 11 22:53:06 DietPi systemd[1]: Starting DietPi-RAMlog...
Apr 11 22:53:07 DietPi bash[399]: DietPi-Ramlog: Starting
Apr 11 22:53:07 DietPi bash[399]: chown: invalid user: ‘debian-tor:nogroup’
Apr 11 22:53:07 DietPi bash[399]: DietPi-Ramlog: Completed
Apr 11 22:53:07 DietPi systemd[1]: Started DietPi-RAMlog.

Fourdee, 2017-04-16:
@WolfganP 

Thanks for the report,

> chown: invalid user: ‘debian-tor:nogroup’

This is for the RAMlog, we have a special case permission for tor log files, safe to ignore.

Whats the status of RAMdisk ```/DietPi```:
```
systemctl status dietpi-ramdisk -l
cat /etc/dietpi/logs/dietpi-ramdisk.log
```

msdos, 2017-05-27:
I received the same error, but I think the problem is the SD card, it's a **really old** 2gb with only DietPi Jessie installed for the first time. I'm going to stop using the card instead of spending more time with it. I'm posting here just to say that `dmesg` gives me info as well, maybe it's useful for your debugging approaches.

```
[   46.855682] FAT-fs (mmcblk0p1): error, corrupted directory (invalid entries)
[   46.855690] FAT-fs (mmcblk0p1): Filesystem has been set read-only
[   47.178695] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
[   47.178855] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
[   47.698696] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
[   47.698891] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
[   47.699018] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
[   48.326605] Adding 815100k swap on /var/swap.  Priority:-1 extents:5 across:864252k SSFS
[  613.251797] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
[  613.783637] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
```

douglasmiranda, 2017-11-08:
Another common cause for these kind of errors is if you have some volume on /etc/fstab that you are mounting on boot and it fails. Sometimes it fails before the system mounts the first volumes.

So you wait a while and the system will let you inform your root password to enter in maintenance mode, then you can edit your /etc/fstab file, remove / comment the entries that may be the cause of error. Check with `mount -a`, if there are no errors, just reboot.

Fffrank, 2018-05-01:
I'm a new user but unfortunately I believe I just joined this club. Installed DietPi on a fresh Odroid C2 w/ a Sandisk 128gb card. Had things up and running and was tinkering with it. Connected an external USB drive and rebooted -- and directory is now empty. Boots and I can login but can't run any dietpi commands.
`
dietpi@DietPi:~$ sudo ls -lha /var/tmp/dietpi/logs
total 80K
drwxr-xr-x 2 root   root   4.0K Apr 29 13:34 .
drw-rw---- 3 dietpi dietpi 4.0K Jan 10 07:16 ..
-rw-r--r-- 1 root   root    28K Apr 29 13:37 dietpi-firstrun-setup.log
-rw-r--r-- 1 root   root   1.7K Apr 30 10:59 dietpi-ramdisk.log
-rw-r--r-- 1 root   root    940 Apr 30 10:59 dietpi-ramlog.log
-rw-r--r-- 1 root   root      0 Apr 30 11:37 dietpi-software_apt.log
-rw-r--r-- 1 root   root    28K Apr 29 13:35 dietpi-update.log
-rw-r--r-- 1 root   root    213 Dec 31  1969 fs_expand.log
-rw-r--r-- 1 root   root   1.8K Dec 31  1969 fs_partition_resize.log`

Fourdee, 2018-05-01:
@Fffrank 

Thanks for the report. This issue is always related to hardware instability.

Please paste results of the logs:
```
systemctl status dietpi-ramdisk -l
cat /etc/dietpi/logs/dietpi-ramdisk.log
dmesg | grep mmc
```

Also, which power supply are you running?

Fourdee, 2018-05-02:
@Fffrank 

> [    0.766820] emmc: error -110 whilst initialising MMC card
[    6.564644] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

Looks like a corrupt SD filesystem. Any recent power cuts? If not, could be SD card failing, or PSU is insufficient (we highly recommend the official RPi PSU's).

You'll need to reinstall DietPi on the system.

mzramna, 2019-04-17:
this also happends over btrfs filesystem,btrfs does not support swapfile like archives , it should be checked into the instalation script before starting the conversion

MichaIng, 2019-04-17:
@mzramna 
Many thanks for reporting. For reference: https://github.com/MichaIng/DietPi/issues/1285#issuecomment-484207090

All our images are shipped with ext4 root partition, however we do not check root fs type during DietPi-PREP. Either we need to check and disallow DietPi-PREP on non-ext4 root fs, or, which I like more, simply refuse swap creation and print error on script level. This then also includes manual script usage to create swap on external drive etc.
_________
Btw we have an open request to allow swap on BTRFS via separate swap partition: https://github.com/MichaIng/DietPi/issues/1715
But this is nothing we can safely apply to drives with existing content, especially not root fs. And before doing huge efforts, we can simply wait for kernel 5.0 supporting swap files on BTRFS: https://kernelnewbies.org/LinuxChanges/#Linux_5.0.Btrfs_swap_file_support
- But yeah will take a while until that kernel is available for devices, and most (currently supported devices) will most properly never see it 😉.
- Allowing to use a whole partition for swap instead of a swap file is a dedicated reasonable request. So it is then up to the user to create a partition of desired space and the script will just toggle it for swap.