MichaIng/DietPi #721: DietPi-Software | OMV 3 - Unable to create Shared Folders

LexiconCode, 2017-01-20:
Under "Shared Folders" you cannot select a Device since there are no options shown. 

**Hardware**
Odroid C2

**Configuration**
Dietpi 142

**Steps to reproduce**
1. Install OMV 3 
2. attach a USB  hard drive (I've tried both setting up through Dietpi and mounting through is the OS) (formatted as ext4 )
3. log into openmediavault
4. check "Physical Devices" - verify external hard drive and OS medium
5. Go to "Shared Folders" and try to add a folder.
1. Install OMV 3 
2. attach a USB  hard drive (I've tried both setting up through Dietpi and mounting through is the OS) (formatted as ext4 )
3. log into openmediavault
4. check "Physical Devices" - verify external hard drive and OS medium
5. Go to "Shared Folders" and try to add a folder.

Fourdee, 2017-01-22:
@LexiconCode 

Thanks, i'll take a look.

Fourdee, 2017-01-22:
Yep, looks like anything to do with:
"Select a device"

Results in no options.

Only error in log is (doesn't look related):
```
root@DietPi:~# cat /var/log/nginx/openmediavault-webgui_error.log
2017/01/22 13:38:39 [error] 1243#0: *28 open() "/var/www/openmediavault/omv" failed (2: No such file or directory), client: 192.168.0.22, server: openmediavault-webgui, request: "GET /omv HTTP/1.1", host: "192.168.0.112"
```

Fourdee, 2017-01-22:
Ok, looks like OMV scraping of ```/etc/fstab``` is not compatible with DietPi-Drive_manager.

OMV adds headers/blocks which OMV code requires to work with:
```
# >>> [openmediavault]
UUID=45aa4dad-dfd2-492b-917f-11fb0ff3e29b /media/45aa4dad-dfd2-492b-917f-11fb0ff3e29b ext4 defaults,nofail,user_xattr,noexec,usrjquota=aquota.user,grpjquota=aquota.group,jqfmt=vfsv0,acl 0 2
# <<< [openmediavault]
```

- ```dietpi-drive_manager```, unmount drive and eject
- exit
- Plug drive back in
- OMV > Physical Disks > Wipe
- OMV > File Systems > Create
- OMV > File Systems > Mount
OMV > Shared folders > Add

Drive is now available.

k-plan, 2017-01-23:
First, I really like OVM.
But my experience is, you have to run it on performant hardware, like i386 platforms.
On SBC it will have weak performance and stability for a NAS system (hardware limitation).
And OVM is a "all-in-one" system, like a complete OS witch system configuration tool. 
If you need something more, you have to install a OVM plugin, if where is one for it.

 Like I write up here ( http://dietpi.com/phpbb/viewtopic.php?f=12&t=1040&p=4478#p4478 ) I personally think it's not a good idea to use DietPi on the same device with OMV.
And please read @Fourdee https://github.com/Fourdee/DietPi/issues/495#issuecomment-271571863 here.

> Hardware
> Odroid C2

For a Odroid-C2 images you can take a look here: 
https://sourceforge.net/projects/openmediavault/files/Odroid-C2%20images/

Fourdee, 2017-01-23:
@k-plan 

> Like I write up here ( http://dietpi.com/phpbb/viewtopic.php?f=12&t=1040&p=4478#p4478 ) I personally think it's not a good idea to use DietPi on the same device with OMV.

Yep, may need to drop this one from ```dietpi-software```, can see it causing no end of incompatibility issues with DietPi configurations, and in the case here (https://github.com/Fourdee/DietPi/issues/721#issuecomment-274332945) manual entries to ```/etc/fstab``` are not supported by OMV.

k-plan, 2017-01-23:
@Fourdee 

> Yep, may need to drop this one from `dietpi-software` ...

Yes, OVM on DietPi is not more as an early beta. We can test it, but ...

Yes, many incompatibility issues with DietPi configurations and error messages while installing and in the logs.
I have not a huge promise to get this fixed.

LexiconCode, 2017-01-23:
The web UI is clunky and slow. The huge draw for this type of software for me is the ability to configure nearly everything from the web user interface. Hopefully there is an alternative.

k-plan, 2017-01-23:
>  ... is the ability to configure nearly everything from the web user interface. Hopefully there is an alternative.

yes, a user has started to create one, see https://github.com/Fourdee/DietPi/issues/448#issuecomment-242536278

[image]

but he has decided to deleted all his work ... 👎 
https://github.com/Fourdee/DietPi/issues/448#issuecomment-244607413

You can test if Webmin do what you are looking for, but I not a big fan to do this on DietPi.

Fourdee, 2017-01-24:
Ok, added a notice to OMV online doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=4859#p4859

Will mark this as closed.

closed 2017-01-24 by Fourdee

LexiconCode, 2017-01-24:
Thanks for looking into it.  @Fourdee 

@k-plan 
I've considered using Webmin. I've decided against because of outstanding security concerns and I found it to be unreliable with setting configurations. In addition its plug-in system is atrocious and not well-maintained.

Anyway I will continue seeking alternatives but this does bring in question how any software like Webmin would interact with DietPi. 

DietPi has various methods for installing software and then applying its configuration. Basically we need a web admin panel that's robust enough to handle DietPi use cases. If you guys think it's worth it I'll open a separate ticket to carry on this discussion.

Fourdee, 2017-01-25:
@LexiconCode 

> DietPi has various methods for installing software and then applying its configuration. Basically we need a web admin panel that's robust enough to handle DietPi use cases. If you guys think it's worth it I'll open a separate ticket to carry on this discussion.

Already have one open, and yes, a web interface for DietPi would be a great addition.

https://github.com/Fourdee/DietPi/issues/448