MichaIng/DietPi #724: PINE64 | Internal WiFi module only works after connecting Ethernet

midwan, 2017-01-22:
On a pine64 2GB with the Wifi/Bluetooth official module:
On the first installation of DietPi v133, during the network configuration stage, if no ethernet is connected the wifi does not connect either. It seems to enable the module, but Scanning for APs does not work and manual setting does not connect either.

If I connect the ethernet cable, then Wifi also connects normally and we can proceed.

Fourdee, 2017-01-22:
@midwan 

Thanks, i'll take a look.

Fourdee, 2017-01-22:
I believe this is resolved in v140, but wont be applied until ```dietpi-update```.

I've updated the image to v142, please can you re-test and let us know if this resolves it?
http://dietpi.com/downloads/images/DietPi_PineA64-arm64-(Jessie).7z

midwan, 2017-01-22:
Confirmed, it works with this image.

closed 2017-01-23 by Fourdee

ogero, 2017-02-06:
I'm in the same situation, (WiFi dongle only works when ETH cable plugged), but dumb me, I disabled ETH adapter hoping the WiFi adapter would come up as fallback and... nope :(
Is there any way to re-enable ETH by editing some file like the one for initial setup, so I can run the `dietpi-update`? 
Or will need to do a fresh reinstall?  
(This Pi has now zero connectivity and its hdmi port is broken)  

Thanks!

ogero, 2017-02-06:
Don't mind me.. after some unplug-plug cycles, somehow, the WiFi dongle flashes and got a stable connection :)