MichaIng/DietPi #726: PINE64 | Upgrading the firmware breaks DietPi tools

midwan, 2017-01-22:
After using the provided menu option to upgrade the Firmware to the latest version, following the reboot the DietPi shortcuts no longer work:

`# dietpi-config`
`-bash: /DietPi/dietpi/dietpi-config: No such file or directory`

The /DietPi/ folder is actually empty!

Fourdee, 2017-01-23:
@midwan 

Thanks, i'll take a look. 

I'll mark this one as closed (dupe) and use the other ticket: https://github.com/Fourdee/DietPi/issues/719

closed 2017-01-23 by Fourdee

Fourdee, 2017-01-23:
Unable to replicate:
```
root@DietPi:~# uname -a
Linux DietPi 3.10.104-2-pine64-longsleep #113 SMP PREEMPT Thu Dec 15 21:46:07 CET 2016 aarch64 GNU/Linux
```

Must be linked to issue: https://github.com/Fourdee/DietPi/issues/719