MichaIng/DietPi #730: exclude from ver 141 is not working.

filemoon, 2017-01-24:
Looks like /DietPi/dietpi/.dietpi-backup_exclude file content is ignored.

Fourdee, 2017-01-25:
@filemoon 

Is this running on a v141 system? 

Can you give me an example of your excludes?

We had significant changes to ```dietpi-backup``` in v142 and I tested exclude file there successfully. So it may also be worth updating to v142 and re-testing:
https://github.com/Fourdee/DietPi/issues/685

filemoon, 2017-01-25:
I don't know what was wrong - but after upgrade and reboot file was ignored. When I edit (recreate)  file again issue was gone.

To be honest I'm not sure I like new backup - I've used nfs map for <= 141 backups but I cannot use them anymore (not compatible volume - or file system).

Fourdee, 2017-01-25:
@filemoon 

>  but after upgrade and reboot file was ignored. When I edit (recreate) file again issue was gone.

As per the update notification, previous backups from v141 are no longer compatible with v142. You will need to recreate a backup and exclude files again:
https://cloud.githubusercontent.com/assets/7077026/21613602/d03e4600-d1cd-11e6-9482-f000f0dce72c.png

It is possible to restore from a v141 backup following:
https://github.com/Fourdee/DietPi/issues/685#issuecomment-270145794

> <= 141 backups but I cannot use them anymore (not compatible volume - or file system).

NFS is still supported by ```dietpi-backup```, that has not changed, so there may be another issue causing this:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-backup#L320-L327

If you can provide any further info on your NFS setup, i'd be grateful.

Fourdee, 2017-02-05:
No response, making as closed.

closed 2017-02-05 by Fourdee