MichaIng/DietPi #731: Software request: SabNZBd & Movian

Invictaz, 2017-01-25:
Requesting SabNZBd & Movian.tv

Invictaz, 2017-03-22:
Movian has source here:

https://github.com/andoma/movian

Invictaz, 2017-04-07:
Is there a Milestone on this?

Fourdee, 2017-04-07:
@Invictaz 

Not yet unfortunately, v149 milestone is pretty stacked already:
https://github.com/Fourdee/DietPi/milestone/51

I've split SabNZBd into another ticket:
https://github.com/Fourdee/DietPi/issues/865

I may be able to get SabNZBd in for v149 if you like? Movian would need to wait until v150 at the earliest.

Invictaz, 2017-04-07:
@Fourdee Would be great SABNZBd for 149. 

Would hope Movian for 150 then.

Invictaz, 2017-04-08:
@Fourdee 

Please see these additional instructions to build it for Raspberry Pi. Don't know about the other systems (OrangePi etc).

https://movian.tv/boards/23/topics/10345?r=10755

Fourdee, 2017-04-08:
@Invictaz 
Thanks for the info, will come in handy 👍

Invictaz, 2017-04-30:
Any chance this is in v150?

Invictaz, 2017-05-08:
Should be compiler with latest Libav from Git to avoid any MPEG2 errors. 

Might also use, not sure: --enable-omx --enable-omx-rpi --enable-static

Fourdee, 2017-05-13:
https://github.com/andoma/movian#how-to-build-for-raspberry-pi

Fourdee, 2017-05-13:
```
apt-get install -y git-core build-essential autoconf bison flex libelf-dev libtool pkg-config texinfo libncurses5-dev libz-dev python-dev libssl-dev libgmp3-dev ccache zip squashfs-tools

wget https://github.com/andoma/movian/archive/master.zip -O package.zip
unzip package.zip
rm package.zip
cd movian*

./Autobuild.sh -t rpi
```

Default build script:
https://github.com/andoma/movian/blob/552ad613b1d0a2225dce2d0941b453a3dda6f03b/Autobuild/rpi.sh#L50-L57

Invictaz, 2017-06-28:
Autobuild.sh -t rpi calls rpi.sh

When you compile you give -v verionnumber

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

Invictaz, 2017-12-30:
@Fourdee See my other comment in your "I'm going to close all software requests issue". If you can help me please setup a compile / buildrobot I can compile this myself. It needs to be compiled first before we can enter this into DietPi. Because you already compiled a lot of software for DietPi, I hope this is possible.

Fourdee, 2018-01-26:
Re:

> nobody has build a Movian .deb or install for Dietpi yet.
But yesterday I received this new configuration file so it might work.
The author said that it might be not the best but will do.
Can you do anything with it? It is for raspbian but might work for DietPi as well.
Other info is in the issue: https://github.com/Fourdee/DietPi/issues/731
Best,