MichaIng/DietPi #732: BPi Pro | Update to 4.9 kernel

Fourdee, 2017-01-25:


Fourdee, 2017-01-25:
Mainline 4.9: http://dietpi.com/downloads/testing/DietPi_BananaPiPro-armv7-(Jessie).7z

Fourdee, 2017-01-26:
- 🈯️ onboard wifi

gpo123, 2017-01-27:
Sound on 3,5mm jack is working on my BananaPi (not PRO) with self compiled Armbian kernel 4.9. Had to reset alsamixer settings though. Settings:  Left Mixer Left DAC : muted  / Power Amplifier DAC : unmuted / Power Amplifier Mixer :muted / Power Amplifier Mute : unmuted / Right Mixer Left DAC : muted / Right Mixer Right DAC : muted. HDMI ouput seems to be gone, but I don't care for my setup.

I have little problem with configuring ondemand governor:  sample rate setting is not accessible due to /sys changes, CPU stay at max-frequency.

Fourdee, 2017-01-27:
@gpo123 

Thanks for info 👍 

> Sound on 3,5mm jack is working on my BananaPi (not PRO) with self compiled Armbian kernel 4.9. Had to reset alsamixer settings though

Yeah, i did play with the mixer settings, but had no joy on both hdmi and 3.5mm. I'll replicate your amixer settings and re-test.

gpo123, 2017-01-27:
ok, got the ondemand bug: line 1195 in dietpi-config should read: MIN_VALUE=$(( $(cat /sys/devices/system/cpu/cpufreq/ondemand/**min_sampling_rate**) / 1000 )). After fiddling with the settings CPU is now scaling up and down nicely. According to [http://linux-sunxi.org/Cpufreq](url) you should also set **io_is_busy** to **1**. Had good results on my cubietruck.
Do you intend to support all governors available on 4.9 ?

Fourdee, 2017-01-27:
@gpo123 

https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-cpuinfo#L153

We'll need to add an array of possible locations for:
```
/sys/devices/system/cpu/cpufreq/ondemand/min_sampling_rate
/sys/devices/system/cpu/cpu$i/cpufreq/scaling_min_freq
```

> Do you intend to support all governors available on 4.9 ?

All, excluding ones not available on mainline (eg:```interactive```)

> io_is_busy to 1

We'll need to add that in here: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-cpu_set#L99-L123

gpo123, 2017-01-27:
Yes, I'm Jörg. I will try to make the approbiate changes to dietpi-config to support both kernels on BananaPi (PRO) later on.

Fourdee, 2017-01-27:
@gpo123 

Yep, still no joy with BPi Pro. Tried various combinations, no sound:
[image]

gpo123, 2017-01-27:
> image

Hi Dan,

These are my settings with alsamixer -V:

Should I upload my kernel / Armbian deb / kernel config somewhere ?

Regards Jörg

Fourdee, 2017-01-27:
@gpo123 

> alsamixer -V

```
alsamixer -V -c 0
```

Legend 👍. Now it works on 3.5mm!

gpo123, 2017-01-27:
Fine. But how do we deal with HDMI gone? Delete from config or wait for upstream changes ? File a bug report at armbian ?

gpo123, 2017-01-27:
Actually I have a liitle problem with noise on BananaPi with 3,5mm jack on mainline kernel. This was not the case with legacy kernel. Did you observe the same ? I first thought it was a faulty cable or power supply but others seem to have this problem too as you can read here:
[https://forum.armbian.com/index.php/topic/397-mainline-a10a20-audio-driver/](url)
Do you know by chance how to convert *.dtb files into readable format ?

gpo123, 2017-01-27:
I took the chance and updated cubietruck also to mainline 4.9. Running good so far:

[image]

These are the the processes running:
[image]

Quite good for this "deprecated" device ! Must prove itself now as stable and I am contended. Although HDMI is also missing.

Fourdee, 2017-01-28:
> This was not the case with legacy kernel. Did you observe the same ?

Didn't check if i'am honest, I believe in the assumption of basic onboard audio support, then USB DAC/HAT's for serious audio usage. 
Although if it was fine on 3.x, its most likely an issue with the codec driver on 4.9. Seems the solution is per:
https://forum.armbian.com/index.php/topic/397-mainline-a10a20-audio-driver/#entry13933

> Do you know by chance how to convert *.dtb files into readable format ?

From compiled back to source? Nope sorry.

> Fine. But how do we deal with HDMI gone? Delete from config or wait for upstream changes ? File a bug report at armbian ?

https://forum.armbian.com/index.php/topic/397-mainline-a10a20-audio-driver/#entry24185

Fourdee, 2017-01-29:
Tasks:
- 🈴 Set BPi Pro to CARD 0 as default, 3.5mm support only
- 🈴 Apply mixer settings when default card is selected: https://github.com/Fourdee/DietPi/issues/732#issuecomment-275702873

I'll get these done.

Fourdee, 2017-01-29:
Image live.

closed 2017-01-29 by Fourdee