MichaIng/DietPi #733: Problem loading Amiberry V2 onto Raspberry Pi 2

Stormhunter632, 2017-01-25:
Hi. i have a problem loading amiberry on my PI 2. it has a 32gb SD card in it. it has been loading code and rebooted a couple of times, now it is stuck on a page saying /etc/rc.local: line 6: /DietPi/dietpi/func/dietpi-notify: No such file or directory. It is repeating above constantly.Can you advise please.

Fourdee, 2017-01-25:
Possibly linked to: https://github.com/Fourdee/DietPi/issues/719

@Stormhunter632 

Thanks for the report.

Is this a fresh installation today, or an existing installation?

And, is the image you used ```DietPi-AmiBerry_v131_RPi-armv6-(Jessie).img``` or ```DietPi_v141_RPi-armv6-(Jessie).img```?

Stormhunter632, 2017-01-25:
It was today and is V131 which is the one on the retro facebook group

Fourdee, 2017-01-27:
http://dietpi.com/downloads/conf/DietPi-AmiBerry/dietpi.txt

Stormhunter632, 2017-01-27:
can you please tell what i do with the above as i am a novice using raspberry pi. thank you

Fourdee, 2017-01-27:
@Stormhunter632 

Working on it now, just put that there for notes, few minutes.

Stormhunter632, 2017-01-27:
ok thanks had me worried there lol

Fourdee, 2017-01-27:
@Stormhunter632 

Ok, i've updated the RPi image which should prevent loss of DietPi files in ```/boot``` and ```/DietPi```.

Please can you:
- Download the new image: http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z
- Write to SD card
- Right click the following link http://dietpi.com/downloads/conf/DietPi-AmiBerry/dietpi.txt , select "Save as", then save to ```/boot/dietpi.txt``` on the SD card
- Eject SD, plug into RPi and power on.

Let me know if this resolves the issue.

Stormhunter632, 2017-01-27:
ok i will have a go and let you know thanks

Fourdee, 2017-01-27:
@Stormhunter632 

Thanks. Strange, I duplicated the same test, works fine on my RPi 3.

Which PSU brand/amperage and SD card are you using?

Stormhunter632, 2017-01-27:
using rpi2 Samsung scandisk 32gb and 5.1v 2.5 amp psu supplied by the Pi Hut
I can try it on my RPI3 if you want

Stormhunter632, 2017-01-27:
I just tried on my pi 3 with same result. will try with different SD card after Tea but they are both brand new from Pi Hut

Fourdee, 2017-01-27:
@Stormhunter632 

Thanks.

Strange, Sandisk/Samsung should be fine.

Did you download and write the updated image: http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z ?

Stormhunter632, 2017-01-27:
I have just done a total restart from scratch. so loaded image and saved file in /boot/diet pi on new SD card. it ran till it stopped at login details. I did not know you had to login again until I pressed the return button. It then loaded everything and is now sitting on the amiga setup page. I am now going to read the read me document to see how to load kickstarts and roms and see if it works. I will let you know the result.

Fourdee, 2017-01-28:
@Stormhunter632 

Thanks 👍 

> I did not know you had to login again until I pressed the return button.

This should not be the case, automatic login for AmiBerry, i'll look into it.

Fourdee, 2017-01-28:
@Stormhunter632 

I've updated the AmiBerry image also. Simply download, write to SD card and poweron:
http://dietpi.com/downloads/images/DietPi-AmiBerry_RPi-armv6-(Jessie).7z

I've tested here, working fully and automatic login.

If you could test this image on another RPi system, i'd be grateful.

Stormhunter632, 2017-01-29:
ok i will do that.

Stormhunter632, 2017-01-30:
Tried it on RPI 2 with a normal micro SD 32gb card and PSU 5v 2.5a and it came up with Line1120 syntax error near unexpected token '{' 

This might be due to the scan disc not sure.

Now running it on same set up with a new Samsung 32gb SD and it is loaded straight into Amiberry no problems. Also tried it on RPi 3 and fine on there as well.

I just need to now find out how to load Roms and Kickstarts into Amiberry to test it fully.

Fourdee, 2017-01-30:
@Stormhunter632 

> This might be due to the scan disc not sure.

Thanks for running the additional tests 👍 
Yep, sounds like an issue with the SD card used. Possibly faulty.

> I just need to now find out how to load Roms and Kickstarts into Amiberry to test it fully.

You may find this guide useful:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=64#p64

Stormhunter632, 2017-02-01:
is it possible to load midnight commander and PUTTY ssh into Amiberry as all of my things are run on Linux and Raspberry Pi. The only Microsoft i have is on a small laptop so hard to work with and linux is easier for me to uas SSH and transfer files that way.

Fourdee, 2017-02-02:
@Stormhunter632 

OpenSSH server is installed by default on AmiBerry image. You can use WinSCP for file transfers:
https://winscp.net/eng/download.php

MC can be installed optionally, by using ```dietpi-software``` > Linux Software > Midnight commander

Stormhunter632, 2017-02-02:
Thank you for your help. i will have a practice at setting up amiberry and other bits over weekend.

Fourdee, 2017-02-04:
@Stormhunter632 

No worries, let me know if any issues.

Fourdee, 2017-02-13:
@Stormhunter632 
I'll mark this as resolved. Please reopen if required.

closed 2017-02-13 by Fourdee