MichaIng/DietPi #734: dietpi-letsencrypt renew don't overwrite certificate

dmaglio, 2017-01-26:
I have installet dietpi and create a cert with dietpi-letsencrypt. but i notice that isn't renew every month. if i enter into /etc/letsencrypt/live/ i found many version of my certficate. example:
if the site is www.sitedefault.com i found:
* www.sitedefault.com
* www.sitedefault.com-001
* www.sitedefault.com-002
* www.sitedefault.com-003
etc....
in log file cannot find any information about overwrite file....

Fourdee, 2017-01-26:
@dmaglio 

Thanks for report 👍 

Lets force a manual renew to generate logs, please run the following command:
```
/DietPi/dietpi/dietpi-letsencrypt 1
```

Then, please paste logs from:
```
cat /var/log/dietpi-letsencrypt.log
```

---

note to self:
http://dietpi.com/phpbb/viewtopic.php?f=9&t=1228

dmaglio, 2017-07-01:
maybe i found theb bug.... into dietpi-letsencrypt there two variables: LETSENCRYPT_DOMAIN and LETSENCRIPT_EMAIL. after the first install this variable remain to mydomain.com and user@email.com

Kengkohr, 2017-07-27:
The problem seems to be lying on the side of certbot, not on the dietpi-letsencrypt-script.

After running `dietpi-letsencrypt 1` i manually go to 
`/etc/letsencrypt/live/my-domain.de-000X` and do `cat privkey.pem cert.pem > combined.pem`

Therafter I rename `/etc/letsencrypt/live/my-domain.de-000X` to `/etc/letsencrypt/live/my-domain` and everything is working fine and the cert in my server is no longer outdated.

dhernan3, 2017-08-06:
I can cooroborate the bug.
I have manually updated today and a new version of cert bot was downloaded (if I remember correctly, from .17 to .18), and same behavior, the new certs downloaded to my domain.dom-000x. possible fix: download to correct dir if possible that fix, or manually (but with script) the new certs to correct dir

dhernan3, 2017-08-07:
My FIX:

Edit /Dietpi/dietpi/dietpi-letsencrypt

Search the two lines containing

   ./certbot-auto

And then:
 - Delete the option --duplicate (to not create the -000x directory)
- Add  these options:
    --cert-name $LETSENCRYPT_DOMAIN --non-interactive --force-renewal

First option sets the certificate name (and use that name as of directory)
Second option to do not ask questions
Third option to force the renewal of certificate, even if it's not expired

With my modifications, runing
dietpi-letsencript 1
works fine and don't create new dirs.

MichaIng, 2018-01-18:
@Fourdee 
On Stretch we use the package included cron job, but on Jessie we use our own, that has the behaviour mentioned above. We should switch the new (?) renew command: https://certbot.eff.org/docs/using.html#renewing-certificates

Fourdee, 2018-01-18:
@MichaIng 

> On Stretch we use the package included cron job, but on Jessie we use our own, that has the behaviour mentioned above. We should switch the new (?) renew command: 

Yep. Its been a while since I last worked on this script and certbot has changed vastly since we first created it. So my knowledge is outdated/lost lol.

Probably be more beneficial if we re-write this script, based on current certbot documentation. I'd imagine there is a few more outstanding issues that would also need resolving.

MichaIng, 2018-01-18:
@Fourdee 
I would not do a big effort for Jessie anymore, so keep it simple and just use the backports, which allows us to handle distro versions the same?
- https://packages.debian.org/jessie-backports/certbot
- https://packages.debian.org/jessie-backports/python-certbot-apache
- https://packages.debian.org/jessie-backports/python-certbot-nginx

Fourdee, 2018-01-18:
@MichaIng 

> I would not do a big effort for Jessie anymore, so keep it simple and just use the backports, which allows us to handle all distro versions the same?

Sounds good 👍

MichaIng, 2018-01-28:
@Fourdee 
Already working on this, but now realized, that for RPi no backports are available by default. On my RPi I enabled it by installing (manuall) `debian-archive-keyring` and then add Debian backports repo to sources.
```
wget http://deb.debian.org/debian/pool/main/d/debian-archive-keyring/debian-archive-keyring_2017.5~deb8u1_all.deb -O package.deb
dpkg -i package.deb
rm package.deb
echo 'deb http://deb.debian.org/debian jessie-backports main' > /etc/apt/sources.list.d/debian-backports.list
mkdir /etc/apt/preferences.d &> /dev/null
cat << _EOF_ > /etc/apt/preferences.d/certbot
Package: certbot
Pin: release a=jessie-backports
Pin-Priority: 990
_EOF_
G_AGUP
G_AGI -t jessie-backports certbot
```
The pin allows apt upgrade to use backports for certbot and it's dependencies automatically.

On the other hand we do not offer Stretch for RPi anymore, and the Odroids do have backports by default. For full compatibility (also in case of user installs DietPi Jessie via preparation script) going with the method above?

Fourdee, 2018-01-29:
@MichaIng 

> On the other hand we do not offer Stretch for RPi anymore, and the Odroids do have backports by default. For full compatibility (also in case of user installs DietPi Jessie via preparation script) going with the method above?

Agree 👍 

To be consistent with all our other images, may be beneficial if we simply add backports to RPi by default, with Pin set to low priority over Raspbian repos?

MichaIng, 2018-01-29:
Backports have low priority by default, but to assure, we could add:
```
Package: *
Pin: release a=jessie-backports
Pin-Priority: -1
```
I will send PR soon. Upgrading my RPi to Buster at the moment. Of course errors occur, in this case `apt --configure udev` 😱.

k-plan, 2018-01-30:
> Upgrading my RPi to Buster at the moment

As I told you before, you need simply more hardware (SBC)  :smiley: 

 /play rumble

MichaIng, 2018-01-30:
@Fourdee 
Great, just finished this, tested on Jessie and Stretch, working great. But I don't know what drove me to enter a real domain:
> Client with the currently selected authenticator does not support any combination of challenges that will satisfy the CA.

Lead me to this: https://github.com/certbot/certbot/issues/5405#issuecomment-356498627

And I already wondered why my RPi had some acme error recently. Found some partial server outage on their side and a planned reboot and after that, my errors were gone. Actually I guess this is due to temporary re-enabling the TLS-SNI-01 challenge for existing certificates.

To sum it up:
- Due to a found security issue, since 09. January certbot clients with version lower than 0.21.0 will not work any more with apache/nginx plugin.
- Even on Debian Buster the version is 0.20.0, thus too low, so no change to see the new version on Jessie backports nor Stretch, as long as they don't push some upgrade over security repo or something to fix otherwise useless software.
- The workaround provided `certbot -a webroot -w /var/www/ -i nginx/apache ...` seem to do the job, as well as the standalone certificate. For certificate renewal the `renew` seems to remember that method, so should work then also, installer should do all the nginx/apache config file adjustments. We could just use this as our default command in dietpi-letsencrypt. But I can't make a full final test on my VMs, especially to test Apache/Nginx autoconfiguration.
- Alternatively we completely revert to github source, to access the new software version. But I do not like the overhead and auto installed dependencies, that will remain on the system out of APT control.

Okay, I found `--dry-run` to further investigate this on my RPi (production server):

```
2018-01-31 00:15:21 root@micha:~# certbot renew --dry-run
Saving debug log to /var/log/letsencrypt/letsencrypt.log

-------------------------------------------------------------------------------
Processing /etc/letsencrypt/renewal/micha.gnoedi.org.conf
-------------------------------------------------------------------------------
Cert not due for renewal, but simulating renewal for dry run
Plugins selected: Authenticator apache, Installer apache
Renewing an existing certificate
Performing the following challenges:
Client with the currently selected authenticator does not support any combination of challenges that will satisfy the CA.
Attempting to renew cert (micha.gnoedi.org) from /etc/letsencrypt/renewal/micha.gnoedi.org.conf produced an unexpected error: Client with the currently selected authenticator does not support any combination of challenges that will satisfy the CA.. Skipping.
All renewal attempts failed. The following certs could not be renewed:
  /etc/letsencrypt/live/micha.gnoedi.org/fullchain.pem (failure)

-------------------------------------------------------------------------------
** DRY RUN: simulating 'certbot renew' close to cert expiry
**          (The test certificates below have not been saved.)

All renewal attempts failed. The following certs could not be renewed:
  /etc/letsencrypt/live/micha.gnoedi.org/fullchain.pem (failure)
** DRY RUN: simulating 'certbot renew' close to cert expiry
**          (The test certificates above have not been saved.)
-------------------------------------------------------------------------------
1 renew failure(s), 0 parse failure(s)
```
Then I was able to force renewal with the method above to really test it:
```
2018-01-31 00:28:36 root@micha:~# certbot renew --force-renewal -a webroot -w /var/www/ -i apache
Saving debug log to /var/log/letsencrypt/letsencrypt.log

-------------------------------------------------------------------------------
Processing /etc/letsencrypt/renewal/micha.gnoedi.org.conf
-------------------------------------------------------------------------------
Plugins selected: Authenticator webroot, Installer apache
Renewing an existing certificate
Performing the following challenges:
http-01 challenge for micha.gnoedi.org
Using the webroot path /var/www for all unmatched domains.
Waiting for verification...
Cleaning up challenges

-------------------------------------------------------------------------------
new certificate deployed with reload of apache server; fullchain is
/etc/letsencrypt/live/micha.gnoedi.org/fullchain.pem
-------------------------------------------------------------------------------

-------------------------------------------------------------------------------

Congratulations, all renewals succeeded. The following certs have been renewed:
  /etc/letsencrypt/live/micha.gnoedi.org/fullchain.pem (success)
-------------------------------------------------------------------------------
```
[... 10 lines cut ...]
Renewing an existing certificate
Performing the following challenges:
http-01 challenge for micha.gnoedi.org
Waiting for verification...
Cleaning up challenges

-------------------------------------------------------------------------------
new certificate deployed with reload of apache server; fullchain is
/etc/letsencrypt/live/micha.gnoedi.org/fullchain.pem
-------------------------------------------------------------------------------

-------------------------------------------------------------------------------
** DRY RUN: simulating 'certbot renew' close to cert expiry
**          (The test certificates below have not been saved.)

Congratulations, all renewals succeeded. The following certs have been renewed:
[... cut ...]
I guess many certbot/letsencrypt user will not yet found that issue, as it is quite young. So we should do this work around step with our next update. If the certbot packages get some unexpected update, then we can revert the renewal method by: `certbot renew --force-renewal --apache/nginx`

So as this works, and I don't see any problems, as we have webroot access, I would go with this solution. But I am not sure what about the existing Jessie installations. As for 6.0 release date, they should have the fixed version from github. Should we overwrite this by (older) backports package to allow similar binary/cron handling and force the renewal fix there also?

MichaIng, 2018-02-02:
@Fourdee 
Found by change another issue, generally using backports:
#### Required Information:
- DietPi Version | v6.1
- SBC Device     | Virtual Machine (x86_64) (index=20)
- Distro         | jessie (index=3)
- Command        | G_AGI: sqlite3
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
```

The problem here is, that I used backports to install certbot package as above. Somehow the dependency tree includes `libsqlite3-0` (even that no sqlite will be installed) and pulls version 3.16 from backports. If now actual SQLite is installed, it needs the older main repo version 3.8. The only real solution would be, as we can't predict which packages will be pulled from backports by `apt-get install -t jessie-backposts`, to directly upgrade everything from there (set pin priority of whole repo to 990). And I would not recommend that for RPi, where a bunch of packages from different distros (Raspbian and Debian) would be mixed.

Really not sure how to handle this. Backports are generally a nice way to get access to never package versions/new packages and ones, that were first introduced a major version later. We could find a good workaround (check for installed certbot on Jessie in this case, to pull also SQLite from backports) for this single issue, or prevent further issues like that by completely switch to backports, or just leave things, as they are, and keep this in mind, if users face apt dependency issue, to check dpkg for installed backport packages.

Maybe combine first and last. I hope, this is a rare case, were just sqlite library, but not actual sqlite package is needed 😆.

MichaIng, 2018-03-02:
I mark as closed due to: https://github.com/Fourdee/DietPi/pull/1459

closed 2018-03-02 by MichaIng