MichaIng/DietPi #738: NanoPi NEO Air - no bluetooth

crazyquark, 2017-01-29:
Looks like bluetooth is not working. I am on the latest DietPi version.
In dmesg I get these errors:
`
[    7.426466] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Jun  6 2014 14:50:39 version 7.10.226.49 (r) FWID 01-8962686a
[    7.451904] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code (0x30 0x30)

[   14.617302] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[   14.617321] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
`
If I do a hcitool scan I get:
`
Device is not available: Address family not supported by protocol
`

Also the ap6212-bluetooth service is in failed state:
`
 ap6212-bluetooth.service - LSB: Patch firmware for ap6212 adapter
   Loaded: loaded (/etc/init.d/ap6212-bluetooth)
   Active: failed (Result: exit-code) since Sun 2017-01-29 07:24:22 GMT; 5min ago
  Process: 607 ExecStart=/etc/init.d/ap6212-bluetooth start (code=exited, status=1/FAILURE)

Jan 29 07:24:21 DietPi-Nano ap6212-bluetooth[607]: Can't open HCI socket.: Address family not supported by protocol
Jan 29 07:24:21 DietPi-Nano ap6212-bluetooth[607]: /etc/init.d/ap6212-bluetooth: 54: /etc/init.d/ap6212-bluetooth: rfkill: not found
Jan 29 07:24:22 DietPi-Nano ap6212-bluetooth[607]: sh: echo: I/O error
Jan 29 07:24:22 DietPi-Nano ap6212-bluetooth[607]: Can't get port settings: Input/output error
Jan 29 07:24:22 DietPi-Nano ap6212-bluetooth[607]: Can't initialize device: Input/output error
Jan 29 07:24:22 DietPi-Nano ap6212-bluetooth[607]: Can't open HCI socket.: Address family not supported by protocol
Jan 29 07:24:22 DietPi-Nano systemd[1]: ap6212-bluetooth.service: control process exited, code=exited status=1
Jan 29 07:24:22 DietPi-Nano systemd[1]: Failed to start LSB: Patch firmware for ap6212 adapter.
Jan 29 07:24:22 DietPi-Nano systemd[1]: Unit ap6212-bluetooth.service entered failed state.
`

crazyquark, 2017-01-29:
After installing ```rfkill``` which seems to have been missing I still get the following errors:
```

● ap6212-bluetooth.service - LSB: Patch firmware for ap6212 adapter
   Loaded: loaded (/etc/init.d/ap6212-bluetooth)
   Active: failed (Result: exit-code) since Sun 2017-01-29 07:32:41 GMT; 5s ago
  Process: 1994 ExecStart=/etc/init.d/ap6212-bluetooth start (code=exited, status=1/FAILURE)

Jan 29 07:32:40 DietPi-Nano ap6212-bluetooth[1994]: Can't open HCI socket.: Address family not supported by protocol
Jan 29 07:32:41 DietPi-Nano ap6212-bluetooth[1994]: sh: echo: I/O error
Jan 29 07:32:41 DietPi-Nano ap6212-bluetooth[1994]: Can't get port settings: Input/output error
Jan 29 07:32:41 DietPi-Nano ap6212-bluetooth[1994]: Can't initialize device: Input/output error
Jan 29 07:32:41 DietPi-Nano ap6212-bluetooth[1994]: Can't open HCI socket.: Address family not supported by protocol
Jan 29 07:32:41 DietPi-Nano systemd[1]: ap6212-bluetooth.service: control process exited, code=exited status=1
Jan 29 07:32:41 DietPi-Nano systemd[1]: Failed to start LSB: Patch firmware for ap6212 adapter.
Jan 29 07:32:41 DietPi-Nano systemd[1]: Unit ap6212-bluetooth.service entered failed state.
```

crazyquark, 2017-01-29:
```lsmod``` gives:
```

sun8i_ths               3284  0
cpufreq_dt              3522  0
thermal_sys            42239  2 cpufreq_dt,sun8i_ths
uio_pdrv_genirq         3354  0
uio                     8012  1 uio_pdrv_genirq
brcmfmac              164920  0
brcmutil                6270  1 brcmfmac
cfg80211              191427  1 brcmfmac
rfkill                 10928  2 cfg80211
fuse                   70616  1
```

Fourdee, 2017-01-29:
@crazyquark 

Thanks for report. Which kernel are you running?
```uname -a```

crazyquark, 2017-01-30:
On the DietPi setup, the kernel is:
```
Linux DietPi-Nano 4.9.0-sun8i #3 SMP Fri Dec 30 15:18:06 GMT 2016 armv7l GNU/Linux
```
The stock Ubuntu Core uses:
```
Linux FriendlyARM 3.4.39-h3 #24 SMP PREEMPT Tue Oct 18 13:31:10 CST 2016 armv7l armv7l armv7l GNU/Linux
```

Fourdee, 2017-01-30:
@crazyquark 

Thanks, i'll check the stock Ubuntu core image, and see if the service for bluetooth attach is any different. It may be a issue specific to 4.9, but we'll take a look.

Fourdee, 2017-02-09:
```
root@nanopiair:~# iwlist wlan0 scan
wlan0     Interface doesn't support scanning : Resource temporarily unavailable

[  230.935080] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[  230.935094] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
```

Fourdee, 2017-02-09:
I refuse to deal with ~~ARM~~ThomasKaiserBian. Marking this as no fix.

mmikel, 2017-02-26:
changing ttyS1 to ttyS3 in /etc/default/ap6212 was help me....

ThomasKaiser, 2017-02-26:
@mmikel this works out of the box with an uncrippled Armbian installation: https://github.com/igorpecovnik/lib/blob/50882530628faffe9c97f015569a31d16f5ceb47/scripts/firstrun#L217-L218

Fourdee, 2017-02-26:
@mmikel 

> changing ttyS1 to ttyS3 in /etc/default/ap6212 was help me....

Thanks 👍 

Appears the ARMbian ```firstrun``` script failed to execute this correctly, on last image build. I'll rebuild the image and see if this has been fixed: 
https://github.com/igorpecovnik/lib/issues/603

Fourdee, 2017-03-13:
Our updated image should be functional with BT, i'll run a test:
-

Fourdee, 2017-03-14:
🈯️ Works fine: http://dietpi.com/downloads/images/DietPi_NanoPiNEOAir-armv7-(Jessie).7z
```
root@DietPi:~# bluetoothctl
[NEW] Controller 43:29:B1:55:01:01 DietPi [default]
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# scan on
Discovery started
[CHG] Controller 43:29:B1:55:01:01 Discovering: yes
```

Completed.

closed 2017-03-14 by Fourdee

egorf, 2017-06-29:
Hey guys,
I've just checked the image posted here for the NanoPi Neo Air. When booting, both bluetooth and wi-fi do not work. And the kernel is `3.4.113-sun8i`, rather than 4.9. Did anyone manage to get it working after all?

ThomasKaiser, 2017-06-29:
> Did anyone manage to get it working after all?

Works out of the box with Armbian. If you bought your Air just recently then you have AP6212A instead of AP6212 on the board which needs new firmware files (`apt upgrade` and needed firmware files are there -- speaking of Armbian though).

troyane, 2018-08-22:
@Fourdee 

> The requested URL /downloads/images/DietPi_NanoPiNEOAir-armv7-(Jessie).7z was not found on this server.

Will there be any updates for NanoPi Neo Air?

Thanks.

troyane, 2018-08-23:
@Fourdee thanks for quick response!

Will try to prepare image. Unfortunately I'll be able to put my hands on real NanoPi Neo Air starting from next week. 

Also it look like good idea to support NanoPi Neo Air at least because it is LTS - long term support.
As official site said:
>  We will not make any changes to this model and will provide support as long as possible.

troyane, 2018-09-04:
@Fourdee 
I've tried mentioned distributive NanoPi Neo Air Jessie for NAS and Armbian. On the middle of execution (I suppose) of PREP I got error (the same for both OSes) -- see [nanopineoair_dietpi.txt](https://github.com/Fourdee/DietPi/files/2349534/nanopineoair_dietpi.txt). Sorry for "bad formatting", since it was hard to copy-paste the message.

Can't understand why did it got such problems. Previously I've connected to wifi (via `nmtui-connect`).

I'm ready to test and provide more information.