MichaIng/DietPi #740: dietpi-services | add support for custom services from file

Fourdee, 2017-01-30:
>

Fourdee, 2017-01-30:
include file = ```/DietPi/dietpi/.dietpi-services_include```

Fourdee, 2017-01-30:
Available for v144, can be tested now with:

```
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-services -O /DietPi/dietpi/dietpi-services
```

Then create include file as per: https://github.com/Fourdee/DietPi/issues/740#issuecomment-276068522

k-plan, 2017-01-30:
Tested, works with `rpimonitor-helper` on NanoPi Neo. 👍 

Hope, no need any more to manually edit  `/DietPi/dietpi/dietpi-services` every time I make a `dietpi-update` . 😃

Fourdee, 2017-01-30:
@k-plan 

Excellent, thanks 👍 . Will mark as completed.

k-plan, 2017-01-30:
Nice.

Sorry, but now it's time to make some documentation for `dietpi-services`

Perhaps you could please drop some lines into your forum, like for `dietpi-cron` or `dietpi-process_tool`

http://dietpi.com/phpbb/viewtopic.php?f=8&t=5

```
 root@NanoPi-Neo:~# dietpi-services -help

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: -help
 Please wait...

DietPi-Services
Available commands: start / stop / restart / enable / disable
```

Fourdee, 2017-01-30:
@k-plan 

> Perhaps you could please drop some lines into your forum, like for dietpi-cron or dietpi-process_tool

😉

thechildofroth, 2017-01-31:
Thanks for this, sadly doesn't seem to work for gmediarender at the moment (although just as likely to be user error as a real issue)

studioai, 2017-02-01:
Works fine on XU4.

List of daemons I added

- Tonido
- couchpotato
- gitea

Fourdee, 2017-02-01:
Completed: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=5393#p5393

closed 2017-02-01 by Fourdee