MichaIng/DietPi #742: BananaPi Pro: ondemand geovernor / WiFi driver inconsistency

gpo123, 2017-01-30:
1. ondemand governor: sample rate can not be set:
line 1235 in dietpi-config should read: MIN_VALUE=$(( $(cat /sys/devices/system/cpu/cpufreq/ondemand/**min_sampling_rate**

2. image has a little inconsistency: wifi driver brcmfmac is listed in /etc/modules, while in the meantime it's blacklisted in /etc/modprobe.d/wifi_disable.

Fourdee, 2017-01-30:
@gpo123 

> image has a little inconsistency: wifi driver brcmfmac is listed in /etc/modules, while in the meantime it's blacklisted in /etc/modprobe.d/wifi_disable.

The blacklist overrides whats in ```/etc/modules```. So for each image, we define the modules the board has in ```/etc/modules```, then apply a blacklist to disable the WiFi/BT modules in ```/etc/modprobe.d/wifi_disable```.

Then when the user enables WiFi in ```dietpi-config```, blacklist is removed, modules are loaded straight away, and, modules in ```/etc/modules``` are loaded during next boot.

Fourdee, 2017-02-28:
Marking this as closed, due to revert from DEV (4.9) to STABLE (3.x) for our main images:
https://github.com/Fourdee/DietPi/issues/778#issuecomment-283112473

closed 2017-02-28 by Fourdee