MichaIng/DietPi #745: transmission-daemon eating op my memory

Thijzer, 2017-02-02:
board : odroid C1

Service "transmission-daemon" with default settings was running in hi memory usage.
found 300MB memory usage after minutes running with 2 torrents.

**Possible fix:**

Changing this value fixed it for me.

edit `/etc/transmission-daemon/settings.json`

change the value from to "cache-size-mb" to 4,

```json
{
  "cache-size-mb": 4,
}
```

Thijzer, 2017-02-02:
I also noticed that transmission-remote-cli was not installed, I know it's not needed but it's such a small package with lots of functionality.

Fourdee, 2017-02-02:
@Thijzer 

Hi,

DietPi allocates 1/7th of total RAM for BitTorrent applications. This reduces IO thrashing on flash media, improving the performance of BitTorrent downloads and increasing flash media lifespan.

closed 2017-02-02 by Fourdee

Thijzer, 2017-02-02:
@Fourdee, Thank you for your quick answer.

I thought that would be the reason :) 

But honestly, who in his right mind would run transmission on the SD card?

Fourdee, 2017-02-03:
@Thijzer 

> But honestly, who in his right mind would run transmission on the SD card?

You'd be surprised lol 😃