MichaIng/DietPi #747: Kodi Fails to Start

joshuamcginnis, 2017-02-03:
I just installed Kodi from a fresh image using `dietpi-software`. On reboot, I attempted to start kodi and got the following errors:

```
/usr/local/bin/kodi: 15: /usr/local/bin/kodi: cannot create /root/.config/pulse/client.conf: Directory nonexistent
/usr/local/bin/kodi: 18: /usr/local/bin/kodi: pulseaudio: not found
ERROR: Unable to create GUI. Exiting
Segmentation fault
rm: cannot remove ‘/root/.config/pulse/client.conf’: No such file or directory
```

Fourdee, 2017-02-03:
@joshuamcginnis 

Thanks for the report.

Odroid C2?

Can you please attach Xorg log?:
```
cat /var/log/Xorg.log
```

joshuamcginnis, 2017-02-03:
@Fourdee 

Yup, Odroid C2. Also, `cat: /var/log/Xorg.log: No such file or directory`.

Fourdee, 2017-02-04:
@joshuamcginnis 

> Also, cat: /var/log/Xorg.log: No such file or directory.

Strange, possibly this hasn't been installed.

Could you please send us a bug report?
http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184
It will contain all logs and installed software, should help us find the cause.

joshuamcginnis, 2017-02-04:
@Fourdee All set. Reference CODE: 5bfef0a1-0e7a-468d-b535-c641bb61f088-0

Fourdee, 2017-02-04:
@joshuamcginnis 

Thanks, looks fine.

You are not using the ```dietpi-autostart``` feature to launch kodi, how are you trying to launch Kodi from command line?

Try running the command:
```
startkodi
```

joshuamcginnis, 2017-02-05:
I was attempting to start via `kodi`. `startkodi` seems to work just fine; though, I'm not sure how I would have know that! Anywho, seems like we can close this one if you'd like.

Fourdee, 2017-02-05:
@joshuamcginnis 

No worries, online doc has this info: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p43

I'll mark as closed.

closed 2017-02-05 by Fourdee

joshuamcginnis, 2017-02-05:
Thanks as always.