MichaIng/DietPi #751: DietPi-Drive_Manager | Add support for C2 EMMC + SD combo

Fourdee, 2017-02-03:
>

joshuamcginnis, 2017-02-07:
Looking forward to this!

Fourdee, 2017-02-17:
Notes:
- After plugging in SD, reboot required to be detected in ```blkid```

Fourdee, 2017-02-17:
[image]

joshuamcginnis, 2017-02-17:
@Fourdee For some reason, I'm not able to get my 64gb sd card to mount with this update.

```
lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0  7.3G  0 disk
├─mmcblk0p1 179:1    0  128M  0 part /boot
└─mmcblk0p2 179:2    0  7.2G  0 part /

fdisk -l

Disk /dev/mmcblk0: 7.3 GiB, 7818182656 bytes, 15269888 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xd3630000

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        2048   264191   262144  128M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      264192 15269887 15005696  7.2G 83 Linux
```

I've installed the update and rebooted with the sd card inserted a couple of times.

Fourdee, 2017-02-17:
@joshuamcginnis 

>  For some reason, I'm not able to get my 64gb sd card to mount with this update.

This is still in devolpment at my end and hasn't been synced to testing branch yet. 

If you are able to assist us with testing, i'll get this synced tomorrow and let you know when its ready (with instructions for testing branch)?

joshuamcginnis, 2017-02-17:
@Fourdee I just noticed you moved this `v145`. When I saw 144 shipped, I assumed this was included. I'll stand by for now and try to test if I can find some time.

Fourdee, 2017-02-17:
@joshuamcginnis 

> I just noticed you moved this v145. When I saw 144 shipped, I assumed this was included. I'll stand by for now and try to test if I can find some time.

Ah yes, apologies 👍 

We had to release v144 early, contained critical fixes for most used software and support for new audio DAC's, hence we nudged a few things to v145. Sorry about that, although, this update will make v145 😉

Fourdee, 2017-02-18:
The updated ```dietpi-drive_manager``` code also fixes hang issue with ```stretch```: 
https://github.com/Fourdee/DietPi/issues/475#issuecomment-252504800

Fourdee, 2017-02-18:
- 🈴 SD to USB is disabled and needs rewrite, not compatible with new system.

Fourdee, 2017-02-20:
- Issue with moving rootFS back to ```p2``` as we require clear of partition tables

Fourdee, 2017-02-20:
RootFS move tests:
- 🈯️ C2  ```/dev/mmcblk0p2``` > ```/dev/mmcblk1p1``` > ```/dev/mmcblk0p2```
- RPi  ```/dev/mmcblk0p2``` > ```/dev/sda1``` > ```/dev/mmcblk0p2```

Fourdee, 2017-02-20:
Notes:
- ```/mnt/usb_1``` is hardcoded throughout DietPi scripts. Update to be compatible with new system.

Fourdee, 2017-02-22:
#### Removal of hard coded USB drives throughout DietPi scripts

Following programs now allows for a list of active mount locations, with menu selection:
- DietPi-Backup
- DietPi-Sync
- DietPi-Software > Userdata location

[image]

Fourdee, 2017-02-22:
ToDo:
- Remove step 5 from: http://dietpi.com/phpbb/viewtopic.php?f=8&t=9
- Update all references of ```/mnt/usb_1``` in online docs to ```/mnt/uuid-of-device```

Fourdee, 2017-02-23:
@midwan

The recent changes here will effect the AmiBerry automated image. The prompt to setup an external drive has now been removed. 
As of DietPi v145, users can setup external drives and user data by running ```dietpi-software``` from command line and selecting ```User Data``` option.

Fourdee, 2017-02-23:
Completed.

Fourdee, 2017-02-24:
Lets update ```dietpi-cloudshell`` storage menu, to use the new mount list selection system.

Fourdee, 2017-02-24:
Done.

closed 2017-02-24 by Fourdee