MichaIng/DietPi #752: VNC strange behaviour after update to 143

ad5trett, 2017-02-04:
After updating to 143, whenever i start a VNC session in VNC Viewer on my Mac (tightvnc server on Pi2), any window that is opened starts in the top left corner of the screen and i'm unable to move it around the screen or resize it. Any windows that open after that do the same and are on top of the previously opened window meaning you can't select the one underneath?

I've tried the built in VNC viewer in macOS as well, same problem.

Could it be a window manager problem? Bit of a novice here so appreciate any pointers.

Thanks.

[image]

Fourdee, 2017-02-04:
@ad5trett 

Thanks for the report.

I run tightvnc on my XU4, although different device, that remembers window locations.

I'll try to replicate on a RPi using Windows + RealVNC viewer.

Fourdee, 2018-02-22:
Outdated <v6.0, marking as closed.

closed 2018-02-22 by Fourdee