MichaIng/DietPi #753: General | Updating PiHole overwrites lighttpd.conf

Fourdee, 2017-02-04:
We prevent this in the ```dietpi-software``` installation, however, ```pihole -up``` will overwrite the lighttpd.conf.

Options:
- Inform users to backup lighttpd before updating, then move it back after update
- Find a method in ```pihole``` -up that prevents overwrite of lighttpd.conf

k-plan, 2017-02-05:
> - Inform users to backup lighttpd before updating, then move it back after update

Hmm, does `pihole -up` at the moment by default.

```
~# pihole -v
::: Pi-hole version is v2.12 (Latest version is v2.12.1)
::: Web-Admin version is v2.4 (Latest version is v2.4)

~# pihole -up

~# ls -lah /etc/lighttpd/
insgesamt 24K
drwxr-xr-x  4 root root 4,0K Feb  3 23:19 .
drwxr-xr-x 91 root root 4,0K Feb  3 23:19 ..
drwxr-xr-x  2 root root 4,0K Jan 24 15:06 conf-available
drwxr-xr-x  2 root root 4,0K Jan 24 15:14 conf-enabled
-rw-r--r--  1 root root 2,4K Feb  3 23:19 lighttpd.conf
-rw-r--r--  1 root root 1,1K Jan 24 15:14 lighttpd.conf.orig

```

After this, I do : 

```
:~# mv /etc/lighttpd/lighttpd.conf /etc/lighttpd/lighttpd.conf.bak
:~# cp /etc/lighttpd/lighttpd.conf.orig /etc/lighttpd/lighttpd.conf
:~# dietpi-services restart
```

... and I will be back in business. 
Can this be a option?

Fourdee, 2017-02-05:
@k-plan 

> Can DietPi have his own pihole update script?

Yes, but then the issue would be trying to remind users to use that, rather than ```pihole -up```. In effect, we could simply say do this as you mentioned:
```
cp /etc/lighttpd/lighttpd.conf.orig /etc/lighttpd/lighttpd.conf
```

mmm, wonder if we made empty file ```/etc/lighttpd/lighttpd.conf.orig```, and read only, that might prevent ```mv```.

Fourdee, 2017-02-07:
Another option, if we want to "fix" this from our side. Set ```lighttpd.conf``` to read only, when user installs PiHole.

Fourdee, 2017-02-09:
Added info to online docs, marking as closed:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=174#p174

closed 2017-02-09 by Fourdee

PromoFaux, 2017-02-22:
Just FYI, there was a line added to our `lighttpd.conf` that allows users to add their own customisations in `/etc/lighttpd/external.conf`, so that we would no longer overwrite any custom configs that people had set up. Seems a good compromise for now?

https://github.com/pi-hole/pi-hole/pull/704

(btw. incoming question for you over on our repo :wink:)

Dagur, 2021-05-19:
You can't change the port in external.conf so this is not a good solution for me

Joulinar, 2021-05-19:
Hi,

You have answered on a 4 years old post. What exactly you are looking for?

MichaIng, 2021-05-19:
> You can't change the port in external.conf

Is it so? The `external.conf` was recently moved to be invoked earlier: https://github.com/pi-hole/pi-hole/commit/e5bfafefb97995ce4cdfadfed1e745141ac57c28
But its still done after the port was assigned. Or does Lighttpd complain when the port is assigned two times, so that it cannot be overridden by an included config?

If so, on a DietPi system, install Pi-hole via `dietpi-software`, so you can choose a different webserver as well set a custom port and `pihole -up` won't overwrite the config 😉.

Dagur, 2021-05-19:
lighttpd complains that it's assigned two times. 

> Duplicate config variable in conditional 0 global: server.port

I did a quick google and read that it's not possible to set the port

Joulinar, 2021-05-19:
> I'm not using DietPi

Now I'm confused and did not know what you are looking for.

Joulinar, 2021-05-19:
Switch to DietPi and your issue will be gone as DietPi support Pihole running nginx 😉 

As well DietPi is based on Raspberry OS 😄

MichaIng, 2021-05-19:
Ah, that covers Debian/Raspbian Buster and up and is even available from Stretch backports, if you're running an old Debian image: https://packages.debian.org/stretch-backports/lighttpd
```
apt install -t stretch-backports lighttpd
```
I was wondering already as I though to have tested this not too long ago.