MichaIng/DietPi #755: Dietpi-Sofware - Internet connection check

elRadix, 2017-02-07:
Hi, 

I just filled a bug report 'Reference CODE: 6e243236-c63d-4315-81cd-eb7b2699e8fd-0'

I have issues installing dietpi-pacakges, it checks for internet connections and it refuses to instal any packages.

My internet connection is working fine, i'm able to apt-get update and install other packages through apt.

Sidenote, I'm running dietpi on a raspi3

Thank you for looking into my case.

oshank, 2017-02-08:
I'm getting a similar issue, fresh install, first run says it can't connect. Checked the internet and it works just fine. Stuck in a loop.

elRadix, 2017-02-09:
@Fourdee can you let me know what IP/URL address does dietpti-software run an ipcheck
I don't understand why all the rest works; I have full WAN connection to the internet with my pi3

Fourdee, 2017-02-09:
> can you let me know what IP/URL address does dietpti-software run an ipcheck

For the main connection test, we pull in ```/etc/apt/sources.list```:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L10576

If this is happening after selecting software for install, which software?

elRadix, 2017-02-09:
@Fourdee its happening on all packaged to be installed on dietpti-software,  its doing internet check which  I found annoying. Is there a way to disable it temporarily so that I can still install packages through dietpi?

Running manually apt-get update and update install xxxx works perfectly

>  cat /etc/apt/sources.list
deb ftp://ftp.fu-berlin.de/linux/raspbian/raspbian jessie main contrib non-free rpi

>

elRadix, 2017-02-09:
I think it's a bug, while I have changed the apt mirror through dietpi-config, dietpi-software still is looking/expecting to **ftp.debian.org** instead the current apt-mirror.

I have added ftp.debian.org in the apt.source.list, and now it works - is this hardcoded ?

> root@raspi:/# dietpi-software install 31
 [Info] Initializing database, please wait...
 [Info] Reading database, please wait...

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated install
 Please wait...

 [Ok] Installing Kodi: the media centre for linux
 [Info] Testing connection to http://ftp.debian.org/debian
 [Info] Max duration of 45 seconds, please wait...
 [Ok] Connection test | Completed

> 

**At end I also get the following error, how can this happen?**

> 
WARNING: The following packages cannot be authenticated!
  debianutils dash diffutils e2fslibs e2fsprogs findutils ncurses-bin tar bzip2 libbz2-1.0 libtinfo5 libncursesw5 libncurses5 libreadline6 libcomerr2 liblzma5 libss2
  zlib1g-dev zlib1g libaudit1 libustr-1.0-1 libsemanage1 libnewt0.52 libgpm2 libkeyutils1 libldap-2.4-2 libcloog-isl4 libffi-dev libffi6 libglib2.0-0 libcroco3
  libfontenc1 libice6 libpciaccess0 libsigsegv2 libsm6 libunistring0 libxt6 libxaw7 libxdmcp6 libxcb1 libxcb-dri2-0 libxcb-dri3-0 libxcb-present0 libxcb-render0
  libxcb-shape0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0 libxfixes3 libxrender1 libxcursor1 libxdamage1 libxi6 libxinerama1 libxrandr2 libxtst6 libxv1 libxxf86dga1
  libxxf86vm1 iptables libxtables10 iputils-ping logrotate net-tools whiptail aptitude xz-utils libipc-sharelite-perl linux-libc-dev python-dbus python-gdbm
  smartmontools unattended-upgrades wput x11-xserver-utils

>

Fourdee, 2017-02-13:
@elRadix 

> I think it's a bug, while I have changed the apt mirror through dietpi-config, dietpi-software still is looking/expecting to ftp.debian.org instead the current apt-mirror.

Can you please send me a bug report, i'd like to check your system setup:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184

elRadix, 2017-02-13:
bug report 'Reference CODE: 6e243236-c63d-4315-81cd-eb7b2699e8fd-0'
is this good?

Fourdee, 2017-02-14:
@elRadix 

Thanks 👍 

> WARNING: The following packages cannot be authenticated!

I believe this may be due to the Ubuntu repo you have added:
```
/etc/apt/sources.list.d/sabnzbdplus.list

# Sabnzbd
deb http://ppa.launchpad.net/jcfp/ppa/ubuntu precise main
# Sabnzbd
#deb http://ppa.launchpad.net/jcfp/ppa/ubuntu precise main
```

> ```deb ftp://ftp.fu-berlin.de/linux/raspbian/raspbian jessie main contrib non-free rpi```

Ah yes, this is a bug, fails to scrape the url for connection test. We'll get that fixed.

Fourdee, 2017-02-14:
[image]

elRadix, 2017-02-14:
Thank you!

Any chance to have sabnzdb in the dietpi repo included ?

Fourdee, 2017-02-14:
@elRadix 

Resolved for v144, thanks for reporting this.

In the mean time, to resolve now, you can change the apt mirror in ```dietpi-config``` > Networking :NAS/MISC > Apt.
Select any of the entries with ```http```.

Fourdee, 2017-02-14:
@elRadix 

> Any chance to have sabnzdb in the dietpi repo included ?

We already have a ticket for this, not sure when i'll be able to get round to it, but i'll bump it up the pile:
https://github.com/Fourdee/DietPi/issues/731

Fourdee, 2017-02-15:
Completed.

tohjg, 2017-02-24:
im facing the same issue just few minutes. dietpi-software try to fetch 'http://ftp.debian.org/debian'. i tried to put the link to the browser and return 404. i replaced 'http://ftp.debian.org/debian' with 'http://ftp.debian.org/debian/' in /etc/apt/sources.lst. it works! where can i commit 'source.lst'?

Fourdee, 2017-02-24:
> i tried to put the link to the browser and return 404. i replaced 'http://ftp.debian.org/debian' with 'http://ftp.debian.org/debian/' in /etc/apt/sources.lst. it works! where can i commit 'source.lst'?

Confirmed, they must of changed their webserver settings overnight lol.

Ok, few locations to change:
- dietpi-config | Where users can select apt mirror: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-config#L3971-L3990
- image prep | https://github.com/Fourdee/DietPi/blob/testing/PREP_SYSTEM_FOR_DIETPI.sh#L72-L77
- Patch system | For existing installations, we'll need to patch for ```ftp.debian.org``` and replace with sed https://github.com/Fourdee/DietPi/blob/testing/dietpi/patch_file#L2349

Fourdee, 2017-02-24:
Hmm, seems uk mirror doesn't require the ```/```, only master mirror:
```
http://ftp.uk.debian.org/debian
```

Lemme add it in anyway:
🈯️ Fine
```
http://ftp.uk.debian.org/debian/
```
Yep, so we can change all ftp.*.debian.org references to include the ```/``` at end of url.

Fourdee, 2017-02-24:
Ok, this is currently effecting all non-RPi images. We need to get this fixed and released asap. @tohjg have you made a start on any pull requests/commits? If not, let me know and i'll make a start.

I've posted a quick fix for users on forum:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1373#p5810

Fourdee, 2017-02-24:
Interesting report from another user, reordering lines in ```/etc/apt/sources.list``` also resolves it.

Fourdee, 2017-02-24:
- 🈯️ Check all apt mirrors work with ```/``` addition, not just global:
```
root@DietPi:~# INTERNET_URL_TEST='http://ftp.uk.debian.org/debian/'
root@DietPi:~# /DietPi/dietpi/func/check_connection "$INTERNET_URL_TEST"
 [Info] Testing connection to http://ftp.uk.debian.org/debian/
 [Info] Max duration of 20 seconds, please wait...
 [Ok] Connection test | Completed

root@DietPi:~# INTERNET_URL_TEST='http://ftp.us.debian.org/debian/'
root@DietPi:~# /DietPi/dietpi/func/check_connection "$INTERNET_URL_TEST"
 [Info] Testing connection to http://ftp.us.debian.org/debian/
 [Info] Max duration of 20 seconds, please wait...
 [Ok] Connection test | Completed

root@DietPi:~# INTERNET_URL_TEST='http://ftp.cn.debian.org/debian/'
root@DietPi:~# /DietPi/dietpi/func/check_connection "$INTERNET_URL_TEST"
 [Info] Testing connection to http://ftp.cn.debian.org/debian/
 [Info] Max duration of 20 seconds, please wait...
 [Ok] Connection test | Completed

root@DietPi:~# INTERNET_URL_TEST='http://ftp.au.debian.org/debian/'
root@DietPi:~# /DietPi/dietpi/func/check_connection "$INTERNET_URL_TEST"
 [Info] Testing connection to http://ftp.au.debian.org/debian/
 [Info] Max duration of 20 seconds, please wait...
 [Ok] Connection test | Completed
```

Fourdee, 2017-02-24:
- Fresh install patch test C2

Fourdee, 2017-02-24:
### Information for our users:

To check for a valid connection, DietPi will run a spider connection test on the 1st url entry in ```/etc/apt/sources.list```. 
As of a few days ago, Debian.org has modified their webserver configuration, requiring an additional ```/``` at the end of the URL, else, it returns a 404 (failure).

We are aiming to have this resolved in DietPi v145, however, all our images will need to be updated.

### Resolution and fix
If you have experienced the issue of a failed connection, here are the possible options:

#### Option 1: SSH users, sed replace
Copy and paste into term, then re-launch ```dietpi-software```
```
sed -i "s@debian.org/debian[[:space:]]@debian.org/debian/ @g" /etc/apt/sources.list
```

#### Option 2: DietPi-Config
- Launch ```dietpi-config```
- Select ```Networking Options: NAS/MISC```
- Select ```Apt Mirror```
- Select any apt mirror, excluding the Global option.
- Exit and re-launch ```dietpi-software```

mzeichner, 2017-02-26:
An alternative fix that might help some people is:

 in the file dietpi.txt that runs at first startup change the line that reads:

Apt_Debian_Mirror=http://ftp.debian.org/debian

to

Apt_Debian_Mirror=ftp://ftp.debian.org/debian

Fourdee, 2017-03-02:
All images have been updated with the fix, v145 will patch existing installs.

Completed.

closed 2017-03-02 by Fourdee