MichaIng/DietPi #756: DietPi-Software | Update Amiberry to the latest version

midwan, 2017-02-07:
@Fourdee 
I've just released a new stable version of Amiberry, it would be great if we could update the DietPi repo with it (through dietpi-update).

The current version of Amiberry (v2.1) can now be found here: https://github.com/midwan/amiberry/releases/tag/v2.1

I'm working on getting the SDL2 version ready, which is currently sitting on it's own branch. Once that is ready, I'll notify you for another update as well. :)

A few notes, regarding the keyboard LED functionality that we talked about earlier (https://github.com/midwan/amiberry/issues/3). In order for this to work properly, we'll need to make sure of the following:

- Launch emulator as a normal (non-root) user
- Launch the emulator from the console (not X11)
- The Scroll Lock fix needs to be applied to the system otherwise the Scroll Lock LED will not work

Fourdee, 2017-02-07:
@midwan 

> I've just released a new stable version of Amiberry, it would be great if we could update the DietPi repo with it (through dietpi-update).

Excellent 👍 

Yep, i'll add this to v144, will get round to it ASAP.

Fourdee, 2017-02-08:
- Launch emulator as a normal (non-root) user
- Launch the emulator from the console (not X11)
- 🈯️ Already applied to RPi image | The Scroll Lock fix needs to be applied to the system otherwise the Scroll Lock LED will not work

Fourdee, 2017-02-11:
@midwan 

Can I see your service for fastboot uae4arm under user pi?

Fourdee, 2017-02-12:
@midwan 

> Can I see your service for fastboot uae4arm under user pi?

Never mind, sussed it ```StandardInput=tty```, all lock keys now functional 👍 :
```
cat << _EOF_ > /etc/systemd/system/uae4arm-rpi.service
[Unit]
Description=AmiBerry (uae4ARM) Amiga Emulator
DefaultDependencies=no
#After=dietpi-service.service
#After=After=remote-fs.target

[Service]
Type=simple
User=root
StandardInput=tty
WorkingDirectory=/etc/uae4arm-rpi
ExecStart=/bin/bash -c 'cd /etc/uae4arm-rpi; sudo -u amiberry ./uae4arm-rpi -f conf/autostart.uae'

[Install]
WantedBy=local-fs.target
_EOF_
systemctl daemon-reload
```

Fourdee, 2017-02-14:
@midwan 

Ok, i'am going to release this with the following known issues:
- Lock keys not functional

I'll create another ticket for running under another user, as this is starting to hold up v144's release. I will run further tests and hopefully get that patched in for v145.

Let me know if you disagree with this, and/or, would rather delay the 2.1 update until lock keys are functional (this will then be reverted and nudged to v145)

midwan, 2017-02-14:
@Fourdee 
Sorry, I've been a little busy and I couldn't help much these days... :(

Are the lock keys not functional at all, or just their LEDs ?

Fourdee, 2017-02-14:
@midwan 

> Sorry, I've been a little busy and I couldn't help much these days... :(

No worries, know the feeling 👍 

> Are the lock keys not functional at all, or just their LEDs ?

LEDS are not functional, i'll test the other now.

midwan, 2017-02-14:
@Fourdee 
Ok, if the LEDs are not functional but the Caps Lock key does work (that's the only key that we use in the Amiga anyway, AFAIK) then we could release this with a fix later on.

If the Caps Lock does not function properly (no effect in the text) then I'd prefer we delay this release until v145, so we have enough time to fix it.

Fourdee, 2017-02-14:
@midwan 

- 🈯️ LEDs are not functional but the Caps Lock key does work 

Yep, caps lock works, just the LED has no effect, tested in AmigaDos 1.3.

Ok, so will release updated 2.1 with v144, known issue with keyboard LEDs, attempt resolve in v145. You ok with that?

midwan, 2017-02-14:
@Fourdee 
Yes, let's go with that :)

Fourdee, 2017-02-15:
@midwan 

As existing ```.uae``` configs are not compatible with the 2.1 update. We'll skip the patch for automatic updates with existing installations. Wouldn't be nice for us to break users config files that they spent time setting up ;)

Existing installations will need to manually reinstall AmiBerry to receive the update.
- Option 1: ```dietpi-software reinstall 108```
- Option 2: ```dietpi-software``` > uninstall | ```dietpi-software``` > Install

New installations after v144, will always install AmiBerry 2.1.

midwan, 2017-02-15:
@Fourdee 
I wasn't aware of any compatibility issues, could you please let me know what you saw?
The config file uses the same settings as the previous version, besides a few new ones (that will default to being disabled if not found anyway), regarding the mapping of controller buttons to enter the GUI or Shutdown the host system.

Fourdee, 2017-02-15:
@midwan 

Strange, ran a test on current install, upgraded to 2.1, config seems fine. Not sure what happened before (probably user error ;)).

Ok, lets get this upgraded during v144 update.

Fourdee, 2017-02-15:
Ok, all done. Minus the LEDs which i'll take a look at in v145:
https://github.com/Fourdee/DietPi/issues/760

Great work, excellent update @midwan , the best Amiga experience yet 👍 

On a personal note:
I just lost 20 minutes playing Stunt Car Racer lol. So easy to get pulled in 😉

closed 2017-02-15 by Fourdee