MichaIng/DietPi #758: Dietpi-Sofware | Consider dropping support for Lighttpd

LexiconCode, 2017-02-10:
One might consider discontinuing Lighttpd as a DietPi Web server in favor of Nginx. There are a number of reasons that come to mind. 

1. Dietpi development overhead
2. Community support
4. Current integration with Dietpi 
5. Performance - Cost benefit analysis

It takes a lot of effort to to build and maintain Web server configurations for all the software that continues to grows by the ticket. By far Nginx and Apache have support from the community with guides that are readily available. While Lighttpd can be configured to work with most software that effort is put onto Dietpi community. In addition Dietpi Certbot Is not compatible with Lighttpd.

Lighttpd has a marginal edge with performance but it's nearly negligible when compared with Nginx,. Although It uses slightly less RAM and has less lag serving webpages with low static traffic. [1](https://help.dreamhost.com/hc/en-us/articles/215945987-Web-server-performance-comparison)  [2](https://www.rootusers.com/linux-web-server-performance-benchmark-2016-results/)  [3](https://www.jeremymorgan.com/blog/programming/raspberry-pi-web-server-comparison/) 

What are your guys's thoughts especially evaluating the performance of Nginx on single board hardware? One other aspect that should be considered a few Dietpi software packages only support Lighttpd.

kronflux, 2017-02-13:
I agree and disagree.
nginx is certainly more popular, better supported, but dietpi seems to be more along the lines of supporting as many preferences as possible with ease of use in mind.

I can't really agree that keeping it would have any negative impact, and nginx is always an option.

There are certainly things that require lighttpd over nginx - pihole for example uses it by default, and although I have seen some commits recently that seem oriented around giving the user more option in terms of web server, it doesn't seem like they will be changing the default anytime soon, nor do their optional commits appear to be ready for the general public.

MichaIng, 2018-01-30:
@LexiconCode @kronflux 
For now I don't see any reason to drop support. The (current) coding overhead for Lighttpd is quite minimal. If nobody is found to provide a new softwares Lighttpd configuration, then it is easy to force users install Nignx and/or Apache instead in combination. But as for my testing and especially ownCloud/Nextcloud scripting, Lighttpd runs very nice out of the box without any configuration "need". Indeed optimization is another question.

The more I read, I also see Nginx developing faster, making some good recent performance increase. But also Lighttpd is not dead and as long as it is just another "option" we should keep it, supporting this other independent webserver solution approach (not comparable with MySQL <> MariaDB).

closed 2018-01-30 by MichaIng

kronflux, 2018-01-30:
I agree. Although I'd love to see everyone switch over to nginx for security purposes alone(let alone all the other fantastic reasons), lighttpd isn't going anywhere. Too many small applications use it by default, and it's extremely optimized for performance overall(which is fantastic for devices like the Raspberry Pi).
It's very easy to set up for basic use, particularly if you're using it for closed network(LAN only) applications such as smart devices and remote control apps.

Fourdee, 2018-01-30:
> You want to add something? I am just going through some old issues right now, closing what is outdated/answered.

👍 

Lighttpd is here to stay. 
Its the lightest functional http server available. For 1/10 connected users, its ideal with minimal overheads, and in those cases faster than Nginx.

k-plan, 2018-01-30:
@MichaIng 

>You want to add something?  I am just going through some old issues right now, closing what is outdated/answered.

I am no @Fourdee, but yes, I want.

 :100:  Nice! :+1: 

Please keep on doing this,  only so we will get  two-digit issue.

Thanks for all your hard work here - coding, support and housekeeping! :heart: