MichaIng/DietPi #759: RPi | Add support for Edimax EW-7811UAC

Fourdee, 2017-02-12:
> https://community.roonlabs.com/t/dietpi-creating-a-lean-and-mean-roon-bridge/13908/328?u=dan_knight

Fourdee, 2017-02-13:
notes:

```
#Pre-reqs
apt-get install build-essential git -y

git clone --depth=1 https://github.com/Fourdee/linux.git
mv linux /lib/modules/$(uname -r)/build

#Driver
wget https://github.com/diederikdehaas/rtl8812AU/archive/driver-4.3.20.zip -O package.zip
unzip package.zip
rm package.zip

cd rtl8812*

#Set for ARM
sed -i '/^CONFIG_PLATFORM_ARM_RPI =/c\CONFIG_PLATFORM_ARM_RPI = y' Makefile
sed -i '/^CONFIG_PLATFORM_I386_PC =/c\CONFIG_PLATFORM_I386_PC = n' Makefile

make -j $(nproc --all)

#Install
insmod 8812au.ko
cp 8812au.ko /lib/modules/$(uname -r)/kernel/drivers/net/wireless
depmod

Fourdee, 2017-03-17:
User testing:
https://community.roonlabs.com/t/dietpi-creating-a-lean-and-mean-roon-bridge/13908/436?u=dan_knight

Fourdee, 2017-03-19:
User confirmed working with his adapter. Making this as completed. Will create a tutorial on forums:
http://dietpi.com/phpbb/viewtopic.php?f=15&t=1502&p=6223#p6223

closed 2017-03-19 by Fourdee

alfanovember, 2017-10-21:
Hello, I'm having some trouble with this.  On an rpi3, Using a fresh DietPi install from DietPi_RPi-armv6-(Jessie) and dietpi-update'd image,  I then copy/paste the commands into an ssh sessions.  Kernel headers and prereqs install okay,  driver download and unzips.  Sed commands are properly editing the makefile.  The make command fails with what looks like problem with a build target for the current kernel version:
```
root@DietPi:~/rtl8812AU-driver-4.3.20# uname -a
Linux DietPi 4.9.52-v7+ #2 SMP Mon Oct 2 23:27:32 UTC 2017 armv7l GNU/Linux
...
root@DietPi:~/rtl8812AU-driver-4.3.20# make -j $(nproc --all)
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.9.52-v7+/build M=/root/rtl8812AU-driver-4.3.20  modules
make[1]: *** /lib/modules/4.9.52-v7+/build: No such file or directory.  Stop.
Makefile:1680: recipe for target 'modules' failed
make: *** [modules] Error 2
```
Any advice?
Thank you so much for DietPi with the RoonBridge package -  Against all advice I'm using it wireless, and getting a  wlan w/ a better antenna going will be the final step for multi-room music happiness.