MichaIng/DietPi #760: DietPi-Software | AmiBerry: Keyboard LEDS not functional

Fourdee, 2017-02-14:
As per:
https://github.com/Fourdee/DietPi/issues/756#issuecomment-279152829

Further testing/methods required.

Fourdee, 2017-03-22:
Ok, tests:

- 🈴 Logged in as user ```amiberry``` = No caps/scroll/num lock updates

Fourdee, 2017-03-22:
@midwan

Apologies for the delay on this. 

I can not get LED's to work, unless we use ```xinit``` to launch a X display with AmiBerry: https://github.com/Fourdee/DietPi/issues/760#issuecomment-288381563

I've pretty much exhausted my options here and possible tests. There must be something related to SDL under FB that prevents LED functionality?

midwan, 2017-03-22:
@Fourdee 
No worries, I've been quite busy with the SDL2 version in the meantime (trying to get it stable enough for release).

I'll have to refresh my memory on what I saw with this, which means I'll need to setup a fresh environment to test with. From what I remember, it seemed to work fine in stock Raspbian (and Raspbian-based, like RetroPie) distros, which run under the user "pi", but not in DietPi.

I'll get back to you on this after I've done some fresh tests and gathered some data.

Fourdee, 2017-03-23:
@midwan 

> I'll have to refresh my memory on what I saw with this, which means I'll need to setup a fresh environment to test with.

Thanks 👍 

I confirmed both on Raspbian Lite and DietPi, same issues with FB and lack of LED response. Maybe RetroPie is doing something different, or even running it under X, mmm.

>  I've been quite busy with the SDL2 version in the meantime (trying to get it stable enough for release).

Yep i've seen, looking forward to seeing this run under SDL2. Any ETA on when possible release may be ready for testing? 
We could even roll out a pre-release optional install for our DietPi users (eg: AmiBerry V2)?

Fourdee, 2017-04-04:
Reference:
https://github.com/Fourdee/DietPi/issues/850

Fourdee, 2017-04-21:
@midwan 

Confirmed ```xinit``` resolves CAPS/NUM LEDS in SDL1. But this is only useful when running from boot.

i'll also try to tweak the service, to check for X from a running desktop, and, launch non ```xinit``` when needed. I think we did this previously, a while back.

Want me to apply ```xinit```` default for SDL1?

midwan, 2017-04-22:
@Fourdee 
Just to be clear, we are talking about the functionality of using them as activity LEDs (e.g. HDD activity, Floppy, Power, etc), right?

I don't mind if we start it with `xinit `if that makes it work. Most of the focus is on the newer SDL2 version which will eventually replace the older one anyway :)

Fourdee, 2017-04-23:
@midwan 

> Just to be clear, we are talking about the functionality of using them as activity LEDs (e.g. HDD activity, Floppy, Power, etc), right?

Yep, keys are fully functional, however, LED status only works with ```xinit```.

> I don't mind if we start it with xinitif that makes it work. Most of the focus is on the newer SDL2 version which will eventually replace the older one anyway :)

Yep, SDL2 is the way forward (performance is much improved), and progressing fast. I'll make the change to SDL1 to use ```xinit```.

Fourdee, 2017-04-23:
Completed.

Note: LED's will only function when launched from console/autostart, not existing X session.

closed 2017-04-23 by Fourdee