MichaIng/DietPi #761: MariaDB not installing

egrueda, 2017-02-15:
Using a fresh install I try to install lighttpd+mariadb and I get this:

` DietPi-Software
─────────────────────────────────────────────────────
 Mode: Install software
 Please wait...

 MariaDB: database
 [Info] Please wait...

 processing | D i e t  P i

 [Info] Apt-get is processing, please wait...

Preconfiguring packages ...
Selecting previously unselected package libaio1:armhf.
(Reading database ... 19356 files and directories currently installed.)
Preparing to unpack .../libaio1_0.3.110-1_armhf.deb ...
Unpacking libaio1:armhf (0.3.110-1) ...
Selecting previously unselected package mysql-common.
Preparing to unpack .../mysql-common_5.5.54-0+deb8u1_all.deb ...
Unpacking mysql-common (5.5.54-0+deb8u1) ...
Selecting previously unselected package libmysqlclient18:armhf.
Preparing to unpack .../libmysqlclient18_5.5.54-0+deb8u1_armhf.deb ...
Unpacking libmysqlclient18:armhf (5.5.54-0+deb8u1) ...
Selecting previously unselected package mariadb-common.
Preparing to unpack .../mariadb-common_10.0.29-0+deb8u1_all.deb ...
Unpacking mariadb-common (10.0.29-0+deb8u1) ...
Selecting previously unselected package libdbi-perl.
Preparing to unpack .../libdbi-perl_1.631-3+b1_armhf.deb ...
Unpacking libdbi-perl (1.631-3+b1) ...
Selecting previously unselected package libreadline5:armhf.
Preparing to unpack .../libreadline5_5.2+dfsg-2_armhf.deb ...
Unpacking libreadline5:armhf (5.2+dfsg-2) ...
Selecting previously unselected package mariadb-client-core-10.0.
Preparing to unpack .../mariadb-client-core-10.0_10.0.29-0+deb8u1_armhf.deb ...
Unpacking mariadb-client-core-10.0 (10.0.29-0+deb8u1) ...
Selecting previously unselected package mariadb-client-10.0.
Preparing to unpack .../mariadb-client-10.0_10.0.29-0+deb8u1_armhf.deb ...
Unpacking mariadb-client-10.0 (10.0.29-0+deb8u1) ...
Selecting previously unselected package mariadb-server-core-10.0.
Preparing to unpack .../mariadb-server-core-10.0_10.0.29-0+deb8u1_armhf.deb ...
Unpacking mariadb-server-core-10.0 (10.0.29-0+deb8u1) ...
Setting up mysql-common (5.5.54-0+deb8u1) ...
Setting up mariadb-common (10.0.29-0+deb8u1) ...
Selecting previously unselected package mariadb-server-10.0.
(Reading database ... 19681 files and directories currently installed.)
Preparing to unpack .../mariadb-server-10.0_10.0.29-0+deb8u1_armhf.deb ...
Unpacking mariadb-server-10.0 (10.0.29-0+deb8u1) ...
Selecting previously unselected package libdbd-mysql-perl.
Preparing to unpack .../libdbd-mysql-perl_4.028-2+deb8u2_armhf.deb ...
Unpacking libdbd-mysql-perl (4.028-2+deb8u2) ...
Selecting previously unselected package libhtml-template-perl.
Preparing to unpack .../libhtml-template-perl_2.95-1_all.deb ...
Unpacking libhtml-template-perl (2.95-1) ...
Selecting previously unselected package mariadb-client.
Preparing to unpack .../mariadb-client_10.0.29-0+deb8u1_all.deb ...
Unpacking mariadb-client (10.0.29-0+deb8u1) ...
Selecting previously unselected package mariadb-server.
Preparing to unpack .../mariadb-server_10.0.29-0+deb8u1_all.deb ...
Unpacking mariadb-server (10.0.29-0+deb8u1) ...
Processing triggers for systemd (215-17+deb8u6) ...
Setting up libaio1:armhf (0.3.110-1) ...
Setting up libmysqlclient18:armhf (5.5.54-0+deb8u1) ...
Setting up libdbi-perl (1.631-3+b1) ...
Setting up libreadline5:armhf (5.2+dfsg-2) ...
Setting up mariadb-client-core-10.0 (10.0.29-0+deb8u1) ...
Setting up mariadb-client-10.0 (10.0.29-0+deb8u1) ...
Setting up mariadb-server-core-10.0 (10.0.29-0+deb8u1) ...
Setting up mariadb-server-10.0 (10.0.29-0+deb8u1) ...
dpkg: error processing package mariadb-server-10.0 (--configure):
 subprocess installed post-installation script returned error exit status 1
Setting up libdbd-mysql-perl (4.028-2+deb8u2) ...
Setting up libhtml-template-perl (2.95-1) ...
Setting up mariadb-client (10.0.29-0+deb8u1) ...
dpkg: dependency problems prevent configuration of mariadb-server:
 mariadb-server depends on mariadb-server-10.0 (>= 10.0.29-0+deb8u1); however:
  Package mariadb-server-10.0 is not configured yet.

dpkg: error processing package mariadb-server (--configure):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.19-18+deb8u7) ...
Processing triggers for systemd (215-17+deb8u6) ...
Errors were encountered while processing:
 mariadb-server-10.0
 mariadb-server
E: Sub-process /usr/bin/dpkg returned an error code (1)`

and
`Apt-get has failed:
- Apt string = mariadb-server mariadb-client
- Error code = 100
- Software Title = MariaDB: database
`

Fourdee, 2017-02-15:
@egrueda 

Hi, thanks for the report.

Could you please send us a bug report, will allow me to check your system setup:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184

egrueda, 2017-02-15:
Sure!
Reference CODE: fe3353cf-a8dc-44e6-b7c3-23e942d16942-0

Fourdee, 2017-02-18:
Tests:
- 🈯️ Fresh install of ```llap: lighttpd | mariadb | php```
- 🈴 Install of ```llmp: lighttpd | mysql   | php``` on top (asking for trouble lol):
```
Errors were encountered while processing:
 /var/cache/apt/archives/mysql-server-5.5_5.5.54-0+deb8u1_armhf.deb
You might want to run 'apt-get -f install' to correct these.
The following packages have unmet dependencies:
 mysql-server : Depends: mysql-server-5.5 but it is not installed
```

As expected, installing MySQL ontop of MariaDB (and visa versa) will cause issues with package conflicts. You should only have one or the other installed.

Fourdee, 2017-02-18:
Ok, i've added some code that will not allow MariaDB and MySQL to be installed, at the same time. 
- if both are no installed, but selected, MySQL is preferred over MariaDB.

This should prevent any re-occurrence of the issue @egrueda has experienced.

Fourdee, 2017-02-18:
- 🈯️ Tested

@egrueda 
Although this doesnt help your current installation state, this will be prevented from v145 onwards. If you are unable to fix the package conflicts manually, i'd recommend a fresh install, and, only selecting either MySQL or MariaDB.

I'll mark this as closed, please reopen if required.

closed 2017-02-18 by Fourdee