MichaIng/DietPi #762: MySQL not installing

egrueda, 2017-02-15:
Can't install mysql using dietpi-software

```
Preconfiguring packages ...
Selecting previously unselected package libwrap0:armhf.
(Reading database ... 19629 files and directories currently installed.)
Preparing to unpack .../libwrap0_7.6.q-25_armhf.deb ...
Unpacking libwrap0:armhf (7.6.q-25) ...
Selecting previously unselected package libaio1:armhf.
Preparing to unpack .../libaio1_0.3.110-1_armhf.deb ...
Unpacking libaio1:armhf (0.3.110-1) ...
Selecting previously unselected package mysql-server-core-5.5.
Preparing to unpack .../mysql-server-core-5.5_5.5.54-0+deb8u1_armhf.deb ...
Unpacking mysql-server-core-5.5 (5.5.54-0+deb8u1) ...
Preparing to unpack .../mysql-server-5.5_5.5.54-0+deb8u1_armhf.deb ...
Aborting downgrade from (at least) 10.0 to 5.5.
If are sure you want to downgrade to 5.5, remove the file
/var/lib/mysql/debian-*.flag and try installing again.
dpkg: error processing archive /var/cache/apt/archives/mysql-server-5.5_5.5.54-0+deb8u1_armhf.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
Selecting previously unselected package libhtml-template-perl.
Preparing to unpack .../libhtml-template-perl_2.95-1_all.deb ...
Unpacking libhtml-template-perl (2.95-1) ...
Selecting previously unselected package mysql-server.
Preparing to unpack .../mysql-server_5.5.54-0+deb8u1_all.deb ...
Unpacking mysql-server (5.5.54-0+deb8u1) ...
Selecting previously unselected package tcpd.
Preparing to unpack .../tcpd_7.6.q-25_armhf.deb ...
Unpacking tcpd (7.6.q-25) ...
Errors were encountered while processing:
 /var/cache/apt/archives/mysql-server-5.5_5.5.54-0+deb8u1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Fourdee, 2017-02-15:
@egrueda 

> If are sure you want to downgrade to 5.5, remove the file
/var/lib/mysql/debian-*.flag and try installing again.
dpkg: error processing archive /var/cache/apt/archives/mysql-server-5.5_5.5.54-0+deb8u1_armhf.deb (--unpack):

Can you remember:
- What you selected for install, before this occurred?
- Did you run this before trying LLAP (MariaDB afterwards)? https://github.com/Fourdee/DietPi/issues/761

Fourdee, 2017-02-18:
> https://github.com/Fourdee/DietPi/issues/761#issuecomment-280850108

closed 2017-02-18 by Fourdee