MichaIng/DietPi #766: Time is badly out of sync

despicablebapple, 2017-02-17:
I was experiencing a issue of time being out of sink by about 9 hours which was breaking some of my applications. 
After looking for a bit it turned out timedatectl had NTP turn off which I turned on by using this command timedatectl set-ntp TRUE and editing the /etc/systemd/timesyncd.conf file to uncomment the Servers line after doing this my issue with the time was fixed. 
This could be a issue with my raspberry pi only as I have customised it to my needs and installed applications not in the dietpi-software utility and have been using this system base for a long time but if not I though it would be useful to document in case other people have the same problem

Fourdee, 2017-02-17:
@despicablebapple 

> After looking for a bit it turned out timedatectl had NTP turn off which I turned on by using this command timedatectl set-ntp TRUE and editing the /etc/systemd/timesyncd.conf file to uncomment the Servers line after doing this my issue with the time was fixed.

Thanks for the report 👍  I'll take a look and see if this effects non-customized installations.

WolfganP, 2017-02-18:
It happened to me as well when I suffered the empty /dietpi folder issue.
Even is the system boots, there are some system services not starting or behaving wildly. Forcing a clock sync with 'ntpd -gq' got the system back to date temporarily till re-installation.

Fourdee, 2017-02-18:
@WolfganP 

> It happened to me as well when I suffered the empty /dietpi folder issue.

Yep, ```ntpd -gq``` is launched during boot ```/DietPi/boot```, and, in ```/etc/cron.daily/dietpi```. Both of these rely on ```/DietPi``` being available.

Fourdee, 2017-02-18:
All seems to be working correctly on a fresh install.

```timesyncd``` is disabled by default as DietPi will launch ```ntpd -gq```:
```
root@DietPi:~# systemctl status systemd-timesyncd -l
● systemd-timesyncd.service - Network Time Synchronization
   Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; disabled)
   Active: inactive (dead)
     Docs: man:systemd-timesyncd.service(8)
```

```
root@DietPi:~# cat /var/log/ntpd.log
18 Feb 13:17:16 ntpd[461]: proto: precision = 1.042 usec
18 Feb 13:17:16 ntpd[461]: Listen and drop on 0 v4wildcard 0.0.0.0 UDP 123
18 Feb 13:17:16 ntpd[461]: Listen and drop on 1 v6wildcard :: UDP 123
18 Feb 13:17:16 ntpd[461]: Listen normally on 2 lo 127.0.0.1 UDP 123
18 Feb 13:17:16 ntpd[461]: Listen normally on 3 lo ::1 UDP 123
18 Feb 13:17:16 ntpd[461]: peers refreshed
18 Feb 13:17:16 ntpd[461]: Listening on routing socket on fd #20 for interface updates
18 Feb 13:17:16 ntpd[461]: Deferring DNS for 0.debian.pool.ntp.org 1
18 Feb 13:17:16 ntpd[461]: Deferring DNS for 1.debian.pool.ntp.org 1
18 Feb 13:17:16 ntpd[461]: Deferring DNS for 2.debian.pool.ntp.org 1
18 Feb 13:17:16 ntpd[461]: Deferring DNS for 3.debian.pool.ntp.org 1
18 Feb 13:17:16 ntpd[473]: signal_no_reset: signal 17 had flags 4000000
18 Feb 13:17:18 ntpd[473]: host name not found: 0.debian.pool.ntp.org
18 Feb 13:17:18 ntpd[473]: host name not found: 1.debian.pool.ntp.org
18 Feb 13:17:18 ntpd[473]: host name not found: 2.debian.pool.ntp.org
18 Feb 13:17:18 ntpd[473]: host name not found: 3.debian.pool.ntp.org
18 Feb 13:17:19 ntpd[461]: Listen normally on 4 eth0 fe80::ba27:ebff:fef6:9f11 UDP 123
18 Feb 13:17:19 ntpd[461]: peers refreshed
18 Feb 13:17:21 ntpd[473]: host name not found: 0.debian.pool.ntp.org
18 Feb 13:17:21 ntpd[473]: host name not found: 1.debian.pool.ntp.org
18 Feb 13:17:21 ntpd[473]: host name not found: 2.debian.pool.ntp.org
18 Feb 13:17:21 ntpd[473]: host name not found: 3.debian.pool.ntp.org
18 Feb 13:17:25 ntpd[461]: Listen normally on 5 eth0 192.168.0.105 UDP 123
18 Feb 13:17:25 ntpd[461]: peers refreshed
18 Feb 13:17:27 ntpd[473]: DNS 0.debian.pool.ntp.org -> 217.114.59.66
18 Feb 13:17:27 ntpd[473]: DNS 1.debian.pool.ntp.org -> 109.74.206.120
18 Feb 13:17:27 ntpd[473]: DNS 2.debian.pool.ntp.org -> 176.58.109.199
18 Feb 13:17:27 ntpd[473]: DNS 3.debian.pool.ntp.org -> 178.79.155.116
18 Feb 13:18:44 ntpd[461]: ntpd: time set +5.625925 s
```

This is run during boot and daily cron: https://github.com/Fourdee/DietPi/issues/766#issuecomment-280845691

Fourdee, 2017-02-18:
@despicablebapple 

I believe you were right, this appears limited to your customized setup. 

In future, I'd highly recommend using our implemented ```dietpi-config``` > Time sync mode, to apply your required timesync mode. Using ```disabled``` would allow you to manually configure a method of your choice, outside of DietPi:
[image]

I'll mark this as closed. Please reopen if required.

closed 2017-02-18 by Fourdee