MichaIng/DietPi #769: DietPi does not boot after patch 144

cellulosa, 2017-02-18:
I have just updated diepi with Patch 144 and after rebooting it it won't boot. The SD card now appears unreadable.

Fourdee, 2017-02-18:
@cellulosa 

Hi, thanks for the report and sorry to hear your having this issue.

Which:
- Device (eg: RPi 3)
- SD card
- Power supply

cellulosa, 2017-02-18:
@Fourdee thanks for your reply. Unfortunately I don't have any external monitor, but I can see the Pi does not do anything (it keeps the red light and no other blinking appear). I think something happened with the SD card, if I run a fsck it returns `Invalid BS_jmpBoot in boot block: 000000`.

- Raspberry Pi Model B 512 MB RAM
- SDHC Sandisk Ultra 30 MB/s 16GB
- iZKA micro USB Mains Power Wall Supply Charger 1000mAh / 1A

Fourdee, 2017-02-20:
@cellulosa 

Thanks for the info.

> iZKA micro USB Mains Power Wall Supply Charger 1000mAh / 1A

This stands out as a possible cause, however, we did experience some issues linked to image corruption on a few of our previous images, that may of caused this to occur over time:
https://github.com/Fourdee/DietPi/issues/719#issuecomment-277533451

This should now be resolved, using our updated image ```DietPi_v143_RPi-armv6-(Jessie).img```.

If you are able to read the following file on partition 2 of the SD card, we can check if the image you used may of been effected:
```
cat /etc/.dietpi_image_version
```

cellulosa, 2017-02-20:
@Fourdee thanks for the support.

Running `cat /etc/.dietpi_image_version` on the Pi returns `142`.

I have now downloaded the updated image `DietPi_v143_RPi-armv6-(Jessie).img`. How do you recommend to proceed to manually update the Pi?

Fourdee, 2017-02-22:
@cellulosa 

> Running cat /etc/.dietpi_image_version on the Pi returns 142.

Ah yes, so this could be caused by the image, although, still cant rule out PSU.

> I have now downloaded the updated image DietPi_v143_RPi-armv6-(Jessie).img.  How do you recommend to proceed to manually update the Pi?

For fresh installations, DietPi updates automatically during 1st run to latest version.

> I have just updated diepi with Patch 144 and after rebooting it it won't boot. The SD card now appears unreadable.

If indeed the SD card is unreadable, and no info is available on HDMI output during boot, a fresh install is the only option.

We could attempt to recover user data (eg: nextcloud files etc) from your previous installation, which software did you install, and was a USB drive used?

cellulosa, 2017-02-22:
@Fourdee it's ok I will just do a fresh install. Thanks for your help!

cellulosa, 2017-02-27:
@Fourdee I have now bought the [Raspberry Pi Official Universal Power Supply Unit](https://www.amazon.co.uk/dp/B00MBH6XNS/ref=pe_1909131_77697121_tnp_email_TE_AMZLdp_1) and re-formatted the SD drive with the updated image `DietPi_v143_RPi-armv6-(Jessie).img`. However, after the first reboot (after updates) the Pi does not boot anymore.

Fourdee, 2017-02-28:
@cellulosa 

Thanks for the report, sorry to hear your still experiencing issues.

I've updated our RPi image to v145, please could you try the following image and let us know results:
http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z

Before writing this image, please also complete a full format on the SD card:
- download SD formatter https://www.sdcard.org/downloads/formatter_4/eula_windows/SDFormatterv4.zip
- Extract and Run
- Goto options, and select the following 
[image]
- Format (This may take 10-30 minutes) depending on write speed.

The above will write zero to all cells on the SD card. Once completed, write the DietPi image.

cellulosa, 2017-03-01:
@Fourdee 

Thank you. By following your process and formatting the SD cart and then installing the updated RPi image it all worked.

closed 2017-03-01 by cellulosa

WolfganP, 2017-03-04:
I suffered a booting issue with a brand new v144 image burn and boot installation on a RPi2. 
Burned the image to an SD card, started configuring it, used for a while, and after that the system didn't boot (just a few green led indications of storage being read, but no display on HDMI so I don't know where the system stands up, as no FTP either)

I removed the the SD card and analyzed it. The FAT partition seems OK (no SDcard block errors, all files seem present including /dietpi), but didn't proceed to analyze the linux type 83 partition (I've just looked for another SD card and reinstalled everything, now RPi boots OK).

Anything you want me to check on the offending SD card?

WolfganP, 2017-04-11:
@Fourdee After having issues with booting again after a reboot (RPi2 read the SDcard for several seconds and then nothing, not even reach initializing the HDMI display for further troubleshooting), I analyzed the SDcard and everything on the FAT partition seems OK, but I can't open the rootfs partition on a PC. Is it an standard ext4 partition or did you protect/disguise it somehow?
Thx.

Fourdee, 2017-04-13:
@WolfganP 

>  but I can't open the rootfs partition on a PC. Is it an standard ext4 partition or did you protect/disguise it somehow?

All our images use a standard ext4 partition on partition 2. If your system supports ext4, it should be visible. If not, it sounds like a failed image write or unstable SD card.
[image]

Have you tried: https://github.com/Fourdee/DietPi/issues/769#issuecomment-283030267 (or ```dd if=/dev/zero of=dev/sdCARD```) to full format SD card prior to image write?