MichaIng/DietPi #770: DietPi | Add support and optimizations for CHIP

Famicoman, 2017-02-18:
Are there any plans to support the $9 CHIP computer?  
https://getchip.com/pages/chip

Fourdee, 2017-02-19:
@Famicoman 

Hi, thanks for the request.

> Are there any plans to support the $9 CHIP computer?

If we can get our hands on one, yes. Although it works out at roughly £30 after shipping + import tax to UK :frowning_face: 

I'll send CHIP an email, see if they would be willing to send us a dev sample board, allowing us to achieve an DietPi image and support for this device.

Fourdee, 2017-02-19:
Tweet, should get more attention than email, show your support and should help get things rolling 👍 : https://twitter.com/DietPi_/status/833279357380591617

Famicoman, 2017-02-19:
Thanks guys, hope they end up sending one your way!

Fourdee, 2017-02-22:
Received a few donations, ordered a CHIP kit for $43.

aliosa27, 2017-02-23:
I have tried to do this on my chip's a few times, however things break when trying to recreate a ubifs from the installed, trimmed system, ill tackle this over the next week. time off work!

woulf, 2017-03-31:
I have an unused chip and hdmi shield if it can help

Fourdee, 2018-04-05:
Marking as closed due to the following:
- I lack the device.
- PREP can be used to convert an existing Debian pre-image into DietPi: https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh

closed 2018-04-05 by Fourdee