MichaIng/DietPi #772: Multiple drives; removed one

oshank, 2017-02-20:
I had 2 USB HDD's set up for a while with no issues, I moved all the userdata to the HDD I plan on keeping attached to the Pi. I shutdown and removed the other drive. After rebooting I cannot access dietpi-software or dietpi-drive_manager. It keeps getting hung up on something and will not provide me with an interface. It also has a hard time shutting down with only one drive plugged in. With both drives plugged in, it has no issue providing me with an interface for drive_manager and software or shutting down.

oshank, 2017-02-21:
Well I might have solved it already; with both drives mounted, I went into drive_manager and changed the mount type to UUID and had it always mount to /mnt/usb_2/.

oshank, 2017-02-21:
Just kidding, it remounted to usb_1 rather than usb_2

Fourdee, 2017-02-22:
@oshank 

Hi,

Thanks for the report 👍 

> Well I might have solved it already; with both drives mounted, I went into drive_manager and changed the mount type to UUID and had it always mount to /mnt/usb_2/.

Yes, when not using UUID mount type, and 2 usb drives, the mount location "could" change/swap during subsequent reboots.

In DietPi v145 (in devolpment), we have made significant changes to ```dietpi-drive_manager``` and UUID is now set for new drives, by default. This should prevent this issue from re-occurring:
https://github.com/Fourdee/DietPi/issues/751#issuecomment-280648324

oshank, 2017-02-22:
`blkid`
/dev/mmcblk0p2: UUID="859cf567-e7d3-4fa7-82b8-cb835cd272c6" TYPE="ext4" PARTUUID="0313e829-02"
/dev/sda1: UUID="18775961-fe7e-4fe4-9527-f6e95883d7b9" TYPE="ext4" PARTLABEL="primary" PARTUUID="f2823ed5-844c-42e6-b971-688cdd601762"
/dev/mmcblk0: PTUUID="0313e829" PTTYPE="dos"
/dev/mmcblk0p1: SEC_TYPE="msdos" LABEL="boot" UUID="70D8-A768" TYPE="vfat" PARTUUID="0313e829-01"

`cat /etc/fstab`
#Internal Drives---------------------------------------------------
proc            /proc           proc    defaults				0 0
/dev/mmcblk0p1  /boot           vfat    defaults,noatime,discard		0 2
/dev/mmcblk0p2  /               ext4    defaults,noatime,discard		0 1
tmpfs 			/tmp  			tmpfs 	defaults,noatime,nodev,nosuid,mode=1777				0 0
tmpfs 			/var/log 		tmpfs 	defaults,size=20m,noatime,nodev,nosuid,mode=1777	0 0
tmpfs 			/DietPi 		tmpfs 	defaults,size=10m,noatime,nodev,nosuid,mode=1777	0 0

#Samba Client------------------------------------------------------
#/mnt/samba . Please use dietpi-config and the networking menu to setup this mount

#FTP Client Mount--------------------------------------------------
#/mnt/ftp_client . Please use dietpi-config and the networking menu to setup this mount

#External Drives---------------------------------------------------
#NB: Please use dietpi-drive_manager to setup and control your external drives.
UUID=18775961-fe7e-4fe4-9527-f6e95883d7b9       /mnt/usb_1      auto     defaults,noatime,nofail,x-systemd.automount  0       0
#/dev/sdb1       /mnt/usb_2      auto     defaults,noatime,nofail,x-systemd.automount  0       0
#/dev/sdc1       /mnt/usb_3      auto     defaults,noatime,nofail,x-systemd.automount  0       0
#/dev/sdd1       /mnt/usb_4      auto     defaults,noatime,nofail,x-systemd.automount  0       0
#/dev/sde1       /mnt/usb_5      auto     defaults,noatime,nofail,x-systemd.automount  0       0

#NFS Client Mount--------------------------------------------------
#/mnt/nfs_client . Please use dietpi-config and the Networking Options: NAS menu to setup this mount
#a swapfile is not a swap partition, no line here
#use  dphys-swapfile swap[on|off]  for that

`ls -lha /mnt`
drwxr-xr-x 10 root root 4.0K Feb 20 17:58 .
drwxr-xr-x 20 root root 4.0K Aug 28 19:03 ..
lrwxrwxrwx  1 root root   10 Feb 20 17:58 dietpi_userdata -> /mnt/usb_1
drwxr-xr-x  2 root root 4.0K Aug  7  2016 ftp_client
drwxr-xr-x  2 root root 4.0K Sep 20 18:17 nfs_client
drwxr-xr-x  2 root root 4.0K Aug  7  2016 samba
drwxrwxr-x 13 root root 4.0K Feb 20 17:58 usb_1
drwxr-xr-x  2 root root 4.0K Feb 20 17:58 usb_2
drwxr-xr-x  2 root root 4.0K Jan 11 16:00 usb_3
drwxr-xr-x  2 root root 4.0K Jan 11 16:00 usb_4
drwxr-xr-x  2 root root 4.0K Jan 11 16:00 usb_5

Fourdee, 2017-02-23:
@oshank 

> /dev/sda1: UUID="18775961-fe7e-4fe4-9527-f6e95883d7b9" TYPE="ext4" PARTLABEL="primary" 

> UUID=18775961-fe7e-4fe4-9527-f6e95883d7b9 /mnt/usb_1 auto defaults,noatime,nofail,x-systemd.automount 0 0

Thanks, that looks fine, USB drive will always mount to ```/mnt/usb_1```. The other drive is not attached, if you want to attach it, using ```dietpi-drive_manager``` to set it up should be fine. Just make sure to set mount type to UUID.

closed 2017-02-24 by oshank