MichaIng/DietPi #773: Error upon installation

floydianslips, 2017-02-21:
Only trying to install LXDE and Chromium:

Apt-get has failed:                                                │ 
     │  - Apt string = xcompmgr xterm xinit xauth xserver-xorg dbus-x11   │ 
     │ xfonts-base x11-xserver-utils x11-common x11-utils                 │ 
     │ --no-install-recommends                                            │ 
     │  - Error code = 100                                                │ 
     │  - Software Title = Xserver: linux display system

Fourdee, 2017-02-22:
@floydianslips 

Thanks for the report 👍 

Which device is this running on?

floydianslips, 2017-02-22:
Raspberry Pi 3. I spent a couple hours trying to get past this error. Tried LXFE and received errors as well. Never could get anything running on it. Please let me know if there is any way to help on this.

Fourdee, 2017-02-23:
@floydianslips 

> Raspberry Pi 3

Thanks 👍 

I'll run a test here to try and replicate.

Fourdee, 2017-02-23:
@floydianslips 

Ok, appears to work fine here on fresh DietPi install with RPi 3, may be limited to your current installation and package combination, or, apt mirror.

We'll need to debug this manually, **please run the following commands and paste results:**
```
apt-get install -y xcompmgr xterm xinit xauth xserver-xorg dbus-x11 xfonts-base x11-xserver-utils x11-common x11-utils --no-install-recommends
cat /etc/apt/sources.list
```

floydianslips, 2017-02-23:
I needed to commonadeer that SD card for another project unfortunately. 
I have tried a new card and just LXDE and it worked swimmingly. Sorry 
about that, I should have tried to keep that card.

If it helps the card is a Lexar 300x 16GB. I will try to load it on 
their again when I get a chance to test it. My goal with Diet Pi is to 
install Internet-in-a-Box (XSCE) on it. We are developing it to work on 
Raspberry Pi and I am hoping it will be easily installed on Diet Pi.

https://github.com/XSCE/xsce

Are there any plans to have Ansible as an install option in the future 
for Diet Pi?

Cheers,

Josh

Fourdee, 2017-02-27:
@floydianslips 

> Are there any plans to have Ansible as an install option in the future
for Diet Pi?

Interesting software 👍  I'll create a ticket and take a look when possible.
https://github.com/Fourdee/DietPi/issues/782

floydianslips, 2017-02-27:
Thanks. You have any plans on adding ansible as an install option?

#782 <https://github.com/Fourdee/DietPi/issues/782>

Fourdee, 2017-03-02:
@floydianslips 

> You have any plans on adding ansible as an install option?

Do now, lol 😄 
https://github.com/Fourdee/DietPi/issues/789

Marking this as closed (that is enough requests for 1 ticket lol 😉 )

I'll do some research when I start, but, in simple terms what does ansible actually do?

closed 2017-03-02 by Fourdee

floydianslips, 2017-03-07:
It is a process automation tool basically. We use it to install and 
enable all the components of our software. ie: moodle, kiwix server, web 
server, etc.

Fourdee, 2017-03-08:
@floydianslips 

Thanks.

I'll need to run some tests and see if Ansible will be compatible with DietPi, and visa versa.