MichaIng/DietPi #775: web admin's setting not able to start/enable dhcp server

tohjg, 2017-02-24:
issue transfer from [pihole](https://github.com/pi-hole/pi-hole/issues/1264)

Fourdee, 2017-02-24:
@tohjg 

> let's call it bob. added him as sudoer. got an error saying sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set while executing pihole command. i googled around about the error and found a magic command
chmod 4755 /usr/bin/sudo

Which device was this on? As far as I know, we dont specifically change ownership/uid bit of ```sudo```, defaults should be functional and not require a change.

tohjg, 2017-02-24:
m using [pine64](https://www.pine64.org/?product=pine-a64-board-1gb) 1gb board

i2064, 2017-03-12:
I am also experiencing this problem on a fresh install of DietPi and Pi-hole on Raspberry Pi 3.

> DietPi_v145_RPi-armv6-(Jessie).img
> Pi-hole Version v2.13.1 Web Interface Version v2.5.1 

I used the DietPi installer to install Pi-hole and the ad-blocking feature is not working. Clicking the Enable button on the left-side menu has no effect and setting "DHCP server enabled" is also ineffective.

I traced the php code to the following statement in `/var/www/html/admin/api.php`:

`exec('sudo pihole enable');`

Now, ad-blocking works if the statement `sudo pihole enable` is executed from the command line as root user. So something is not right with the way DietPi modifies the Pi-hole installer or in the way it is set up. The script is running under the www-data user and there is an entry for this in /etc/sudoers:

`www-data ALL=NOPASSWD: /usr/local/bin/pihole`

I had a fresh working install of Pi-hole on Raspbian before trying DietPi so I'll maybe compare the install scripts between both distributions tomorrow if I have some time.

Fourdee, 2017-03-13:
@i2064 

> Now, ad-blocking works if the statement sudo pihole enable is executed from the command line as root user. So something is not right with the way DietPi modifies the Pi-hole installer or in the way it is set up. 

Thanks for the report, i'll see if we can replicate at our end.

Fourdee, 2017-03-13:
@i2064 

Fresh install with v146 (which includes sudo permissions fix: https://github.com/Fourdee/DietPi/issues/794):
```
root@DietPi:~# pihole status
::: DNS service is running
::: Pi-hole blocking is Enabled

```

Disable in web interface:
```
root@DietPi:~# pihole status
::: DNS service is running
::: Pi-hole blocking is Disabled
```

> and setting "DHCP server enabled" is also ineffective.

Also appears to be working with v146

Looks like https://github.com/Fourdee/DietPi/issues/794 is the cause, this is already fixed for v146. I'll try and get v146 released in the next few days.

Marking as resolved.

closed 2017-03-13 by Fourdee