MichaIng/DietPi #776: DietPi-Software | Upgrade Pi-Hole to the latest version

midwan, 2017-02-24:
Is there a reason that Pi-hole is still on v2.9.5, while the are several versions after that available?

The current latest one is v2.12.1: https://github.com/pi-hole/pi-hole/releases

Fourdee, 2017-02-24:
@midwan 

> Is there a reason that Pi-hole is still on v2.9.5, while the are several versions after that available?

PiHole needs to be updated manually as it overwrites the ```lighttpd.conf```. i'd also recommend making a system backup before running this ```dietpi-backup```, as we cant predict/control what ```pihole -up``` will do:
```
mv /etc/lighttpd/lighttpd.conf /etc/lighttpd/lighttpd.conf.bak
pihole -up
mv /etc/lighttpd/lighttpd.conf.bak /etc/lighttpd/lighttpd.conf
```

midwan, 2017-02-24:
Yeap, I read that in the forum post.. but I was wondering why installing it from scratch using Dietpi-Software doesn't install the latest version?

Also, is it normal that I don't see a Login link in the Pi-hole web interface?

Fourdee, 2017-02-24:
> Yeap, I read that in the forum post.. but I was wondering why installing it from scratch using Dietpi-Software doesn't install the latest version?

Strange, ```dietpi-software``` does run the official installation script during install, it should be latest version:
http://install.pi-hole.net

> Also, is it normal that I don't see a Login link in the Pi-hole web interface?

Should be located on the left hand side. Is this a fresh installation of PiHole recently?

midwan, 2017-02-24:
When I installed it it got v2.9.5, maybe if I remove it and install it again it would get the latest one (didn't test that). :)

This is what I see on the left hand side:
[image]

Fourdee, 2017-02-24:
> When I installed it it got v2.9.5, maybe if I remove it and install it again it would get the latest one (didn't test that). :)

Yep, uninstall and install though ```dietpi-software``` would work.

In older versions, the login option is in the top right hand corner, the PiHole button. Just did a fresh install, Pi-hole Version v2.12.1 Web Interface Version v2.4, login button is on left hand side.

midwan, 2017-02-24:
Hm, I'll try a clean re-installation to see if it helps then

midwan, 2017-02-24:
OK, re-installing from scratch works.

Note: the site is now accessible under <IP>/admin instead of <IP>/pihole, you may want to update the documentation on the forum... ;-)

closed 2017-02-24 by midwan

Fourdee, 2017-02-24:
@midwan 

> Note: the site is now accessible under /admin instead of /pihole, you may want to update the documentation on the forum... ;-)

It should be available under both. We create a symlink for /pihole to /admin aswell. So either should work:
ip/pihole
ip/admin

midwan, 2017-02-24:
Yeap, but it comes up with this:

[image]

Fourdee, 2017-02-24:
@midwan 

Weird lol. Didn't happen for me on Chrome, which browser?

midwan, 2017-02-24:
@Fourdee 
I tried it with Chrome... :) The block is from PiHole itself it you notice, and that's why it has the option to whitelist it. However, when I try to whitelist it from here it fails with an "unknown error".

It's not a big deal for me, as I can use the /admin page without a problem. Just thought that you should know in case other people come across this.

Fourdee, 2017-02-24:
@midwan 

> Just thought that you should know in case other people come across this.

Aye, thanks 👍 

Its one of those days lol...
[image]